tmk/tmk_keyboard #485: Terminal-USB Converter Recognized in macOS but not Windows

adiron, 2017-08-18:
I built a converter running on an atmega32u4. I previously had Soarer's converter on it, and it worked fine, but I prefer TMK for the layers and such.

Anyway, I configured the PS/2 mode to `BUSYWAIT` (only mode that worked with this Set 3 AT/PS2 terminal connector Model M), programmed the layers to my liking, everything seemed to be on track.

Everything was fine as far as macOS is concerned, but when I plugged it into a Windows 10 machine, nothing works. I can't connect with hid_listen, I can't type anything, etc. The device *does* show up in the device manager, however.

What could cause such a thing?

tmk, 2017-08-18:
Try USART or INT?

tmk, 2017-08-19:
confirmed BUSYWAIT and INT works on Ubuntu and Windows10.

You better plug keyboard into converter first then conneted to computer USB port. If you don't you may suffer from some power/intialization issues because IBM keyboards gnerally draw much current at plugin/startup.

adiron, 2017-08-19:
USART doesn't work (displays read errors from the keyboard)
INT doesn't work either (displays wFF infinitely in the console and doesn't do anything)

I'm not sure what you mean about connecting the keyboard to the converter. I plugged and unplugged the controller with the keyboard already connected to it many times if that's what you mean. So if there were any init related things, they were likely not missed.

The keyboard in question is very old (1988), part number is 1391792 I think.

Anything else I should try?

tmk, 2017-08-19:
Yea, that is what I meant. I also got repeated wFF sometime when I connect keyboard after plug converter into computer. I suspect some combination of computer and converter hardware causes initialization problem.  

I guess fixing/improving initialization code may avoid this problem but have had no time to look into. If you want to pursue this problem, look into power line at startup/plugin with oscilloscope and examine code USB initialization process.

adiron, 2017-08-19:
Unfortunately I don't know that much about controllers/USBHID for now. I'll try to look into and see what happens.

It's interesting to note that Soarer's Converter does not suffer from any of these issues. Maybe contact him and see how he solved those problems.

tmk, 2017-08-21:
let us know if you get any clue!

Arakula, 2017-12-09:
adiron -
If you manage to contact Soarer, please let me know. You'd be the first in years to do that.
Soarer's Converter, BTW, works interrupt-based. Unfortunately, I don't have enough experience with tmk keyboard to give an educated guess on the initialization differences that might cause your problem.

adiron, 2017-12-09:
I haven't tried to contact him. I understand that he hasn't been very communicative.