tmk/tmk_keyboard #489: Put teh license up in the root folder

MironAtHome, 2017-10-15:
As soon as I see GPL anywhere near any project I run.

I can't use it. Even read it.

So, please do publish license, as to avoid unnecessary troubles. GPL does bond people, who read code and didn't know an project was GPL protected. And given time and all the trouble related to the GPL, I strongly recommend making GPL licensing of project as evident as humanly possible. Because currently your project has elements of aggressive preying on curious programmers by sharing out knowledge, which really is incorporated by unbreakable legal tenet.

tmk, 2017-10-15:
Can you create pull request for this issue?
Some of codes are published under Modified BSD License, do we need also copy of the license?

MironAtHome, 2018-01-06:
License was applied.
Thank you

closed 2018-01-06 by MironAtHome