tmk/tmk_keyboard #492: Keyboard powers down when host goes to sleep

kaihuang201, 2017-10-16:
I am using tmk on GH60 Rev. CHN ("Satan"). And even with `SLEEP_LED_ENABLE = yes`, the keyboard would power down when the host goes to sleep. However, I noticed that the sleep led works when the computer is booting up.

tmk, 2017-10-16:
Sleep led is just a toy and infringes USB spec, just disable it.
If you are happy to debuging it yourself I can help some.

kaihuang201, 2017-10-17:
Yeah, sleep LED is just a nice eye candy to have. But what I really wanted here is to have the keyboard sleep when the host goes to sleep, so I can wake up with the keyboard later, instead of having to press a button on the computer. I noticed in the doc/hooks.txt there are hooks for hybernating:
```
`hook_usb_suspend_entry(void)`  | When the device enters USB suspend state.
`hook_usb_suspend_loop(void)` | Continuously, while the device is in USB suspend state. *Default action:* power down and periodically check the matrix, causing wakeup if needed.
```
Do I just need to implement those in my `matrix.c` to make it work? What function should I call to wake up the keyboard? How do I wake up the host?

Your help would be very much appreciated!

tmk, 2017-10-17:
You don't need `SLEEP_LED_ENABLE` to turn keyboard  into sleep and wake it. If you are lucky it works by default. If not  debug USB related code around here:
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/protocol/lufa/lufa.c#L662-L667

And note that in some OS you need configure device setting to wake up computer by keyboard. I think FAQ in wiki has entry about this, check it.

tmk, 2017-10-17:
And  on second thought, yeah looking into why sleep led doesn't work would be useful to debug this perhaps. You can find related code in:
tmk_core/protocol/lufa/lufa.c
tmk_core/common/avr/sleep_led.c

kaihuang201, 2017-10-31:
I tried commenting out the whole while loop for suspension (https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/protocol/lufa/lufa.c#L662-L667), but GH60 Rev. CHN would still power down when the host hibernates. Any suggestions on what I can do to investigate this further?

Mean while, I finished building another 60% with a teensy-lc controller (also using tmk of course =)). And the sleep/wakeup works flawlessly with Linux, but it doesn't seem to work at all with Windows. I saw a related open issue with sleep/wake for Windows (https://github.com/tmk/tmk_keyboard/issues/435), any progress on that yet?

tmk, 2017-11-01:
what is your host? you have same issue on linux,  windows and mac?

I don't think #435 is related to your issue. No progress on #435 so far.

tmk, 2017-11-01:
What is exactly your problem? what do you mean by "power down"?

Let's forget about teensy-lc here, which is based on chibios and is probably different problem. TMK/Chibios is not well supported yet and we'll have to debug it more.

kaihuang201, 2017-11-01:
For the GH60 Rev. CHN, I have the same issue on Linux and Windows, but haven't tested on Mac. The keyboard shows the same behavior of turning off when the host goes to sleep, the LEDs all turn off even if I comment out the suspension while loop. It cannot wake up either Linux or Windows.

tmk, 2017-11-01:
hm, it seems to be legitimate behaviour, keyboard should try not to consume power as possible when host goes sleep. Sleep LED function is exceptional, it works during sleep and consume power in vain.
You have to check your code and build configuration of firmware, keyboard hardware(schematic) and host computer setting. Also check these FAQ entries.
https://github.com/tmk/tmk_keyboard/wiki/FAQ#wakeup-from-sleep-doesnt-work
https://github.com/tmk/tmk_keyboard/wiki/FAQ#problem-on-biosuefiresumesleepwakepower-cycles

For the record, I just confirmed that my HHKB and Alps64 can resume sleeping host on Linux and Windows10, my test computer is Thinkpad X201s.

tmk, 2017-11-01:
What is differnece of the GH60 Rev.CHN to genuine GH60? Do you have schematic of it?
And where can you get official firmware for it?

kaihuang201, 2017-11-04:
I never managed to find a schematic for GH60 Rev.CHN, I found the pinout on https://kbfirmware.com/ and then rewrote some functions in `matrix.c` based on that. 

I'm not sure where I can get the "official" firmware. The PCB came with a default firmware flashed, but I couldn't use it because I have a custom layout. So I flashed tmk keyboard at the very beginning.

closed 2023-11-14 by tmk