tmk/tmk_keyboard #493: How can I reset the layer_state to one layer only and switch off all higher layers?

tobinam, 2017-10-21:
Hi,

I want an action to set the layer-state to one layer only. All other layers should be switched off, regardless of their state when the action is called.

I might need somesting like in [this section of the keymap documentation](https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/doc/keymap.md#2210-bitwise-operation) .

> 2.2.10 Bitwise operation
> part indicates which part of 32bit layer state(0-7). bits is 5-bit value. on indicates when the action is executed.

I do not really understand what the *part* parameter means. Can anybody give me an example how I can activate say layer 0 only and turn off all other layers?

Kind regards,
Tobi

tmk, 2017-10-22:
I'll update and improve its description of the Bitwise action later.

hm, `ACTION_LAYER_SET` doesn't seems to work in some cases. This patch may work for you, can you confirm this?
```
diff --git a/tmk_core/common/action_code.h b/tmk_core/common/action_code.h
index 1eb6910..69eb545 100644
--- a/tmk_core/common/action_code.h
+++ b/tmk_core/common/action_code.h
@@ -267,7 +267,7 @@ enum layer_pram_tap_op {
 #define ACTION_LAYER_INVERT(layer, on)              ACTION_LAYER_BIT_XOR((layer)/4,   1<<((layer)%4),  (on))
 #define ACTION_LAYER_ON(layer, on)                  ACTION_LAYER_BIT_OR( (layer)/4,   1<<((layer)%4),  (on))
 #define ACTION_LAYER_OFF(layer, on)                 ACTION_LAYER_BIT_AND((layer)/4, ~(1<<((layer)%4)), (on))
-#define ACTION_LAYER_SET(layer, on)                 ACTION_LAYER_BIT_SET((layer)/4,   1<<((layer)%4),  (on))
+#define ACTION_LAYER_SET(layer, on)                 ACTION_LAYER_BIT_SET((layer)/4, ((1<<((layer)%4)) | (1<<5)),  (on))
 #define ACTION_LAYER_ON_OFF(layer)                  ACTION_LAYER_TAP((layer), OP_ON_OFF)
 #define ACTION_LAYER_OFF_ON(layer)                  ACTION_LAYER_TAP((layer), OP_OFF_ON)
 #define ACTION_LAYER_SET_CLEAR(layer)               ACTION_LAYER_TAP((layer), OP_SET_CLEAR)
```

tmk, 2017-10-22:
Hmm, I confirmed `ACTION_LAYER_SET(0, ON_PRESS)` works for me with 'ACTION_LAYER_TOGGLE(4)'.

closed 2019-05-29 by tmk