tmk/tmk_keyboard #494: One-shot modifier doesn't get recognized randomly on teensy-lc

kaihuang201, 2017-10-31:
The one shot modifier tap doesn't get recognized randomly on teensy-lc controller, and the tap action doesn't register the one shot modifier. This would usually happen a number of times in a row and then back to normal. Another keyboard with GH60 Rev. CHN and tmk firmware doesn't seem to have this issue.

kaihuang201, 2017-11-04:
Interestingly, the problem went away when I moved the `timer_read()` out of the timer diff macro as shown below:

https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_util.c#L61
```
#if (defined(ONESHOT_TIMEOUT) && (ONESHOT_TIMEOUT > 0))
+        int16_t time_now  = timer_read();
+        if (TIMER_DIFF_16(time_now, oneshot_time) >= ONESHOT_TIMEOUT) {
-        if (TIMER_DIFF_16(timer_read(), oneshot_time) >= ONESHOT_TIMEOUT) {
            dprintf("Oneshot: timeout\n");
            clear_oneshot_mods();
        }
#endif
```

It may be some weird timing issue that causes wrong values to be evaluated, but I do not intend to dig this further now that the problem is solved.

closed 2017-11-04 by kaihuang201

tmk, 2017-11-05:
The macro probably evaluates `timer_read()` twice and causes the problem.
`TIMER_DIFF` should be fixed I'll look into later.

https://github.com/tmk/tmk_keyboard/blob/51326077b085ad5c82cb0ea7c63405364c052cd5/tmk_core/common/timer.h#L28-L32