tmk/tmk_keyboard #495: Enable tmk_core debug printing?

kaihuang201, 2017-10-31:
How do I enable tmk_core debug print?

tmk, 2017-11-01:
Set these variables to true somewhere.

https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/debug.h#L49-L53

kaihuang201, 2017-11-03:
This still doesn't enable debug print that are inside **tmk_core**.

tmk, 2017-11-03:
Which one for example?

On 11/03/2017 11:49 AM, Kai Huang wrote:
> This still doesn't enable debug print that are inside *tmk_core*.

tmk, 2017-11-03:
And this may help?
https://github.com/tmk/tmk_keyboard/wiki/FAQ#debug-console

kaihuang201, 2017-11-03:
For example: https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_tapping.c#L42

My debug console works fine. It shows `Listening:` and `Keyboard starting`. And `dprintf` in `matrix.c` shows up as intended. But I can't see anything in tmk_core, unless I add `dprintf` myself.

tmk, 2017-11-03:
To enable debug print there you have to define `DEBUG_ACTION` somewhere, for example at line before these lines, or also you can define in your makefile.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_tapping.c#L9-L13

closed 2017-11-04 by kaihuang201