tmk/tmk_keyboard #497: One-shot modifier hit and miss in Remote Desktop

kaihuang201, 2017-11-09:
I am a big fan of the one-shot modifier feature, and I work with remote desktop a lot. 

I discovered that one-shot modifier fails to modify the key press that follows when typing inside a Windows Remote Desktop session frequently. 

Thoughts: (correct me if I'm wrong) current implementation only sets a modifier flag when sending the keyboard report when the one-shot modifier is active, however, it does not generate a de-facto keypress of the modifier key (e.g. Left Shift key down). And I suspect this to be the cause of this issue. 

Maybe generating a real keypress for the one-shot would solve the problem. I will let you know once I get around to experiment that.

tmk, 2017-11-09:
I think you have right idea. Yes, I like to see your code and result, let me know!
Thanks,

jfrmilner, 2019-08-22:
Solution
Change the RDP client settings as follows:
1. Click Options in the Remote Desktop Connection User Interface.
2. Select the Local Resources tab.
3. Under Keyboard > Apply Windows key combinations, select On the local computer.
After doing this, the SHIFT/ALT/CTRL keys work as expected in the RDP session including One-Shot and Macros