tmk/tmk_keyboard #501: action_code.h:187:41: error: expected expression before ‘{’ token #define ACTION_NO { .code = 0 } #1

243750496, 2017-11-20:
test platform Ubuntu budigie 17.10

i have no idea how to fix the last bug:
**action_code.h:187:41: error: expected expression before ‘{’ token  #define ACTION_NO                       { .code = 0 }**

what i had done:
```
sudo apt-get install avr-gcc

sudo apt-get install avr-libc

make
```
```

-------- begin --------
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mkdir -p obj_onekey_lufa
Compiling C: keymap.c
avr-gcc -c -mmcu=atmega32u4 -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=512 -DCONSOLE_ENABLE -DVERSION=unknown -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(USB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_onekey_lufa/keymap.lst -I. -I../../tmk_core -I../../tmk_core/common -I../../tmk_core/protocol -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lufa/LUFA-git -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_onekey_lufa_keymap.o.d  keymap.c -o obj_onekey_lufa/keymap.o 
In file included from ../../tmk_core/common/action.h:24:0,
                 from keymap.c:21:
keymap.c: In function ‘keymap_fn_to_action’:
../../tmk_core/common/action_code.h:187:41: error: expected expression before ‘{’ token
 #define ACTION_NO                       { .code = 0 }
                                         ^
keymap.c:98:23: note: in expansion of macro ‘ACTION_NO’
         action.code = ACTION_NO;
                       ^
../../tmk_core/rules.mk:557: recipe for target 'obj_onekey_lufa/keymap.o' failed
make: *** [obj_onekey_lufa/keymap.o] Error 1
atc@ATC:~/Downloads/tmk_keyboard-master/keyboard/onekey$ sudo apt-get install libgcrypt
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libgcrypt
atc@ATC:~/Downloads/tmk_keyboard-master/keyboard/onekey$ sudo apt-get install libgcrypt11
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libgcrypt11 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libgcrypt11' has no installation candidate
atc@ATC:~/Downloads/tmk_keyboard-master/keyboard/onekey$ sudo apt-get install libgcrypt20
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgcrypt20 is already the newest version (1.7.8-2ubuntu1).
The following packages were automatically installed and are no longer required:
  binutils-arm-none-eabi dvgrab gcc-arm-none-eabi libboost-wave1.62.0
  libcdparanoia0:i386 libgdl-3-5 libgdl-3-common libieee1284-3:i386
  libnewlib-dev libopenimageio1.7 libopenjpeg5 libopenshadinglanguage1.8
  libopensubdiv libopus0:i386 libpugixml1v5 libtheora-bin
  libtopmenu-client-gtk3-0 libunicap2 libvisual-0.4-0:i386 libweed0
  libwine-development lives-data lives-plugins mkvtoolnix ogmtools perlmagick
  scribus-data topmenu-gtk3 wine-stable-amd64 wine64-development
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 44 not upgraded.
atc@ATC:~/Downloads/tmk_keyboard-master/keyboard/onekey$ sudo apt-get install libgcrypt20-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  binutils-arm-none-eabi dvgrab gcc-arm-none-eabi libboost-wave1.62.0
[... 262 lines cut ...]
-------- end --------

atc@ATC:~/Downloads/tmk_keyboard-master/keyboard/onekey$ make -f Makefile

-------- begin --------
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mkdir -p obj_onekey_lufa
Compiling C: keymap.c
avr-gcc -c -mmcu=atmega32u4 -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=512 -DCONSOLE_ENABLE -DVERSION=unknown -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(USB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_onekey_lufa/keymap.lst -I. -I../../tmk_core -I../../tmk_core/common -I../../tmk_core/protocol -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lufa/LUFA-git -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_onekey_lufa_keymap.o.d  keymap.c -o obj_onekey_lufa/keymap.o 
In file included from ../../tmk_core/common/action.h:24:0,
                 from keymap.c:21:
keymap.c: In function ‘keymap_fn_to_action’:
../../tmk_core/common/action_code.h:187:41: error: expected expression before ‘{’ token
 #define ACTION_NO                       { .code = 0 }
                                         ^
keymap.c:98:23: note: in expansion of macro ‘ACTION_NO’
         action.code = ACTION_NO;
                       ^
../../tmk_core/rules.mk:557: recipe for target 'obj_onekey_lufa/keymap.o' failed
make: *** [obj_onekey_lufa/keymap.o] Error 1
```

243750496, 2017-11-20:
https://github.com/tmk/tmk_keyboard/tree/v2.9
after download this version the compile problem sloved but show up another problem 
```
atc@ATC:~/Downloads/tmk_keyboard-2.9/keyboard/onekey$ sudo make -f Makefile.pjrc dfu
dfu-programmer atmega32u4 erase
dfu-programmer: no device present.
../../tmk_core/rules.mk:426: recipe for target 'dfu' failed
make: *** [dfu] Error 1
atc@ATC:~/Downloads/tmk_keyboard-2.9/keyboard/onekey$ 

```

how to slove it?or any other way to load hex file? or what's the next step is if not load the hex file? 
btw i am using LEOPOLO FC 210TP
found drivers here:
https://github.com/nathanrosspowell/FC210TP

tmk, 2017-11-21:
are you sure your keyboard supports atmel DFU?
'make dfu' flashes firmware with atmel DFU bootloader.

243750496, 2017-11-21:
```
#define KEYMAPS_SIZE    (sizeof(keymaps) / sizeof(keymaps[0]))
#define FN_ACTIONS_SIZE (sizeof(fn_actions) / sizeof(fn_actions[0]))

/* translates key to keycode */
uint8_t keymap_key_to_keycode(uint8_t layer, keypos_t key)
{
    if (layer < KEYMAPS_SIZE) {
        return pgm_read_byte(&keymaps[(layer)][(key.row)][(key.col)]);
    } else {
        // fall back to layer 0
        return pgm_read_byte(&keymaps[0][(key.row)][(key.col)]);
    }
}

/* translates Fn keycode to action */
action_t keymap_fn_to_action(uint8_t keycode)
{
    action_t action;
    if (FN_INDEX(keycode) < FN_ACTIONS_SIZE) {
        action.code = pgm_read_word(&fn_actions[FN_INDEX(keycode)]);
    } else {
        action.code = ACTION_NO;
    }
    return action;
}

```
what's these line means? in official keymap.c  i havent seen them or something similar.
btw:i got these code also from https://github.com/nathanrosspowell/FC210TP
and i want to set my own keymap 

could u plese help me ? thx a lot.

[keymap.c.txt](https://github.com/tmk/tmk_keyboard/files/1491263/keymap.c.txt)

243750496, 2017-11-21:
/*
Copyright 2012 Jun Wako <wakojun@gmail.com>

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 2 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

/*
 * scan matrix
 */
#include <stdint.h>
#include <stdbool.h>
#include <avr/io.h>
#include <util/delay.h>
#include "print.h"
#include "debug.h"
#include "util.h"
#include "matrix.h"

#ifndef DEBOUNCE
#   define DEBOUNCE	5
#endif
static uint8_t debouncing = DEBOUNCE;

/* matrix state(1:on, 0:off) */
static matrix_row_t matrix[MATRIX_ROWS];
static matrix_row_t matrix_debouncing[MATRIX_ROWS];

static matrix_row_t read_cols(void);
static void init_cols(void);
static void unselect_rows(void);
static void select_row(uint8_t row);

inline
uint8_t matrix_rows(void)
{
    return MATRIX_ROWS;
}

inline
uint8_t matrix_cols(void)
{
    return MATRIX_COLS;
}

void matrix_init(void)
{
    debug_enable = true;
    debug_matrix = true;
    debug_mouse = true;
    // initialize row and col
    unselect_rows();
    init_cols();

    // initialize matrix state: all keys off
    for (uint8_t i=0; i < MATRIX_ROWS; i++) {
        matrix[i] = 0;
        matrix_debouncing[i] = 0;
    }
}

uint8_t matrix_scan(void)
{
    for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
        select_row(i);
        _delay_us(30);  // without this wait read unstable value.
        matrix_row_t cols = read_cols();
        if (matrix_debouncing[i] != cols) {
            matrix_debouncing[i] = cols;
            if (debouncing) {
                debug("bounce!: "); debug_hex(debouncing); debug("\n");
            }
            debouncing = DEBOUNCE;
        }
        unselect_rows();
    }

    if (debouncing) {
        if (--debouncing) {
            _delay_ms(1);
        } else {
            for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
                matrix[i] = matrix_debouncing[i];
            }
        }
    }

    return 1;
}

bool matrix_is_modified(void)
{
    if (debouncing) return false;
    return true;
}

inline
bool matrix_is_on(uint8_t row, uint8_t col)
{
    return (matrix[row] & ((matrix_row_t)1<<col));
}

inline
matrix_row_t matrix_get_row(uint8_t row)
{
    return matrix[row];
}

void matrix_print(void)
{
    print("\nr/c 0123456789ABCDEF\n");
    for (uint8_t row = 0; row < MATRIX_ROWS; row++) {
        phex(row); print(": ");
        pbin_reverse16(matrix_get_row(row));
        print("\n");
    }
}

uint8_t matrix_key_count(void)
{
    uint8_t count = 0;
    for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
        count += bitpop16(matrix[i]);
    }
    return count;
}

where these lines coming from ?  i got it searched no wiki and doc said something like that

but got understood other code from https://deskthority.net/workshop-f7/how-to-build-your-very-own-keyboard-firmware-t7177.html

tmk, 2017-11-22:
Q1. as comments says former function translate key event to keycode using keymaps you defined, and the latter maps special keycode into action code.
Q2. matrix.c is used to scan your switch matrix.
Q3. which lines?

243750496, 2017-11-22:
keymap.c
```
#define KEYMAPS_SIZE    (sizeof(keymaps) / sizeof(keymaps[0]))
#define FN_ACTIONS_SIZE (sizeof(fn_actions) / sizeof(fn_actions[0]))

/* translates key to keycode */
uint8_t keymap_key_to_keycode(uint8_t layer, keypos_t key)
{
    if (layer < KEYMAPS_SIZE) {
        return pgm_read_byte(&keymaps[(layer)][(key.row)][(key.col)]);
    } else {
        // fall back to layer 0
        return pgm_read_byte(&keymaps[0][(key.row)][(key.col)]);
    }
}

/* translates Fn keycode to action */
action_t keymap_fn_to_action(uint8_t keycode)
{
    action_t action;
    if (FN_INDEX(keycode) < FN_ACTIONS_SIZE) {
        action.code = pgm_read_word(&fn_actions[FN_INDEX(keycode)]);
    } else {
        action.code = ACTION_NO;
    }
    return action;
}

```
matrix.c:
```
/*
Copyright 2012 Jun Wako <wakojun@gmail.com>

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 2 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

/*
 * scan matrix
 */
#include <stdint.h>
#include <stdbool.h>
#include <avr/io.h>
#include <util/delay.h>
#include "print.h"
#include "debug.h"
#include "util.h"
#include "matrix.h"

#ifndef DEBOUNCE
#   define DEBOUNCE	5
#endif
static uint8_t debouncing = DEBOUNCE;

/* matrix state(1:on, 0:off) */
static matrix_row_t matrix[MATRIX_ROWS];
static matrix_row_t matrix_debouncing[MATRIX_ROWS];

static matrix_row_t read_cols(void);
static void init_cols(void);
static void unselect_rows(void);
static void select_row(uint8_t row);

inline
uint8_t matrix_rows(void)
{
    return MATRIX_ROWS;
}

inline
uint8_t matrix_cols(void)
{
    return MATRIX_COLS;
}

void matrix_init(void)
{
    debug_enable = true;
    debug_matrix = true;
    debug_mouse = true;
    // initialize row and col
    unselect_rows();
    init_cols();

    // initialize matrix state: all keys off
    for (uint8_t i=0; i < MATRIX_ROWS; i++) {
        matrix[i] = 0;
        matrix_debouncing[i] = 0;
    }
}

uint8_t matrix_scan(void)
{
    for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
        select_row(i);
        _delay_us(30);  // without this wait read unstable value.
        matrix_row_t cols = read_cols();
        if (matrix_debouncing[i] != cols) {
            matrix_debouncing[i] = cols;
            if (debouncing) {
                debug("bounce!: "); debug_hex(debouncing); debug("\n");
            }
            debouncing = DEBOUNCE;
        }
        unselect_rows();
    }

    if (debouncing) {
        if (--debouncing) {
            _delay_ms(1);
        } else {
            for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
                matrix[i] = matrix_debouncing[i];
            }
        }
    }

    return 1;
}

bool matrix_is_modified(void)
{
    if (debouncing) return false;
    return true;
}

inline
bool matrix_is_on(uint8_t row, uint8_t col)
{
    return (matrix[row] & ((matrix_row_t)1<<col));
}

inline
matrix_row_t matrix_get_row(uint8_t row)
{
    return matrix[row];
}

void matrix_print(void)
{
    print("\nr/c 0123456789ABCDEF\n");
    for (uint8_t row = 0; row < MATRIX_ROWS; row++) {
        phex(row); print(": ");
        pbin_reverse16(matrix_get_row(row));
        print("\n");
    }
}

uint8_t matrix_key_count(void)
{
    uint8_t count = 0;
    for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
        count += bitpop16(matrix[i]);
    }
    return count;
}

```

tmk, 2017-11-22:
if you can't find info on wiki or google you have to read source codes.

tmk, 2017-11-22:
what's actually your question or problem with those lines you quoted?

243750496, 2017-11-22:
what do these lines in matrix.c mean? i do a lot of work learning what's the meaning of 
inline 
uint8_t
static 
ifndef & endif

and compare the code with G60's matrix.c only these code get puzzled ,and let me check the code i post 
if still have question i will ask here :)
```
#define KEYMAPS_SIZE    (sizeof(keymaps) / sizeof(keymaps[0]))
#define FN_ACTIONS_SIZE (sizeof(fn_actions) / sizeof(fn_actions[0]))

/* translates key to keycode */
uint8_t keymap_key_to_keycode(uint8_t layer, keypos_t key)
{
    if (layer < KEYMAPS_SIZE) {
        return pgm_read_byte(&keymaps[(layer)][(key.row)][(key.col)]);
    } else {
        // fall back to layer 0
        return pgm_read_byte(&keymaps[0][(key.row)][(key.col)]);
    }
}

/* translates Fn keycode to action */
action_t keymap_fn_to_action(uint8_t keycode)
{
    action_t action;
    if (FN_INDEX(keycode) < FN_ACTIONS_SIZE) {
        action.code = pgm_read_word(&fn_actions[FN_INDEX(keycode)]);
    } else {
        action.code = ACTION_NO;
    }
    return action;
}

```

so now the only help i need is to understand what's the meaning of the code below and the using of **3 bold function**(
1.phex();
2.pbin_reverse16();
3.bitpop16();
)

inline
uint8_t matrix_rows(void)
{
    return MATRIX_ROWS;
}

inline
uint8_t matrix_cols(void)
{
    return MATRIX_COLS;
}

bool matrix_is_modified(void)
{
    if (debouncing) return false;
    return true;
}

inline
bool matrix_is_on(uint8_t row, uint8_t col)
{
    return (matrix[row] & ((matrix_row_t)1<<col));
}

void matrix_print(void)
{
    print("\nr/c 0123456789ABCDEF\n");
    for (uint8_t row = 0; row < MATRIX_ROWS; row++) {
        **phex(row); print(": ");
        pbin_reverse16(matrix_get_row(row));**
        print("\n");
    }
}

uint8_t matrix_key_count(void)
{
    uint8_t count = 0;
    for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
        count += **bitpop16(matrix[i]);**
    }
    return count;
}

tmk, 2017-11-22:
On 11/22/2017 03:01 PM, 243750496 wrote:
> what do these lines in matrix.c mean? i do a lot of work learning what's 
> the meaning of
> inline
> uint8_t
> static
> ifndef & endif

those are general words in C, not specific to this project. You can 
learn them easily with book and web search. Take time to learn about C 
and AVR programing.

> and compare the code with G60's matrix.c only these code get puzzled
> ,and let me check the code i post
> if still have question i will ask here :)
>
> |#define KEYMAPS_SIZE (sizeof(keymaps) / [...]

Those codes should be not needed if you are using current version of 
TMK. You will need them with old versions.

> so now the only help i need is to understand what's the meaning of the 
> code below and the using of *3 bold function*(
> 1.phex();
> 2.pbin_reverse16();
> 3.bitpop16();

The functions works like:
1. print 8-bit value in hexadecimal form
2. print 16-bit value in binary form in reverse
3. count "bit population" of 16-bit value

You can find their definitions under tmk_core/common directory, use your 
favorite search command like grep or ctags.

243750496, 2017-11-22:
I have another question that after understand all code, it's a hardware question: 
Shall i need to remove the old controler to use micro pro as a new one to get it have able to  flash my firmware

closed 2017-12-17 by 243750496