tmk/tmk_keyboard #502: GUI key not working

nikolapintaric, 2017-11-20:
My LGUI key stopped working for no reason but the HID listener get's something from the keyboard, Here's the output:
```
r/c 0123456789ABCDEF
00: 0000000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0100000000000000
r/c 0123456789ABCDEF
00: 0000000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000
```
When i change the key to another keycode it works

tmk, 2017-11-21:
My LGUI key is still working. And I can use the key with another code.

```
r/c 01234567
00: 00000000
01: 00000000
02: 00000000
03: 00000000
04: 10000000
05: 00000000
06: 00000000
07: 00000000
```

closed 2017-11-22 by tmk

gcollic, 2019-02-11:
Hello,
same problem here. @nikolapintaric did you find the source of your problem ?
I'm on Windows 10, and it happened out of the blue, I did not change anything about the keyboard, it could be an automatic OS update (but at first I thought it was a mechanical problem and hadn't the time to investigate so I don't remember when and which update could have provoked this). Are you on Windows 10 too ?
Thanks

tmk, 2019-02-11:
Try disable to 'boot magic' in makefile.

gcollic, 2019-02-12:
It worked ! Thanks so much !

wojalin, 2019-04-29:
I had the same issue on Mac. 
Helped me as well.
@tmk could you tell, what happens internally, that this solves the problem?

tmk, 2019-04-30:
see my reply above.