tmk/tmk_keyboard #503: opensuse compilation

voodoomsr, 2017-11-26:
Hello guys, I'm having issues building the hex file for my hhkb keyboard. I'm doing the same things I did on my mac but here in my pc with opensuse, I'm unable to build succesfully.

The error that I'm seeing is the following:

`In file included from keymap_common.h:20:0,
                 from keymap_ph.c:4:
/usr/lib64/gcc/avr/7/include/stdint.h:9:26: error: no include path in which to search for stdint.h
 # include_next <stdint.h>
                          ^
In file included from ../../tmk_core/common/action.h:22:0,
                 from keymap_common.h:23,
                 from keymap_ph.c:4:
../../tmk_core/common/keyboard.h:31:5: error: unknown type name 'uint8_t'
     uint8_t col;
     ^~~~~~~
../../tmk_core/common/keyboard.h:32:5: error: unknown type name 'uint8_t'
     uint8_t row;
     ^~~~~~~
`

the error message continues for a couple of lines, as it is not recognizing the type uint8_t and other related types.

I think it has something to do with the avr-gcc library. In opensuse I couldn't find it explicitly. What I found was the library cross-avr-gcc.

Someone has experience building this on this type of OS, that could share some help on this?

tmk, 2017-11-27:
Can you compile intact sources without your changes?
First to know is whether build tools, tmk sources or your changes.

what is your avr-libc and avr-gcc version?

voodoomsr, 2017-11-29:
My changes are minimum, I could reproduce the same behaviour with other keymaps files too. 
I couldn't find the avr-libc library for this distribution, so I used cross-avr-gcc instead. Maybe that is the problem. In the noon I will put more details about libraries versions, I don't have access to that pc in this moment.

closed 2018-01-23 by tmk