tmk/tmk_keyboard #505: 3 Question about makefile

243750496, 2017-12-17:
Q1:
* lufa/     - LUFA USB stack **For only Pro Micro? or all micro usb device?**
* pjrc/     - PJRC USB stack **For all Teensy series? or all mini usb device?**

Q2:
# Target architecture (see library "Board Types" documentation).
ARCH = AVR8
where to find the documentation

Q3:
# Boot Section Size in *bytes*
#   Teensy halfKay   512
#   Teensy++ halfKay 1024
#   Atmel DFU loader 4096
#   LUFA bootloader  4096
#   USBaspLoader     2048
OPT_DEFS += -DBOOTLOADER_SIZE=4096

Does it have Teensy 2.0++ choice?

tmk, 2017-12-17:
Q1. No it doesn't depend on usb connector. those two stacks can be used  with USB capbable 8bit AVR controller.  what is "123" you are refering?

Q2. check AVR GCC documentation.

Q3. I can remember. but try 4096 or 8192. You can check its bootloader section size on datasheet of AVR.

Q4. no. use lufa.

closed 2017-12-17 by tmk

243750496, 2017-12-18:
Q1-3:can i change lufa to 123 or something else?
i mean can i set teensy(**because i use teensy 2.0++ as controler**) or set gh60_cc because i write the code by my own (**so i set an assign to it**)

243750496, 2017-12-23:
Could u please give me an reply ?
Btw whether there have irc channel?and the server address is?

tmk, 2017-12-23:
As for Q1, hmm, still not sure what you are refering.  what is gh60_cc? I can't find anything in this repo.
Oh, are you asking that it is possible to set TARGET=gh60_123? if so yes but why don't you try yourself before asking? I'm confused.

Q3. any of them is not correct.  You need bootloader section size. as i said refer datasheet. But  DBOOTLOADER_SIZE doesn't matter so much, firmware works with wrong value.

I think 16MHz is or for the teensy.

I don't think I have time much to answer these kind fo questions. you better find good mentor that can help you.

243750496, 2017-12-23:
```
# Target file name (without extension).
TARGET = gh60_lufa
```
Q1-3:can i change gh60_lufa to gh60_teensy as i want (because i use teensy 2.0++ controler)

tmk, 2017-12-24:
`TARGET` is just used as name of file and directory during build process, doesn't affect build configuration.