tmk/tmk_keyboard #506: Help me write a user defined function for 4-shift-state mapping?

DreymaR, 2017-12-18:
I want to add AltGr mappings to a few keys in my layout:
* Pressing the key with/-out Shift just sends the key (the Shift state will sort itself out)
* Pressing the key with AltGr/RAlt sends Unicode input depending on the Shift state
    * E.g., the A key should send A/a without AltGr but Æ/æ with AltGr
* I'm prepared to # define a way to send Unicode (it will be OS dependent)

So, the user function should:
* Check the state of the AltGr modifier
* If no AltGr, send the key
* If AltGr but no Shift, send Unicode four-digit string 1 (by the defined method)
* If AltGr and Shift, send Unicode four-digit string 2

I can't figure out how to do this, sorry... :-(

tmk, 2017-12-20:
You can get current modifier state by `get_mods()` and `get_weak_mods()`.
https://github.com/tmk/tmk_keyboard/blob/51326077b085ad5c82cb0ea7c63405364c052cd5/tmk_core/common/action_util.h#L37

those function returns 8-bit value whose each bit implies modifier state(1:on 0:off).
```
Modifier state:                                  
76543210
|||||||`-- LCTRL
||||||`--- LSHIFT                                
|||||`---- LALT                                  
||||`----- LGUI
|||`------ RCTRL
||`------- RSHIFT                                
|`-------- RALT                                  
`--------- RGUI
```

So check them like below:
```
if (get_mods() & MOD_BIT(KC_RALT)) {
  // with AltGr
  if (get_mods() & (MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT))) {
     // with Shift
  } else {
     // without Shift
  }
else {
  // without AltGr
}

DreymaR, 2018-01-11:
OK, I almost got it now. My function works for AltGr vs non-AltGr, and correctly sends Unicode. Yay!

**BUT**: It can't see the Shift state! I've tried polling both strong and weak mods. Thought the problem might be sticky Shift (ONESHOT_MOD), but with that deactivated the problem still persists.

If I set `shft` below to 1 manually, I get the capitalized letters as I should. What's wrong?

```
#include "action_util.h"
. . .
            if (record->event.pressed) {
                if ( get_mods() & MOD_BIT(KC_RALT) ) {          /* If AltGr is pressed:     */
                    uint8_t shft = ( get_mods()         | get_weak_mods()    )
                                 & ( MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT) );
                    shft = shft ? 1 : 0;
```

tmk, 2018-01-11:
Ah, you can't check oneshot mods state at this time unfortunately. `get_oneshot_mods()` should be added probably.

If you are using normal shift modifier keycode instead of oneshot shift your code seems to work.

DreymaR, 2018-01-12:
No, it isn't working for me. Even if I make sure to define the key as KC_LSFT, or if test for MOD_BIT(KC_LCTL) instead, my shft bit isn't set. Adding a line like this `shft = shft ? 0 : 1; /* debug */` after the above will make the letters capital but then I cannot get the unshifted letters. Ugh.

As for `get_oneshot_mods()`, maybe that'd be nice. But for now I'll settle for checking whether a Shift key is pressed down! Does `record->event.key` contain only the key that triggered the callback? How can I check whether a key mapped to `KC_LSFT` is being held down?

tmk, 2018-01-13:
it seems we missed something in your code, otherwise it is a bug of tmk_core.
I think you have to see what get_mods() and get_weak_mods() actually return with xprintf() and hid_listen, for example,
`xprintf("%02X %02X\n", get_mods(), get_weak_mods());`

And if you can post all of your code here it would be helpful to track the problem.

DreymaR, 2018-01-15:
Thanks for the helpful answers! I haven't understood or used debugging yet, and it seems my BASH for Windows doesn't have the hid_listen command.

But the pertinent code looks like this:
```
# define AC_FLBr ACTION_FUNCTION_OPT(FOURLVL,0)      // FLBr (LBRC key) is 4-level (åÅ)
# define AC_FRBr ACTION_FUNCTION_OPT(FOURLVL,1)      // FRBr (RBRC key) is 4-level (æÆ)
# define AC_FLgt ACTION_FUNCTION_OPT(FOURLVL,2)      // FLgt (NUBS key) is 4-level (øØ)
```
```
/* Extra includes/definitions (for FOURLVL user function): */
#include "action_util.h"
#define type_code(key) register_code(key); unregister_code(key)

/* UNICODEHEADER and UNICODEFOOTER handle OS dependency in the FOURLVL user function.                                 */
/* For XOrg,    Unicode 4-digit hex input uses Ctrl+Shift+u, ####, Enter.                                             
#define UNICODEHEADER add_weak_mods( MOD_BIT(KC_LCTL) | MOD_BIT(KC_LSFT) ); type_code( KC_U    );   // Ctrl+Shift+U,
#define UNICODEFOOTER del_weak_mods( MOD_BIT(KC_LCTL) | MOD_BIT(KC_LSFT) ); type_code( KC_ENT  );   // Del. mods; Enter.
*/ /* <-- FOURLVL Unicode header/footer for XOrg */
/* For MacOS,   Unicode 4-digit hex Unicode input uses Alt+#### (as Alt equals Mac Option).                           */
/* For Windows, Unicode 4-digit decimal nr. input uses Alt+Keypad(####).                                              */
#define UNICODEHEADER add_weak_mods( MOD_BIT(KC_LALT) );          // Hold LAlt,
#define UNICODEFOOTER del_weak_mods( MOD_BIT(KC_LALT) );          // Release LAlt.
/**/ /* <-- FOURLVL Unicode header/footer for MacOS and Windows (but they take different input codes!) */

void action_function(keyrecord_t *record, uint8_t id, uint8_t opt)
{
    /* Base keycode and 4-digit Windows input codes for several glyphs, for use with 4-level ISO adaptation           */
    static const uint8_t KeyToCode[3][2][5] =     {             /* [key#][shifted][codes]   */
        {{ KC_LBRC, KC_P0, KC_P2, KC_P2, KC_P9 },               /* å on Win is Alt+KP(0229) */
         {       0, KC_P0, KC_P1, KC_P9, KC_P7 }} ,             /* Å on Win is Alt+KP(0197) */
        {{ KC_RBRC, KC_P0, KC_P2, KC_P3, KC_P0 },               /* æ on Win is Alt+KP(0230) */
         {       0, KC_P0, KC_P1, KC_P9, KC_P8 }} ,             /* Æ on Win is Alt+KP(0198) */
        {{ KC_NUBS, KC_P0, KC_P2, KC_P4, KC_P8 },               /* ø on Win is Alt+KP(0248) */
         {       0, KC_P0, KC_P2, KC_P1, KC_P6 }} };            /* Ø on Win is Alt+KP(0216) */
    
    switch (id) {
        case FOURLVL:   /* Four-level keys, producing OS specific Unicode input with AltGr/RAlt held down             */
            if (record->event.pressed) {
                if ( get_mods() & MOD_BIT(KC_RALT) ) {          /* If AltGr is pressed:     */
                    uint8_t shft = ( get_mods()         | get_weak_mods()    )
                                 & ( MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT) );
                    shft = shft ? 1 : 0;    // shft = shft ? 0 : 1;    /* debug */
                    UNICODEHEADER                               /* 1) OS specific start,    */
                    send_keyboard_report();
                    for ( uint8_t i = 1; i < 5; i++ ) {         /* 2) Send the 4 digits,    */
                        type_code( KeyToCode[opt][shft][i] );
                    }
                    UNICODEFOOTER                               /* 3) OS specific finish.   */
                } else {                                        /* If AltGr isn't pressed:  */
                    type_code( KeyToCode[opt][0][0] );          /* -- Send the base key.    */
                }
                send_keyboard_report();
//          } else {                /* Do we need to do anything if our key isn't pressed?  */
            }
            break;
    }
}
```

Hope that makes sense! As said before, I've tested it and the keys seem to work well except for detecting the Shift state bit (`shft`).

tmk, 2018-01-16:
I tried your code and confirmed `shft` flag works as expected.

When pressing `AC_FLBr` with right ALT you will sends P0, P2, P2 and P9 with holding down **both** ALT keys. In the result this doesn't work with Windows(at least in US layout), because Unicode input requires only one of either Alt keys, not both. You will have to remove right ALT from mods before sending unicode numbers. Also you may need to remove Shift, though, it seems Windows don't care about Shift.

tmk, 2018-01-16:
You can download hid_listen from PJRC and I think you can use it on bash on windows probably.
https://github.com/tmk/tmk_keyboard/wiki#pjrc-hid_listen

DreymaR, 2018-01-17:
Excellent point! Of course I should clear (and afterwards restore) the Shift mods! The RAlt mod doesn't seem to bother me since I could get the unshifted glyphs without problem. But for safety I should stash that too, in case the device will be used with an OS which has 4-level layouts. This would be a problem with Linux and Mac I think, but not on Windows since the NumPad isn't AltGr sensitive there.

However, when I tried this I got a compiler error saying that I'm overwriting memory! Has my function become too large or what? I don't understand.
```
            if (record->event.pressed) {
                if ( get_mods() & MOD_BIT(KC_RALT) ) {          /* If AltGr is pressed:     */
                /* WIP/TODO: Clear/restore Shift (and RAlt?) mod or the below won't work! */
                    uint8_t mshft = get_mods()      & SHIFTMODBITS;
                    uint8_t wshft = get_weak_mods() & SHIFTMODBITS;
                    uint8_t shftd = ( mshft | wshft ) ? 1 : 0;  //shftd = 1;    /* debug */
//ERROR                    clear_mods(); clear_weak_mods();
//ERROR                    del_mods(mshft); del_weak_mods(wshft);      /* First, deactivate shift. */
//                    send_keyboard_report();
                    UNICODEHEADER                               /* 1) OS specific start,    */
                    send_keyboard_report();
                    for ( uint8_t i = 1; i < 5; i++ ) {         /* 2) Send the 4 digits,    */
                        type_code( KeyToCode[opt][shftd][i] );
                    }
                    UNICODEFOOTER                               /* 3) OS specific finish.   */
//ERROR                    add_mods(mshft); add_weak_mods(wshft);      /* Reactivate shift.        */
```
Uncommenting either of the //ERROR lines above results in compiler error with this message:
```
/usr/lib/gcc/avr/4.8.2/../../../avr/bin/ld: section .keymap loaded at [0000000000006800,0000000000006bff] overlaps section .data loaded at [0000000000006766,000000000000681b]
collect2: error: ld returned 1 exit status
make: *** [usb_usb_unimap.elf] Error 1
```

tmk, 2018-01-23:
I can still compile your code with that ERROR lines. 'make clean' can help you?

DreymaR, 2018-01-24:
I'm a good boy and run make clean every time! ^_^

Indeed, the problem doesn't seem to be any particular command but general memory shortage. According to the build info, my .elf file is 28395 dec (6eeb hex) which must be pushing the limits. The error I got suggested that the data area was overlapping the layout area; I haven't got more than 4 layers so maybe that block isn't full? If so, could I move the layout area so the data area becomes bigger?

Anyway, I got it to work now! **SUCCESS! YAY!** But:
• I didn't have room for the fourth key I wanted to remap. Ah well.
• I need to remove the RAlt mod so it doesn't activate menus on key release.

Here's the function now:
```
#include "action_util.h"
void type_code( uint8_t key1 );

void type_code( uint8_t key1 ) {
    register_code(key1);
    unregister_code(key1);
}
/* UNICODEHEADER and UNICODEFOOTER handle OS dependency in the FOURLVL user function.                                 */
/* For XOrg,    Unicode 4-digit hex input uses Ctrl+Shift+u, ####, Enter.                                             
#define UNICODEHEADER add_weak_mods( MOD_BIT(KC_LCTL) | MOD_BIT(KC_LSFT) ); type_code( KC_U    );   // Ctrl+Shift+U,
#define UNICODEFOOTER del_weak_mods( MOD_BIT(KC_LCTL) | MOD_BIT(KC_LSFT) ); type_code( KC_ENT  );   // Del. mods; Enter.
*/ /* <-- FOURLVL Unicode header/footer for XOrg */
/* For MacOS,   Unicode 4-digit hex Unicode input uses Alt+#### (as Alt equals Mac Option).                           */
/* For Windows, Unicode 4-digit decimal nr. input uses Alt+Keypad(####).                                              */
#define UNICODEHEADER add_weak_mods( MOD_BIT(KC_LALT) );          // Hold LAlt,
#define UNICODEFOOTER del_weak_mods( MOD_BIT(KC_LALT) );          // Release LAlt.
/**/ /* <-- FOURLVL Unicode header/footer for MacOS and Windows (but they take different input codes!) */

#define SHIFTBITS   ( MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT) )
void action_function(keyrecord_t *record, uint8_t id, uint8_t opt)
{
    /* Base keycode and 4-digit Windows input codes for several glyphs, for use with 4-level ISO adaptation           */
    /* TODO: Add the backslash key for two more glyphs (•→?). But it seems I'm out of data memory area...?            */
    static const uint8_t KeyDict[3][2][5] =         {               /* [key#][shifted][codes]       */
        {{ KC_LBRC, KC_P0, KC_P2, KC_P2, KC_P9 },                   /* å on Win is Alt+KP(0229)     */
         {       0, KC_P0, KC_P1, KC_P9, KC_P7 }}   ,               /* Å on Win is Alt+KP(0197)     */
        {{ KC_RBRC, KC_P0, KC_P2, KC_P3, KC_P0 },                   /* æ on Win is Alt+KP(0230)     */
         {       0, KC_P0, KC_P1, KC_P9, KC_P8 }}   ,               /* Æ on Win is Alt+KP(0198)     */
        {{ KC_NUBS, KC_P0, KC_P2, KC_P4, KC_P8 },                   /* ø on Win is Alt+KP(0248)     */
         {       0, KC_P0, KC_P2, KC_P1, KC_P6 }}   };              /* Ø on Win is Alt+KP(0216)     */
    
    switch (id) {
        case FOURLVL:   /* Four-level keys, producing OS specific Unicode input with AltGr/RAlt held down             */
            if (record->event.pressed) {
               uint8_t hmods = get_mods(); // uint8_t wmods = get_weak_mods();
                /* NOTE: Removing the RAlt mod to avoid menu activation means you must re-press RAlt for another go.  */
                /* NOTE: Due to data memory shortage, only the hard mods are considered. This seems enough for me.    */
                if ( hmods & MOD_BIT(KC_RALT) ) {                   /* - If AltGr is pressed:       */
                    uint8_t shftd = ( hmods & SHIFTBITS ) ? 1 : 0;
                    clear_mods();                                   /*   Temporarily remove mods,   */
                    UNICODEHEADER                                   /*   OS specific start,         */
                    send_keyboard_report();
                    for ( uint8_t i = 1; i < 5; i++ ) {             /*   Send the 4-digit code,     */
                        type_code( KeyDict[opt][shftd][i] );
                    }
                    UNICODEFOOTER                                   /*   OS specific finish,        */
                    set_mods( hmods & ~MOD_BIT(KC_RALT) );          /*   Restore mods (mostly).     */
                } else {                                            /* - If AltGr isn't pressed:    */
                    type_code( KeyDict[opt][0][0] );                /*   Send the base key.         */
                }
                send_keyboard_report();
//          } else {                           /* No need to do anything if our key wasn't pressed. */
            }
            break;
    }
}
```

tmk, 2018-01-25:
Glad to hear your success!

As for firmware size.
ATmega32u4 has 32KB flash memory  and bottom 4K section(0x7000-7fff) is reserved and proteced for bootloader. So you can use 28KB(0x0000-6fff) for your firmware basically, or 28672 in decimal size.

In addition to bootloader section 'unimap' build reserves 2K section at bottom for keymap. This is intended to make keymap edittable on web keymap editor. You can use 26KB or 26624 in decimal size for your firmware program except for keymap.

When using with old 8-bit 'keymap' and 'actionmap' you can assign full 28KB for program and key mapping automatically.

To reduce size, you can do first:
- comment out unused functions in Makefile, suchas MOUSEKEY, EXTRAKEY, CONSOLE ...

DreymaR, 2018-01-27:
Again, thanks!

It's strange, I don't have much memory trouble any more now. Maybe those 60 bytes you saved in the last update helped me? ^_^

I noted that I could disable the debug console and save a bunch though (I need mouse/extra keys in my layout so those have to remain on). Why is that on by default? I should think that most end users don't need it and the ones who do know how to turn it on?

On Windows, I noted that my former function worked well with Notepad but not Wordpad/Word/Outlook etc. I think that my use of the LALT mod activated the menus somehow. So eventually I discovered that I could just keep the RALT bit instead and it works! I've updated the function I posted above. Still not sure how well this works for Mac or Linux.

tmk, 2018-01-28:
To be honest, I'm astonished that you are doing coding this without debug console and printf :D Debug console is critical when I write and debug codes, and I often ask users infos from the console to track their issue when they are in trouble. I'd strongly recommend you have debug console as long as flash size allows while writing code at least.
https://github.com/tmk/tmk_keyboard/wiki#debug
https://github.com/tmk/tmk_keyboard/wiki/FAQ#debug-console

closed 2018-10-14 by tmk