tmk/tmk_keyboard #507: error: field name not in record or union initializer   #define ACTION(kind, param)             { .code = ((kind)<<12 | (param)) }

243750496, 2018-01-02:
```
atc@ATC:~/Downloads/tmk_keyboard-master/keyboard/onekey$ make

-------- begin --------
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mkdir -p obj_gh60_cc
Compiling C: keymap_gh60_cc.c
avr-gcc -c -mmcu=at90usb1286 -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=8192 -DCONSOLE_ENABLE -DVERSION=unknown -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(USB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_gh60_cc/keymap_gh60_cc.lst -I. -I../../tmk_core -I../../tmk_core/common -I../../tmk_core/protocol -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lufa/LUFA-git -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_gh60_cc_keymap_gh60_cc.o.d  keymap_gh60_cc.c -o obj_gh60_cc/keymap_gh60_cc.o 
keymap_gh60_cc.c:49:1: warning: braces around scalar initializer
 [0] = ACTION_LAYER_MOMENTARY(1)
 ^
keymap_gh60_cc.c:49:1: note: (near initialization for ‘fn_actions[0]’)
In file included from ../../tmk_core/common/action.h:24:0,
                 from keymap_common_gh60_cc.h:8,
                 from keymap_gh60_cc.c:1:
../../tmk_core/common/action_code.h:189:43: error: field name not in record or union initializer
 #define ACTION(kind, param)             { .code = ((kind)<<12 | (param)) }
                                           ^
../../tmk_core/common/action_code.h:259:53: note: in expansion of macro ‘ACTION’
 #define ACTION_LAYER_TAP(layer, key)                ACTION(ACT_LAYER_TAP, (laye
                                                     ^
../../tmk_core/common/action_code.h:271:53: note: in expansion of macro ‘ACTION_LAYER_TAP’
 #define ACTION_LAYER_ON_OFF(layer)                  ACTION_LAYER_TAP((layer), O
                                                     ^
../../tmk_core/common/action_code.h:265:53: note: in expansion of macro ‘ACTION_LAYER_ON_OFF’
 #define ACTION_LAYER_MOMENTARY(layer)               ACTION_LAYER_ON_OFF(layer)
                                                     ^
keymap_gh60_cc.c:49:7: note: in expansion of macro ‘ACTION_LAYER_MOMENTARY’
 [0] = ACTION_LAYER_MOMENTARY(1)
       ^
../../tmk_core/common/action_code.h:189:43: note: (near initialization for ‘fn_actions[0]’)
 #define ACTION(kind, param)             { .code = ((kind)<<12 | (param)) }
                                           ^
../../tmk_core/common/action_code.h:259:53: note: in expansion of macro ‘ACTION’
 #define ACTION_LAYER_TAP(layer, key)                ACTION(ACT_LAYER_TAP, (laye
                                                     ^
../../tmk_core/common/action_code.h:271:53: note: in expansion of macro ‘ACTION_LAYER_TAP’
 #define ACTION_LAYER_ON_OFF(layer)                  ACTION_LAYER_TAP((layer), O
                                                     ^
../../tmk_core/common/action_code.h:265:53: note: in expansion of macro ‘ACTION_LAYER_ON_OFF’
 #define ACTION_LAYER_MOMENTARY(layer)               ACTION_LAYER_ON_OFF(layer)
                                                     ^
keymap_gh60_cc.c:49:7: note: in expansion of macro ‘ACTION_LAYER_MOMENTARY’
 [0] = ACTION_LAYER_MOMENTARY(1)
       ^
../../tmk_core/rules.mk:557: recipe for target 'obj_gh60_cc/keymap_gh60_cc.o' failed
make: *** [obj_gh60_cc/keymap_gh60_cc.o] Error 1
atc@ATC:~/Downloads/tmk_keyboard-master/keyboard/onekey$ 

```
here's my source code (if it is needed for checking)

[onekey.zip](https://github.com/tmk/tmk_keyboard/files/1596896/onekey.zip)

tmk, 2018-01-04:
what if you use gcc 4.9.x insteaf of 5.4.0?

243750496, 2018-01-04:
sloved by change 

```
const uint16_t PROGMEM fn_actions[] = {
[0] = ACTION_LAYER_MOMENTARY(1)
};

```
To
```
const action_t PROGMEM fn_actions[] = {
[0] = ACTION_LAYER_MOMENTARY(1)
};
```

but why i cannot use uint16_t? is it only support this defination in old code?

New Q:How to disable led.c compile if i don't want to test led function now?
Btw:i changed 

```
# project specific files
SRC =	keymap_gh60_cc.c \
	matrix_gh60_cc.c \
	led_gh60_cc.c

CONFIG_H = config.h
```
To
```
# project specific files
SRC =	keymap_gh60_cc.c \
	matrix_gh60_cc.c# \
#	led_gh60_cc.c

CONFIG_H = config.h
```
but get an error 

```
obj_gh60_cc/protocol/lufa/lufa.o: In function `hook_usb_suspend_entry':
/home/atc/Downloads/tmk_keyboard-master/keyboard/onekey/../../tmk_core/protocol/lufa/lufa.c:694: undefined reference to `led_set'
obj_gh60_cc/common/keyboard.o: In function `keyboard_set_leds':
/home/atc/Downloads/tmk_keyboard-master/keyboard/onekey/../../tmk_core/common/keyboard.c:182: undefined reference to `led_set'
collect2: error: ld returned 1 exit status
../../tmk_core/rules.mk:550: recipe for target 'gh60_cc.elf' failed
make: *** [gh60_cc.elf] Error 1

```

tmk, 2018-01-04:
ah great. `action_t` is a 16-bit width construct , but not equivalent with `uint16_t `.
at some point in the past `fn_actions` was changed from `unit16_t` to `actoin_t`, I believe you refered to some old code.

I think you can't  omit `led_set` function at this point at least. 
But it would be useful that  tmk_core includes default implementation of `led_set` and users can omit it.

closed 2018-01-04 by tmk