tmk/tmk_keyboard #511: Left shift key clears any other held keys

bennyhahmeen, 2018-01-09:
Let me preface this by saying I tested using a hasu USB controller on HHKB Pro 2 (US), but I'm not sure if this is an issue with the board or TMK.

When I hold down any key and then press (or hold) left shift key, it clears the keys that I was previously holding. Any key pressed AFTER left shift is pressed and held will register fine however. 

This is easily demonstrated using the NKRO (n-key rollover) test on the following page: http://www.gigahype.com/nkey-rollover-test-page/

This behavior does not occur with the right shift key or any other modifier key that I can find.

This problem is specific to the left shift physical KEY, not the left shift keycode. The right shift key behaves fine when either the right shift or left shift actions are mapped to it. I tried remapping the left shift key to use the right shift action, but the problem persists.

This is a problem for any games where you press and hold WASD and then press left shift (to sprint, for instance). You can begin to move, then pressing shift instantly stops you because it clears any other key you were holding such as WASD.

bennyhahmeen, 2018-01-09:
For what it's worth, someone else reported this with the Leopold FC660C here, but I don't think they've filed an issue:

https://www.reddit.com/r/MechanicalKeyboards/comments/6ukkoj/strange_problem_after_changing_to_the_hasu/

tmk, 2018-01-09:
You referred to game, I'm assuming you are using Windows, right?
I tested with HHKB USB Alt controller with default keymap(LSFT) and didn't see problem in both the web page and switch hitter on Windows 10. Also no problem was found on Linux.

what is actually your os and the game you are suffering?
and can you share your keymap?

tmk, 2018-01-09:
Btw, my test method was:
1. hold w, a, s, d keys
2. hold Left Shift
And confirmed that any of the four keys were not released  by pressing Left Shift.

tmk, 2018-01-09:
Also confirmed FC660C Alt controler with default keymap doesn't show the problem on my site.
http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?fc660c_emu

tmk, 2018-01-09:
If this is not firmware issue, I'd recommend making sure internal connectors mate each other firmly, first.
Second, try other USB ports or powered HUB, quality of power source may cause sesing key switches.

bennyhahmeen, 2018-01-09:
This is now resolved. It was user error running Atmel Flip I believe and the updated firmware not actually being flashed to the controller. I flashed a new firmware using dfu-programmer from my Mac and resolved the issue. Thank you for looking at this! I believe the shift issue was caused by an old keymap I had and I was simply not able to successfully update it.

closed 2018-01-09 by bennyhahmeen