tmk/tmk_keyboard #513: change layer by normal key

RaymonSHan, 2018-01-10:
is it possible to change layer while i HOLDDOWN a normal key (such as char 'F'), then the key 'J' mean LEFT. and back to default layer when i release the normal key('F').
I know ,the difficulty is what to do when i press the 'F", is it means a normal press for f, or change layer.
I think it can be solve by send key press to PC when the key is RELEASE, not when the key is PRESS. if a normal key is HOLDDOWN, it always means change layer.
if it is possible, 40% keyboard is ENOUGH.

tmk, 2018-01-10:
yes, it is possible. It is called as 'dual-role key/modifier' in keyboard community.
https://en.wikipedia.org/wiki/Modifier_key#Dual-role_keys

In this firmeware the function is referred as 'tap key', btw.

RaymonSHan, 2018-01-10:
thanks for answer. 
and add

ACTION_LAYER_TAP_KEY(30, KC_F)

is enough for my sample?

tmk, 2018-01-10:
yes, i think it kind of works at least, it may depend on your usage and typing habit.
if you don't like how it works it is time to implement dual-role key yourself.

RaymonSHan, 2018-01-10:
in my image，
while F down,  right hand control cursor move
while V down, just as numlock is ON for right hand
while M down, become C-x ahead, B for buffer and F for file
the question is : is it possible to define multi key press in layer , to send C-x C-F to PC ?

tmk, 2018-01-10:
First, check wiki and tweak  your keymap and try the dual-role key yourself. If you don't have keyboard yet, get it first. Without keyboard  it is difficult to learn how to define keymap.
https://github.com/tmk/tmk_keyboard/wiki
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap

RaymonSHan, 2018-01-10:
my main keyboard is HHKB, while second is Poker 2. 
the poker flash its capslock when power on too frequently, and I am waiting my GH60 now.
i found this repo when choose new keyboard, and asked some newbie question.
thanks for you answers, and i will define my layer.

THANKS

closed 2018-01-11 by RaymonSHan