tmk/tmk_keyboard #514: Adafruit Feather 32u4 Bluefruit support (fork)

ramonpoca, 2018-01-11:
Hi! 

I've forked tmk_keyboard and committed my modifications to support Adafruit's Feather 32u4.
The feather BLE firmware supports HID emulation and works via AT commands.

The fork is at: https://github.com/ramonpoca/tmk_keyboard
I've added a bluefruitle_nrf51 folder and makefile include in tmk_core (a bit messy as I had to import a second arduino runtime). 

Right now I have it working with an old Datacomp in AT mode. The converter is at converters/ps2_bluefruitle.

The only problem I have right now is with the cursor keys producing keypad scancodes instead of cursor scancodes. Probably some layout/configuration issue. 

Any hints?
Also, would it be interesting to push the bluefruit support to tmk_core?

tmk, 2018-01-11:
Hi,
Actually what scancode do the cursor keys spit out?
I guess your keyboard uses code set 3(Terminal) instead of 2(PS/2).

ramonpoca, 2018-01-12:
The keyboard is a Datacomp DKF192A, and has selectable 88-286-30 dip switch. I use it in the 286 setting with the ps2_usb_rev2 converter via usb and works smoothly.

tmk, 2018-01-13:
hmm, then, I have no idea about the cursor keys.

As for Bluefruit LE support, yea, Ive been interested in. I'll try to get the board and see your code.

It seems qmk_firmware has suppport it, this may be useful to you(and me) for reference.
https://github.com/qmk/qmk_firmware/blob/9de443cbf10ac41bc15762c9c83f138f51dd3b9a/tmk_core/protocol/lufa/adafruit_ble.cpp

ramonpoca, 2018-01-15:
Ok, after messing my repo (turns out I have to use Busywait, INT leaves the keyboard blinking all leds at high frequency) I can confirm that:
- Keyboard DIP is in the 286 setting.
- send_keyboard receives 00-00-60-00-00-00-00-00 both for cursor-up and numpad 8 (and the same with other cursor keys (up 60, left 5C, right 5E, down 5A).
- Keymap linked in is keymap_plain.c from ps2_usb converter.

Right now only clue is the Busywait vs. INT, but I don't think that should change the keymap mapping.

ramonpoca, 2018-01-15:
As for qmk they've (re)implemented the Adafruit AT protocol. It's probably cleaner to adopt that than to drop all the Arduino tree, even though they only implement SPI communication, while Adafruit has different boards communicating over other channels with the bluetooth firmware (same AT command set though).

ramonpoca, 2018-01-15:
Doh! As I finished writing this I figured out both Adafruit and the keyboard handler where using INT1! xD

ramonpoca, 2018-01-15:
OOOOk! I'm dumb! I had copied the terminal matrix.c over the ps2 one! .
Now I've fixed it all keys work as God and IBM intended X-D.

ramonpoca, 2018-01-18:
I've created a PR (#517) for this, closing the issue.

closed 2018-01-18 by ramonpoca