tmk/tmk_keyboard #516: likely bugs in quantum/matrix.c#matrix_is_on and quantum/process_keycode/process_chording.c#process_chording

jjowdy, 2018-01-17:
I'm still working on getting my (Mac) environment so I can write and run tests. (I'm running into Xcode/clang issues with incompatible `-Wa,` flags (`-adhlns`) and haven't deciphered the customized makefiles yet.)

So, as a result, I can't yet test this with any confidence and haven't really determined whether this code is actually used. 

But, in any case, I've been reading through the code to try to understand the best way to configure my Ergodox EZ, and I spotted a few things that don't look like they could possibly be right:

(Need to use bit shift rather than lt operation.)
`
diff --git a/quantum/matrix.c b/quantum/matrix.c
index 5337e2626..d93b67353 100644
--- a/quantum/matrix.c
+++ b/quantum/matrix.c
@@ -235,7 +235,7 @@ bool matrix_is_modified(void)
 inline
 bool matrix_is_on(uint8_t row, uint8_t col)
 {
-    return (matrix[row] & ((matrix_row_t)1<col));
+    return (matrix[row] & ((matrix_row_t)1<<col));
 }
`

(Probably need to register/unregister all keys in the chord instead of doing an early exit after the first key.)
`
diff --git a/quantum/process_keycode/process_chording.c b/quantum/process_keycod
e/process_chording.c
index 6c6ebe300..0f076181e 100644
--- a/quantum/process_keycode/process_chording.c
+++ b/quantum/process_keycode/process_chording.c
@@ -66,6 +66,8 @@ bool process_chording(uint16_t keycode, keyrecord_t *record) {
           for (uint8_t i = 0; i < chord_key_count; i++) {
             register_code(chord_keys[i]);
             unregister_code(chord_keys[i]);
+          }
+          if (chord_key_count > 0) {
             return false;
           }
         }
`

tmk, 2018-01-17:
I'm a bit confused. are using Xcode to build firmeware?

Also are you sure you are using tmk_firmware? This is wrong place if you are talking about qmk_firmware.

jjowdy, 2018-01-17:
Whoops. I was totally confused! My apologies. I thought I was looking at qmk_firmware issues. 
As for Xcode, I'm trying to figure out the right toolchain to use. I'm using avr-gcc for building the keyboard configs, but wasn't sure what to use for running/writing googletest.

closed 2018-01-17 by jjowdy