tmk/tmk_keyboard #528: Utility functions for easier key entry?

DreymaR, 2018-01-24:
I'd like to suggest a few functions to enhance (un)`register_code()` functionality. You often want to press-release a – possibly modified – key, which can be done with these suggestions. Would it be a good idea?

I'm asking here instead of a pull request since it's a small matter and I'd like to hear your opinion on it first.

In tmk_core/common/action_util.h:
```
void type_code( uint8_t key1 );
void dual_code( uint8_t key1, uint8_t key2 );
```
In tmk_core/common/action_util.c:
```
void type_code( uint8_t key1 ) {
    register_code(key1);
    unregister_code(key1);
}

void dual_code( uint8_t key1, uint8_t key2 ) {
    register_code(key1);
    type_code(key2);
    unregister_code(key1);
}
```

DreymaR, 2018-01-25:
Just one thing: Originally I had done this the quick-n-dirty way, by `#define`:
```
#define type_code(key1)         register_code(key1);                  unregister_code(key1)
#define dual_code(key1,key2)    register_code(key1); type_code(key2); unregister_code(key1)
```

Well, some people dislike `#define` pseudofunctions and I see the risk of using them in general. In this case, however, I should think it'd be safe enough since the input is always well-formed and there's minimal risk of name confusion.

However, what is the best way? Am I right that neither way affects memory usage in the device? Anything else I should be aware of?

tmk, 2018-01-25:
to be honest, I hesitate to add those function in core, you can add them in your file easily when you need.

DreymaR, 2018-01-27:
Hmmm. I agree that dual/trpl may be overmuch. They're probably not that useful. But type_code() seems very very useful indeed! It's so much more readable and user-friendly than register/unregister for each and every key stroke you want to generate. And it seems quite safe.

Could we at least have this one in core? I'm gravitating towards the simple `#define` version:
```
#define type_code(key1)         register_code(key1);                  unregister_code(key1)
```

tmk, 2018-01-28:
OK. Added.

closed 2018-01-28 by tmk