tmk/tmk_keyboard #529: Can't build the teensy_lc_onekey keyboard

jpablobr, 2018-01-25:
Hi,

I just cloned the repo (also the ChibiOS and ChibiOS-Contrib repos) and then `cd`'d into the `keyboard/teensy_lc_onekey` directory and ran the `make` command but it failed with the following:

```
$ make
.Compiler Options
arm-none-eabi-gcc -c -mcpu=cortex-m0plus -O2 -ggdb -fomit-frame-pointer -falign-functions=16 -std=gnu99 -DPROTOCOL_CHIBIOS -ffunction-sections -fdata-sections -fno-common -include config.h -Wall -Wextra -Wundef -Wstrict-prototypes -Wno-missing-field-initializers -Wa,-alms=build/lst/ -DCORTEX_USE_FPU=FALSE -DBOOTMAGIC_ENABLE -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DCOMMAND_ENABLE -DNKRO_ENABLE -DSLEEP_LED_ENABLE -DNO_SUSPEND_POWER_DOWN -DVERSION=ad6059ad -DTHUMB_PRESENT -mno-thumb-interwork -DTHUMB_NO_INTERWORKING -MD -MP -MF .dep/build.d -I. -I../../tmk_core/tool/chibios/ChibiOS/os/license -I../../tmk_core/tool/chibios/ChibiOS/os/common/startup/ARMCMx/compilers/GCC -I../../tmk_core/tool/chibios/ChibiOS-Contrib/os/common/startup/ARMCMx/devices/KL2x -I../../tmk_core/tool/chibios/ChibiOS/os/common/ext/CMSIS/include -I../../tmk_core/tool/chibios/ChibiOS-Contrib/os/common/ext/CMSIS/KINETIS -I../../tmk_core/tool/chibios/ChibiOS/os/rt/include -I../../tmk_core/tool/chibios/ChibiOS/os/common/oslib/include -I../../tmk_core/tool/chibios/ChibiOS/os/common/ports/ARMCMx -I../../tmk_core/tool/chibios/ChibiOS/os/common/ports/ARMCMx/compilers/GCC -I../../tmk_core/tool/chibios/ChibiOS/os/hal/osal/rt -I../../tmk_core/tool/chibios/ChibiOS/os/hal/include -I../../tmk_core/tool/chibios/ChibiOS/os/hal/ports/common/ARMCMx -I../../tmk_core/tool/chibios/ChibiOS-Contrib/os/hal/ports/KINETIS/KL2x -I../../tmk_core/tool/chibios/ChibiOS-Contrib/os/hal/ports/KINETIS/LLD -I../../tmk_core/tool/chibios/ChibiOS-Contrib/os/hal/boards/PJRC_TEENSY_LC -I../../tmk_core/tool/chibios/ChibiOS/os/hal/lib/streams -I../../tmk_core/tool/chibios/ChibiOS/os/various -I../../tmk_core -I../../tmk_core/common -I../../tmk_core/protocol/chibios -I../../tmk_core/protocol -I. main.c -o main.o

Compiling crt0_v6m.S
Compiling chcoreasm_v6m.S
Compiling crt1.c
In file included from ../../tmk_core/tool/chibios/ChibiOS-Contrib/os/common/ext/CMSIS/KINETIS/kl2xz.h:32:0,
                 from ../../tmk_core/tool/chibios/ChibiOS-Contrib/os/common/startup/ARMCMx/devices/KL2x/cmparams.h:78,
                 from ../../tmk_core/tool/chibios/ChibiOS/os/common/startup/ARMCMx/compilers/GCC/crt1.c:27:
../../tmk_core/tool/chibios/ChibiOS-Contrib/os/common/ext/CMSIS/KINETIS/kl26z.h:85:10: fatal error: core_cm0plus.h: No such file or directory
 #include "core_cm0plus.h"        /* Cortex-M0+ processor and core peripherals */
          ^~~~~~~~~~~~~~~~
compilation terminated.
make: *** [build/obj/crt1.o] Error 1
```

I also tried to `include` the "core_cm0plus.h" file by modifying:

```diff
modified   tmk_core/tool/chibios/chibios.mk
@@ -179,6 +179,7 @@ INCDIR = $(CHIBIOS)/os/license \
          $(HALINC) $(PLATFORMINC) $(BOARDINC) $(TESTINC) \
          $(STREAMSINC) $(CHIBIOS)/os/various \
          $(TMK_DIR) $(COMMON_DIR) $(TMK_DIR)/protocol/chibios \
+         $(TMK_DIR)/tool/mbed/mbed-sdk/libraries/mbed/targets/cmsis \
          $(TMK_DIR)/protocol $(TARGET_DIR)

```

But still got this other error:

```
$ make
Compiling crt0_v6m.S
Compiling chcoreasm_v6m.S
Compiling crt1.c
Compiling vectors.c
Compiling chsys.c
In file included from ../../tmk_core/tool/chibios/ChibiOS/os/rt/src/chsys.c:36:0:
../../tmk_core/tool/chibios/ChibiOS/os/rt/include/ch.h:78:2: error: #error "obsolete or unknown configuration file"
 #error "obsolete or unknown configuration file"
  ^~~~~
make: *** [build/obj/chsys.o] Error 1
```

Seems like that isn't the right place (or way) to include those files, or not sure. Obviously I have very little idea of what I'm doing. Any help as to getting me in the right direction would be great.

I also tried building the teensy_lc* keyboards on [this fork](https://github.com/di0ib/tmk_keyboard/tree/master/keyboard) but they all failed in the same way.

(I'm on OS X)

tmk, 2018-01-25:
I guess Chibios or Chibios-Contrib changed something before and is not compatible now. I can build teensy_lc_oneky with using d78431c638 of ChibiOS(Jan 2017) and  b54282fa of ChibiOS-Contrib(Dec 2016).
I would be grateful if you can look into this problem and find fix to use the latest ChibiOS.

btw, Why did you try that fork by di0ib ? I'm interested if the repo has any good change or fix.

jpablobr, 2018-01-27:
Ok, here's what I've tried so far:

1.
Rollback ChibiOS to make sure everything compiles fine with those previous versions.

```
cd $tmkchibios & git checkout d78431c638
cd $tmkchibiosc & git checkout 328f640
cd $tmktlco & make
```

All good.

2.
Only revert ChibiOS (not ChibiOS-Contrib).

```
cd $tmkchibios & git checkout d78431c638
cd $tmkchibiosc & git checkout master 
cd $tmktlco & make
```

All good, so doesn't seem to be a problem with ChibiOS-Contrib.

3.
Only revert ChibiOS-Contrib (not ChibiOS). 

```
cd $tmkchibios & git checkout master
cd $tmkchibiosc & git checkout 328f640
cd $tmktlco & make
```

Same "missing core_cm0plus.h file" error. So seems to be a problem with ChibiOS

4. 
First ChibiOS commit (6b9d13411 - msg: PAL synchronous API.) to break the build

``` 
gco 6b9d13411
cd $tmktlco & make
```

Error:
```
In file included from ../../tmk_core/tool/chibios/ChibiOS/os/hal/include/hal.h:124:0,
                 from ../../tmk_core/tool/chibios/ChibiOS/os/hal/src/hal.c:25:
../../tmk_core/tool/chibios/ChibiOS/os/hal/include/hal_pal.h:933:45: error: unknown type name 'iopadid_t'; did you mean 'ioportid_t'?
   msg_t palWaitPadTimeoutS(ioportid_t port, iopadid_t pad, systime_t timeout);
                                             ^~~~~~~~~
                                             ioportid_t
make: *** [build/obj/hal.o] Error 1
```

Fixed it with (not sure if defining a type similar to `ioportid_id` would be better):

```diff
modified   os/hal/include/hal_pal.h
@@ -1001,12 +1001,12 @@ extern "C" {
   void palWriteBus(IOBus *bus, ioportmask_t bits);
   void palSetBusMode(IOBus *bus, iomode_t mode);
 #if PAL_USE_CALLBACKS || defined(__DOXYGEN__)
-  void palSetPadCallbackI(ioportid_t port, iopadid_t pad,
+  void palSetPadCallbackI(ioportid_uint16_t pad,
                           palcallback_t cb, void *arg);
   void palSetLineCallbackI(ioline_t line, palcallback_t cb, void *arg);
 #endif /* PAL_USE_CALLBACKS */
 #if PAL_USE_WAIT || defined(__DOXYGEN__)
-  msg_t palWaitPadTimeoutS(ioportid_t port, iopadid_t pad,
+  msg_t palWaitPadTimeoutS(ioportid_t port, uint16_t pad,
                            sysinterval_t timeout);
   msg_t palWaitLineTimeoutS(ioline_t line, sysinterval_t timeout);
 #endif /* PAL_USE_WAIT */
modified   os/hal/src/hal_pal.c
@@ -128,7 +128,7 @@ void palSetBusMode(IOBus *bus, iomode_t mode) {
  *
  * @api
  */
-void palSetPadCallbackI(ioportid_t port, iopadid_t pad,
+void palSetPadCallbackI(ioportid_t port, uint16_t pad,
                         palcallback_t cb, void *arg) {
 
   palevent_t *pep = pal_lld_get_pad_event(port, pad);
@@ -168,7 +168,7 @@ void palSetLineCallbackI(ioline_t line, palcallback_t cb, void *arg) {
  * @sclass
  */
 msg_t palWaitPadTimeoutS(ioportid_t port,
-                         iopadid_t pad,
+                         uint16_t pad,
                          sysinterval_t timeout) {
 
   palevent_t *pep = pal_lld_get_pad_event(port, pad);
```

5.
Next commit to break the build.

```
gco 33a3c8f73 # CMSIS update and cleanup. (3 months ago)
```

Lots of "/ChibioOS/os/common/" file/directory renames.

Error:
```
Compiling crt1.c
Compiling vectors.c
Compiling chsys.c
In file included from ../../tmk_core/tool/chibios/ChibiOS/os/rt/src/chsys.c:36:0:
../../tmk_core/tool/chibios/ChibiOS/os/rt/include/ch.h:78:2: error: #error "obsolete or unknown configuration file"
 #error "obsolete or unknown configuration file"
  ^~~~~
make: *** [build/obj/chsys.o] Error 1
```

Fixed with:

```diff
modified   keyboard/teensy_lc_onekey/chconf.h
@@ -29,6 +29,7 @@
 #define CHCONF_H
 
 #define _CHIBIOS_RT_CONF_
+#define _CHIBIOS_RT_CONF_VER_5_0_
 
 /*===========================================================================*/
 /**
```

6.
Next error at same commit (33a3c8f73).

```
Compiling crt0_v6m.S
Compiling chcoreasm_v6m.S
Compiling crt1.c
In file included from ../../tmk_core/tool/chibios/ChibiOS-Contrib/os/common/ext/CMSIS/KINETIS/kl2xz.h:32:0,
                 from ../../tmk_core/tool/chibios/ChibiOS-Contrib/os/common/startup/ARMCMx/devices/KL2x/cmparams.h:78,
                 from ../../tmk_core/tool/chibios/ChibiOS/os/common/startup/ARMCMx/compilers/GCC/crt1.c:27:
../../tmk_core/tool/chibios/ChibiOS-Contrib/os/common/ext/CMSIS/KINETIS/kl26z.h:85:10: fatal error: core_cm0plus.h: No such file or directory
 #include "core_cm0plus.h"        /* Cortex-M0+ processor and core peripherals */
          ^~~~~~~~~~~~~~~~
compilation terminated.
make: *** [build/obj/crt1.o] Error 1
```

Fixed with:

```diff
modified   tmk_core/tool/chibios/chibios.mk
@@ -175,6 +174,7 @@ ASMSRC =
 ASMXSRC = $(STARTUPASM) $(PORTASM) $(OSALASM)
 
 INCDIR = $(CHIBIOS)/os/license \
+         $(CHIBIOS)/os/common/ext/ARM/CMSIS/Core/Include \
          $(STARTUPINC) $(KERNINC) $(PORTINC) $(OSALINC) \
          $(HALINC) $(PLATFORMINC) $(BOARDINC) $(TESTINC) \
          $(STREAMSINC) $(CHIBIOS)/os/various \

```

The next set errors (on the same commit and master) are(seem) mostly due to the outdated `chconf.h` and possibly the `chibios/usb_main.c` which I'm also not familiar with. I'm getting errors along the lines of "undefined reference to `MS2ST'" in function `keyboard_idle_time_cb'.

I'll see if I can play with it a bit more to get the teensy_lc_* keyboards to build with the latest ChibiOS, but if not, at least, I can rollback to those previous versions (which too bad since the ChibiOS project seems quite active and probably it'll be well worth it).

jpablobr, 2018-01-27:
> btw, Why did you try that fork by di0ib ?

It was mostly because his teensy_lc examples are closer to what I want. I want to build something similar to the Planck. He also have other teensy_lc based keyboards - so I was just curious.

tmk, 2018-01-27:
Nice catch! Keep us updated.

As for not found error of "core_cm0plus.h" I think ChibiOS-Contrib should be updated, but it seems that not so many users of Kinetis are around there and the repo is not updated actively unfortunately. Yes, we Kinetis ChibiOS users should update it!

```
diff --git a/os/common/startup/ARMCMx/compilers/GCC/mk/startup_kl2x.mk b/os/common/startup/ARMCMx/compilers/GCC/mk/startup_kl2x.mk                                                            [0/1804]
index ca67f10..bf9bad3 100644
--- a/os/common/startup/ARMCMx/compilers/GCC/mk/startup_kl2x.mk
+++ b/os/common/startup/ARMCMx/compilers/GCC/mk/startup_kl2x.mk
@@ -6,7 +6,7 @@ STARTUPASM = $(CHIBIOS)/os/common/startup/ARMCMx/compilers/GCC/crt0_v6m.S
 
 STARTUPINC = $(CHIBIOS)/os/common/startup/ARMCMx/compilers/GCC \
              $(CHIBIOS_CONTRIB)/os/common/startup/ARMCMx/devices/KL2x \
-             $(CHIBIOS)/os/common/ext/CMSIS/include \
+             $(CHIBIOS)/os/common/ext/ARM/CMSIS/Core/Include \

              $(CHIBIOS_CONTRIB)/os/common/ext/CMSIS/KINETIS
 
 STARTUPLD  = $(CHIBIOS_CONTRIB)/os/common/startup/ARMCMx/compilers/GCC/ld
```