tmk/tmk_keyboard #533: Actions on function key combinations

fritziphantom, 2018-02-02:
Hello,
I built myself an Atreus keyboard and it's awesome! Now I'm trying to adjust the keymapping to my needs and most of it works as intended.
However, there's just one thing that I can't figure out. How do I define an action an a key combination? I'd like to have it like this:
Press FN1:          go to Layer 1
Press FN2:          go to Layer 2
Press FN1+FN2: go to Layer 3

For  now I have defined FN3, which goes to layer 3 and is in place of FN1 on layer 2 and vice versa. Unfortunately it gets stuck on layer 3 if I don't release the FN keys in the reverse order.
Is there a better way to do this? 
Thanks!

fritziphantom, 2018-02-02:
This might be a reason, I'm using the https://github.com/technomancy/tmk_keyboard fork. I'll try if I can use this repo here for it. Hope it is still ok to ask here.

Here is my keymap:

``` #include "keymap_common.h"

/* The default Atreus layout with ctrl and alt swapped, in case it was 
   soldered like this. First layer is normal keys, second
   (momentary fn layer) is numbers, most punctuation, and
   arrows. Third (modal, persistent) layer is function keys and other
   rarely-used keys. */

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
/* 0: BASE */
KEYMAP(KC_Q        ,KC_W       ,KC_E       ,KC_R       ,KC_T       ,KC_Y         ,KC_U       ,KC_I            ,KC_O              ,KC_P              ,\
KC_A               ,KC_S       ,KC_D       ,KC_F       ,KC_G       ,KC_H         ,KC_J       ,KC_K            ,KC_L              ,KC_SCLN           ,\
KC_FN30            ,KC_X       ,KC_C       ,KC_V       ,KC_B       ,KC_N         ,KC_M       ,KC_COMM         ,KC_DOT            ,KC_FN31           ,\
KC_ESC             ,KC_TAB     ,KC_LGUI    ,KC_FN1     ,KC_BSPC    ,KC_LALT      ,KC_LCTL    ,KC_SPC          ,KC_FN0            ,KC_MINS           ,KC_QUOT,KC_ENT),
                                                            // REMEMBER TO SWAP ALT AND CTL!!!! because it is wired like that
/* 1: RAISE  */
KEYMAP(KC_1        ,KC_2       ,KC_3       ,KC_4       ,KC_5       ,KC_6         ,KC_7       ,KC_8            ,KC_9              ,KC_0              ,\
KC_CAPSLOCK        ,KC_NO      ,KC_FN4     ,KC_FN5     ,KC_NO      ,KC_LEFT      ,KC_DOWN    ,KC_UP           ,KC_RIGHT          ,KC_EQUAL          ,\
KC_FN28            ,KC_NO      ,KC_FN7     ,KC_FN6     ,KC_INSERT  ,KC_RALT      ,KC_PGDN    ,KC_PGUP         ,KC_HOME           ,KC_FN29           ,\
KC_NO              ,KC_NO      ,KC_LGUI    ,KC_FN2     ,KC_DEL     ,KC_LALT      ,KC_LCTL    ,KC_SPC          ,KC_FN0            ,KC_NO             ,KC_NO  ,KC_ENTER),

/* 2: LOWER  */
KEYMAP( SHIFT(KC_1),SHIFT(KC_2),SHIFT(KC_3),SHIFT(KC_4),SHIFT(KC_5),SHIFT(KC_6)  ,SHIFT(KC_7),SHIFT(KC_8)     ,SHIFT(KC_9)       ,SHIFT(KC_0)       ,\
KC_NO              ,KC_NO      ,KC_NO      ,KC_NO      ,KC__VOLUP  ,KC_GRV       ,KC_PPLS    ,KC_BSLASH       ,KC_LBRACKET       ,KC_RBRACKET       ,\
KC_FN27            ,KC_NO      ,KC_NO      ,KC_NO      ,KC__VOLDOWN,SHIFT(KC_GRV),KC_NO      ,SHIFT(KC_BSLASH),SHIFT(KC_LBRACKET),SHIFT(KC_RBRACKET),\
KC_NO              ,KC_NO      ,KC_LGUI    ,KC_FN1     ,KC_BSPC    ,KC_LALT      ,KC_LCTL    ,KC_SPC          ,KC_FN2            ,KC_PSCREEN        ,KC_SLCK,KC_PAUSE),

    /* 3: EXTRA */
KEYMAP(  
KC_F1              ,KC_F2      ,KC_F3      ,KC_F4      ,KC_F5      ,KC_6         ,KC_F7      ,KC_F8           ,KC_F9             ,KC_F10,\
KC_NO              ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_NO             ,KC_F11,\
KC_NO              ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_NO             ,KC_F12,\
KC_FN3             ,KC_NO      ,KC_NO      ,KC_TRNS    ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_TRNS           ,KC_NO    ,KC_NO  ,KC_NO   ),

};

  //
const uint16_t PROGMEM fn_actions[] = {
  // change layers
  [0]  = ACTION_LAYER_MOMENTARY(1),  // RAISE
  [1]  = ACTION_LAYER_MOMENTARY(2),  // LOWER
  [2]  = ACTION_LAYER_MOMENTARY(3),  // EXTRA LAYER
  [3]  = ACTION_FUNCTION(BOOTLOADER),

  // handy functions
  [4]  = ACTION_MODS_KEY(MOD_LALT, KC_F4),
  [5]  = ACTION_MODS_KEY(MOD_LALT, KC_TAB),
  [6]  = ACTION_MODS_KEY(MOD_LCTL, KC_TAB),
  [7]  = ACTION_MODS_KEY(MOD_LCTL, KC_C),

  // DUAL ROLE SHIFT KEYS
  // layer: lower
  [27] = ACTION_MODS_TAP_KEY(MOD_LSFT, KC_NO),
  // layer: raise
  [28] = ACTION_MODS_TAP_KEY(MOD_LSFT, KC_NO), 
  [29] = ACTION_MODS_TAP_KEY(MOD_RSFT, KC_END),
  // layer: base
  [30] = ACTION_MODS_TAP_KEY(MOD_LSFT, KC_Z),      // FN30 is left shift and z
  [31] = ACTION_MODS_TAP_KEY(MOD_RSFT, KC_SLSH)    // FN31 is right shift and /

};

void action_function(keyrecord_t *record, uint8_t id, uint8_t opt)
{
  if (id == BOOTLOADER) {
    bootloader();
  }
}
```

tmk, 2018-02-02:
It makes sense, it seems that the repo forked before and didn't update. If you can use the latest tmk_core your keymap would work without fix. But with old core it doens't work due to layer stuck.
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#modifierlayer-stuck

Your current keymaps are configured like below, where L1, L2..., Ln indicate `ACTION_LAYER_MOMENTARY(n)`.

```
Layer   Left    Right
---------------------
0       L2      L1 
1       L3      L1 
2       L2      L3 
3       TRNS    TRNS
```

With old core it get stuck on Layer3 when you press L1 in Layer0 and L3(Left) on Layer1 then release L1(Right), for example. Because Layer1 turns off when releasing L1(Right) so Left key will act as L2(meaningless here) on Layer0, not L3 on Layer1.

You have to use some tricks to prevent situation above. This configuration allows you to select four diffent layers depending order of the two keys pressed. But if you need only three you can define Layer3 and Layer4 as same mapping.

```
Layer   Left    Right
---------------------
0       L2      L1
1       L3      L1
2       L2      L4
3       L3      L1
4       L2      L4
```

tmk, 2018-02-02:
## Note for users of the latest tmk_keyboard repo:
**You don't suffer from this copmlex issue anymore with current tmk_core.**

See 
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#modifierlayer-stuck

fritziphantom, 2018-02-02:
Thank you so much! Your trick worked, now it does what I want it to.

I tried to update the forked repository for some time, but I couldn't figure it out. Thank you again :)

Here is my keymap.c, maybe it is useful for somebody else who can't update to the newer tmk version:
```
#include "keymap_common.h"

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
/* 0: BASE */
KEYMAP(
KC_Q       ,KC_W       ,KC_E       ,KC_R       ,KC_T       ,KC_Y         ,KC_U       ,KC_I            ,KC_O              ,KC_P              ,\
KC_A       ,KC_S       ,KC_D       ,KC_F       ,KC_G       ,KC_H         ,KC_J       ,KC_K            ,KC_L              ,KC_SCLN           ,\
KC_FN30    ,KC_X       ,KC_C       ,KC_V       ,KC_B       ,KC_N         ,KC_M       ,KC_COMM         ,KC_DOT            ,KC_FN31           ,\
KC_ESC     ,KC_TAB     ,KC_LGUI    ,KC_FN2     ,KC_BSPC    ,KC_LALT      ,KC_LCTL    ,KC_SPC          ,KC_FN1            ,KC_MINS           ,KC_QUOT,KC_ENT)  ,
                                                            // REMEMBER TO SWAP ALT AND CTL!!!! because it is wired like that
/* 1: RAISE  */
KEYMAP(#include "keymap_common.h"

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
/* 0: BASE */
KEYMAP(KC_Q        ,KC_W       ,KC_E       ,KC_R       ,KC_T       ,KC_Y         ,KC_U       ,KC_I            ,KC_O              ,KC_P              ,\
KC_A               ,KC_S       ,KC_D       ,KC_F       ,KC_G       ,KC_H         ,KC_J       ,KC_K            ,KC_L              ,KC_SCLN           ,\
KC_FN30            ,KC_X       ,KC_C       ,KC_V       ,KC_B       ,KC_N         ,KC_M       ,KC_COMM         ,KC_DOT            ,KC_FN31           ,\
KC_ESC             ,KC_TAB     ,KC_LGUI    ,KC_FN2     ,KC_BSPC    ,KC_LALT      ,KC_LCTL    ,KC_SPC          ,KC_FN1            ,KC_MINS           ,KC_QUOT,KC_ENT),
                                                            // REMEMBER TO SWAP ALT AND CTL!!!! because it is wired like that
/* 1: RAISE  */
KEYMAP(KC_1        ,KC_2       ,KC_3       ,KC_4       ,KC_5       ,KC_6         ,KC_7       ,KC_8            ,KC_9              ,KC_0              ,\
KC_CAPSLOCK        ,KC_NO      ,KC_FN14    ,KC_FN15    ,KC_NO      ,KC_LEFT      ,KC_DOWN    ,KC_UP           ,KC_RIGHT          ,KC_EQUAL          ,\
KC_FN28            ,KC_NO      ,KC_FN17    ,KC_FN16    ,KC_INSERT  ,KC_RALT      ,KC_PGDN    ,KC_PGUP         ,KC_HOME           ,KC_FN29           ,\
KC_NO              ,KC_NO      ,KC_LGUI    ,KC_FN3     ,KC_DEL     ,KC_LALT      ,KC_LCTL    ,KC_SPC          ,KC_FN1            ,KC_NO             ,KC_NO  ,KC_ENTER),

/* 2: LOWER  */
KEYMAP( SHIFT(KC_1),SHIFT(KC_2),SHIFT(KC_3),SHIFT(KC_4),SHIFT(KC_5),SHIFT(KC_6)  ,SHIFT(KC_7),SHIFT(KC_8)     ,SHIFT(KC_9)       ,SHIFT(KC_0)       ,\
KC_NO              ,KC_NO      ,KC_NO      ,KC_NO      ,KC__VOLUP  ,KC_GRV       ,KC_PPLS    ,KC_BSLASH       ,KC_LBRACKET       ,KC_RBRACKET       ,\
KC_FN27            ,KC_NO      ,KC_NO      ,KC_NO      ,KC__VOLDOWN,SHIFT(KC_GRV),KC_NO      ,SHIFT(KC_BSLASH),SHIFT(KC_LBRACKET),SHIFT(KC_RBRACKET),\
KC_NO              ,KC_NO      ,KC_LGUI    ,KC_FN2     ,KC_BSPC    ,KC_LALT      ,KC_LCTL    ,KC_SPC          ,KC_FN5            ,KC_PSCREEN        ,KC_SLCK,KC_PAUSE),

/* 3: EXTRA */
KEYMAP(  
KC_F1              ,KC_F2      ,KC_F3      ,KC_F4      ,KC_F5      ,KC_6         ,KC_F7      ,KC_F8           ,KC_F9             ,KC_F10            ,\
KC_NO              ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_NO             ,KC_F11            ,\
KC_NO              ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_NO             ,KC_F12            ,\
KC_FN10            ,KC_NO      ,KC_NO      ,KC_FN3     ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_FN1            ,KC_NO             ,KC_NO,KC_NO   ),

/* 4: EXTRA */
KEYMAP(  
KC_F1              ,KC_F2      ,KC_F3      ,KC_F4      ,KC_F5      ,KC_6         ,KC_F7      ,KC_F8           ,KC_F9             ,KC_F10            ,\
KC_NO              ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_NO             ,KC_F11            ,\
KC_NO              ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_NO             ,KC_F12            ,\
KC_FN10            ,KC_NO      ,KC_NO      ,KC_FN2     ,KC_NO      ,KC_NO        ,KC_NO      ,KC_NO           ,KC_FN5            ,KC_NO             ,KC_NO,KC_NO   ),

};

  //
const uint16_t PROGMEM fn_actions[] = {
  // change layers
  //[0]  = ACTION_LAYER_MOMENTARY(1),  // RAISE
  [1]  = ACTION_LAYER_MOMENTARY(1),  // RAISE
  [2]  = ACTION_LAYER_MOMENTARY(2),  // LOWER
  [3]  = ACTION_LAYER_MOMENTARY(3),  // EXTRA LAYER
  [5]  = ACTION_LAYER_MOMENTARY(4),  // EXTRA LAYER
  [10] = ACTION_FUNCTION(BOOTLOADER),

  // handy functions
  [14]  = ACTION_MODS_KEY(MOD_LALT, KC_F4),
  [15]  = ACTION_MODS_KEY(MOD_LALT, KC_TAB),
  [16]  = ACTION_MODS_KEY(MOD_LCTL, KC_TAB),
  [17]  = ACTION_MODS_KEY(MOD_LCTL, KC_C),

[... 57 lines cut ...]
  [14]  = ACTION_MODS_KEY(MOD_LALT, KC_F4),
  [15]  = ACTION_MODS_KEY(MOD_LALT, KC_TAB),
  [16]  = ACTION_MODS_KEY(MOD_LCTL, KC_TAB),
  [17]  = ACTION_MODS_KEY(MOD_LCTL, KC_C),

  // DUAL ROLE SHIFT KEYS
  // layer: lower
  [27] = ACTION_MODS_TAP_KEY(MOD_LSFT, KC_NO),
  // layer: raise
  [28] = ACTION_MODS_TAP_KEY(MOD_LSFT, KC_NO), 
  [29] = ACTION_MODS_TAP_KEY(MOD_RSFT, KC_END),
  // layer: base
  [30] = ACTION_MODS_TAP_KEY(MOD_LSFT, KC_Z),      // FN30 is left shift and z
  [31] = ACTION_MODS_TAP_KEY(MOD_RSFT, KC_SLSH)    // FN31 is right shift and /

};

void action_function(keyrecord_t *record, uint8_t id, uint8_t opt)
{
  if (id == BOOTLOADER) {
    bootloader();
  }
}

```

closed 2018-03-03 by tmk