tmk/tmk_keyboard #535: Create custom key

Svalorzen, 2018-02-15:
In my keyboard I'd love to have a normal key that when pressed outputs "&", while when shifted it outputs "^".

I was wondering whether it would be simpler to create this key with the layers/actions (my guess is probably no), or if it would be possible to simply create a new `KC_MY_SPECIAL_KEY` that does what I need (and if so, which files should I look into to do this).

Thanks in advance for the help, and thanks for your awesome project!

tmk, 2018-02-16:
How to get the charactors depends on keyobard layout on your OS, but let's assume that  you are using US layout  here, for example.

To get '&' I would use `ACTION_MODS_KEY(MOD_LSFT, KC_7)` on a key in layer 0.
https://github.com/tmk/tmk_core/blob/master/doc/keymap.md#212-modified-key

then place `ACTION_LAYER_MODS(1, MOD_LSFT)` on Shift key in layer 0 to activate layer 1 when the  Shift is depressed.
https://github.com/tmk/tmk_core/blob/master/doc/keymap.md#35-momentary-switching-with-modifiers

and just locate '6' on the key in layer 1 to get '^'(Shift+6).

Svalorzen, 2018-02-16:
This seems easy enough, thanks, I'll definitely use it as a first approximation.

Do you have any pointers on what to look into in case it was also possible to define a new key? So I can try to do it in a fancier, layout independent way in my spare time =)

tmk, 2018-02-16:
I don't have any pointer but I think you can find something with search in github/google.

I'd prefer this way because it you can defien in keymap editor without writing code.

closed 2018-03-03 by tmk