tmk/tmk_keyboard #536: [pingmaster, arduino micro (clone)] only pins 1 (CLOCK) & 2 (DATA) work

mhinsch, 2018-02-16:
I've spent a couple of days trying to get the converter for a pingmaster to work using a cheap Chinese Arduino micro clone ("DM Micro-AU"). 

I tried a number of pin configurations changing the respective values in config.h each time. If I had the correct pin numbers configured the converter would boot until ``keyboard start`` and the internal speaker would work (with the wrong numbers it wouldn't), however, there was no input from the keyboard. I had the pull-up resistor connecting DATA and 5V installed at all times.

It turns out that the only working configuration was CLOCK on PD1 (SDA on the silkscreen) and DATA on PD0 (SCL). Swapping those two produces at least *some* kind of output (error messages of the form ``S:4D R:09FF``) , but only this combination actually works properly.

I have no idea if this is a hardware problem with the cheap clone board or a symptom of some underlying software issue, but even in the former case it might be helpful to mention it somewhere in the docs.

tmk, 2018-02-17:
You have to configure correctly to setup pin change interrupt when you change pin for clock.

https://github.com/tmk/tmk_keyboard/blob/master/converter/ibm4704_usb/config.h#L54-L58

mhinsch, 2018-02-17:
Ok, I wasn't aware of that. It would perhaps be helpful to mention that somewhere in the docs. I can't be the only one who tried using different pins.
If you would want to do that, is there a straightforward way to find out which interrupts to change to which value?
BTW, thanks a lot for the awesome work on the converter!

tmk, 2018-02-17:
Not sure what your question are but I think you will find answer in controller's datasheet .

Feel free to add useful info you found to wiki for future users!

mhinsch, 2018-02-17:
Ok, I'm pretty new to this embedded stuff, but I'll have a look around. If I find out how to do it I'll share the information :-).
I'm closing this since the original problem doesn't apply any more.

closed 2018-02-17 by mhinsch