tmk/tmk_keyboard #537: USB_USB Converter - Bootloader and code overlap. (Soft-Brick)

kaansoral, 2018-03-02:
> `make dfu KEYMAP=kaan`
```-n dfu-programmer: waiting

dfu-programmer atmega32u4 erase
Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
dfu-programmer atmega32u4 flash usb_usb.hex
Bootloader and code overlap.
Use --suppress-bootloader-mem to ignore
make: *** [dfu] Error 4```

After this, I tried flashing an earlier usb_usb.hex (79KB versus the new 82KB one), however:

```
dfu.c:844: -EPIPE: a) Babble detect or b) Endpoint stalled 0xffffffe0 (-32)
Device is write protected.
dfu.c:252: dfu_clear_status( 0x7ffeed932a10 )
atmel.c:1434: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:360: Error writing memory data. (err -4)
```

Any advice on how to proceed?

tmk, 2018-03-02:
Probably your firmware is larger than available flash memory(28KB). You will have to slim down your firmware.

kaansoral, 2018-03-02:
Thanks for the reply, probably

Back in the day, I remember a method which gave the % of usage for the firmware, I remember being at 95% - Do you know how I can learn that info?

Most importantly, the USB-USB Converter doesn't light up any more, I tried to manually flash the older usb_usb.hex that I was using since 2016 with `dfu-programmer atmega32u4 flash usb_usb.hex --debug 99999` - however, received the second error

```
dfu.c:844: -EPIPE: a) Babble detect or b) Endpoint stalled 0xffffffe0 (-32)
Device is write protected.
dfu.c:252: dfu_clear_status( 0x7ffeed932a10 )
atmel.c:1434: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:360: Error writing memory data. (err -4)
```

So basically the converter is practically bricked until I solve this issue - Any ideas? / Have you ever experienced something similar?

kaansoral, 2018-03-02:
Stumbled onto: https://github.com/tmk/tmk_keyboard/issues/316

`dfu-programmer atmega32u4 erase --force` Solves the major issue

I can't get my keymap to work with the latest version of tmk tho (mine is from 2 years ago) - I even deleted one layer, no luck, would be great to have a tutorial to cut off some churn - I wonder if there are features I don't use that take up space

For the usb-usb converter, the default keymap is also huge, for an 60%, I don't need a complete remapping, but I guess there's nothing that can be done about it easily

closed 2018-03-02 by kaansoral

tmk, 2018-03-03:
Great.
`make dfu` should use `erase --force` by default, I don't why this doesn't work for you. There may be something room to improve.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/rules.mk#L430-L436

And this wiki page instructs to use `--force` option when flashing with dfu-programmer.
https://github.com/tmk/tmk_keyboard/wiki#dfu-programmer-for-windows-mac-and-linux