tmk/tmk_keyboard #546: Help needed with porting (Linker error)

nuess0r, 2018-04-07:
Hi

Thanks a lot for your great project and your effort to continuously improve and enhance it.

in my fork I'm porting tmk to an older USB controller for the IBM Model M based on a ATmega32 using the VUSB driver. https://github.com/nuess0r/tmk_keyboard

I got it to compile but I'm stuck with two linker errors I couldn't figure out myself what is missing/wrong.

```
Linking: dulcimer.elf
avr-gcc -mmcu=atmega32 -gdwarf-2 -DF_CPU=12000000UL -DUCSR -DNO_CLKPR -DTIMER0_COMP -DTIMER_PRESCALER=1024 -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=4096 -DBOOTMAGIC_ENABLE -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DCOMMAND_ENABLE -DVERSION=637c332b -DPROTOCOL_VUSB -DNO_UART -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=dulcimer.elf -I. -I../../tmk_core -I../../tmk_core/protocol -I../../tmk_core/common -I../../tmk_core/protocol/vusb -I../../tmk_core/protocol/vusb/usbdrv -std=gnu99 -include config.h -MMD -MP -MF .dep/dulcimer.elf.d  obj_dulcimer/keymap.o obj_dulcimer/matrix.o obj_dulcimer/led.o obj_dulcimer/common/host.o obj_dulcimer/common/keyboard.o obj_dulcimer/common/matrix.o obj_dulcimer/common/action.o obj_dulcimer/common/action_tapping.o obj_dulcimer/common/action_macro.o obj_dulcimer/common/action_layer.o obj_dulcimer/common/action_util.o obj_dulcimer/common/print.o obj_dulcimer/common/debug.o obj_dulcimer/common/util.o obj_dulcimer/common/hook.o obj_dulcimer/common/avr/suspend.o obj_dulcimer/common/avr/xprintf.o obj_dulcimer/common/avr/timer.o obj_dulcimer/common/avr/bootloader.o obj_dulcimer/common/keymap.o obj_dulcimer/common/bootmagic.o obj_dulcimer/common/avr/eeconfig.o obj_dulcimer/common/mousekey.o obj_dulcimer/common/command.o obj_dulcimer/protocol/vusb/main.o obj_dulcimer/protocol/vusb/vusb.o obj_dulcimer/protocol/vusb/usbdrv/usbdrv.o obj_dulcimer/protocol/vusb/usbdrv/usbdrvasm.o obj_dulcimer/protocol/vusb/usbdrv/oddebug.o obj_dulcimer/common/sendchar_null.o --output dulcimer.elf -Wl,-Map=dulcimer.map,--cref -Wl,--gc-sections     -lm 
obj_dulcimer/common/command.o: In function `command_common':
/home/chrigi/Projects/Model M/tmk_keyboard/keyboard/dulcimer/../../tmk_core/common/command.c:337: undefined reference to `keyboard_protocol'
/home/chrigi/Projects/Model M/tmk_keyboard/keyboard/dulcimer/../../tmk_core/common/command.c:338: undefined reference to `keyboard_idle'
collect2: error: ld returned 1 exit status
make: *** [../../tmk_core/rules.mk:552: dulcimer.elf] Error 1
```

Thanks for any guidance.

tmk, 2018-04-07:
VUSB doesn't seem to implement those parameters for some reason. It should be fixed  but you will be able to build with removing these lines at this time.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/command.c#L337-L338

nuess0r, 2018-04-08:
How does it work then for the onekey keyboard (attiny85)?

Did I enable a feature that makes it different?