tmk/tmk_keyboard #550: How to get dual-role keys to work using ACTION_MODS_TAP_KEY?

nicluo, 2018-04-11:
I've been trying to get dual-role keys to work on my HHKB Pro 2 with BT Controller.

I'm sorry if I've missed some trivial configuration but I've been reading up and trying to resolve this on my own for two evenings now. Any pointers will be very helpful to me as well.

Steps to reproduce:

1. Reset to master branch (everything is default configuration)
2. Modify `keymap_hhkb.c` to include `ACTION_MODS_TAP_KEY(MOD_LCTL, KC_ESC),`
3. Run `make -f Makefile.rn42`
4. Run `make -f Makefile.rn42 dfu`

Expected: When I tap left Control it should register Escape.

Issue: When I tap left Control nothing happens, it works normally when on hold as Control.

I know that the new firmware is flashed because if I change the keymap new keys get registered (e.g. 'a' to 'z', then when I press the key labeled 'a' on the keyboard it shows z on my text editor instead). I am using Mac OS X on a MacBook.

Full `keycap_hhkb.c`

```
/*
 * HHKB Layout
 */
#include "keymap_common.h"

#ifdef KEYMAP_SECTION_ENABLE
const uint8_t keymaps[][MATRIX_ROWS][MATRIX_COLS] __attribute__ ((section (".keymap.keymaps"))) = {
#else
const uint8_t keymaps[][MATRIX_ROWS][MATRIX_COLS] PROGMEM = {
#endif
    /* Layer 0: Default Layer
     * ,-----------------------------------------------------------.
     * |Esc|  1|  2|  3|  4|  5|  6|  7|  8|  9|  0|  -|  =|  \|  `|
     * |-----------------------------------------------------------|
     * |Tab  |  Q|  W|  E|  R|  T|  Y|  U|  I|  O|  P|  [|  ]|Backs|
     * |-----------------------------------------------------------|
     * |Contro|  A|  S|  D|  F|  G|  H|  J|  K|  L|  ;|  '|Enter   |
     * |-----------------------------------------------------------|
     * |Shift   |  Z|  X|  C|  V|  B|  N|  M|  ,|  .|  /|Shift |Fn0|
     * `-----------------------------------------------------------'
     *       |Alt|Gui  |         Space         |Gui  |Alt|
     *       `-------------------------------------------'
     */
    KEYMAP(ESC, 1,   2,   3,   4,   5,   6,   7,   8,   9,   0,   MINS,EQL, BSLS,GRV,   \
           TAB, Q,   W,   E,   R,   T,   Y,   U,   I,   O,   P,   LBRC,RBRC,BSPC,       \
           LCTL,A,   S,   D,   F,   G,   H,   J,   K,   L,   SCLN,QUOT,ENT,             \
           LSFT,Z,   X,   C,   V,   B,   N,   M,   COMM,DOT, SLSH,RSFT,FN0,             \
                LALT,LGUI,          SPC,                RGUI,RALT),

    /* Layer 1: HHKB mode (HHKB Fn)
     * ,-----------------------------------------------------------.
     * |Pwr| F1| F2| F3| F4| F5| F6| F7| F8| F9|F10|F11|F12|Ins|Del|
     * |-----------------------------------------------------------|
     * |Caps |   |   |   |   |   |   |   |Psc|Slk|Pus|Up |   |Backs|
     * |-----------------------------------------------------------|
     * |      |VoD|VoU|Mut|   |   |  *|  /|Hom|PgU|Lef|Rig|Enter   |
     * |-----------------------------------------------------------|
     * |        |   |   |   |   |   |  +|  -|End|PgD|Dow|      |   |
     * `-----------------------------------------------------------'
     *       |   |     |                       |     |   |
     *       `-------------------------------------------'
     */
    KEYMAP(PWR, F1,  F2,  F3,  F4,  F5,  F6,  F7,  F8,  F9,  F10, F11, F12, INS, DEL,   \
           CAPS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,PSCR,SLCK,PAUS, UP, TRNS, BSPC,      \
           TRNS,VOLD,VOLU,MUTE,TRNS,TRNS,PAST,PSLS,HOME,PGUP,LEFT,RGHT,PENT,            \
           TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,PPLS,PMNS,END, PGDN,DOWN,TRNS,TRNS,            \
                TRNS,TRNS,          TRNS,               TRNS,TRNS),
};

/*
 * Fn action definition
 */
#ifdef KEYMAP_SECTION_ENABLE
const action_t fn_actions[] __attribute__ ((section (".keymap.fn_actions"))) = {
#else
const action_t fn_actions[] PROGMEM = {
#endif
    [0]  = ACTION_LAYER_MOMENTARY(1),
    [1]  = ACTION_MODS_TAP_KEY(MOD_LCTL, KC_ESC),
};

```

tmk, 2018-04-11:
To use special function defined in `fn_actions[]` array you have to refer to the function with `FNx` in your `KEYMAP()` where x is number 0-31 which indicates index in the array.

In your keymap file the tap key function is defined with index '1' and you have to use `FN1` instead of `LCTL` in `KEYMAP()`.

tmk, 2018-04-11:
You can define tap key in Keymap Editor too.  You can see definition of Left Control key in 'Code Edit' tab.

https://goo.gl/C6fdpK