tmk/tmk_keyboard #551: can i compile led function only?

243750496, 2018-04-11:
because i have designed the keyboard with 3 controller , the reason is simple is that usb only can provide a limited current so that i need for 3 to afford full led and the keys,
it's used librepcb (like Kicad)to produce 

Q:as said previous i have 3 controller : 
1 for keys and core leds 
2 for Other Leds

so , if i can not get single led fuction compiled , i will not be able to use the other leds, so i hope to get a way to slove the problem 
[LibrePCB-Local.zip](https://github.com/tmk/tmk_keyboard/files/1898665/LibrePCB-Local.zip)

tmk, 2018-04-11:
Assuming 'core leds' means LED idicators like capslock,
I don't think this firmware includes code for leds other than lock key indicators.

You will have to write code for the 'Other Leds' and probably code to comuunicate between microcntrollers.

243750496, 2018-04-12:
what about backlight.c and backlight.h? how are these files use? is it contain the function i want? why it's in tmk_core (can i compile it single?)

243750496, 2018-04-12:
> You will have to write code for the 'Other Leds' and probably code to comuunicate between microcntrollers.

but if Teensy 2.0++ still contain some led function both with keyboard function, but how can i use tmk keyboard firmware both with own led firmware?(how to compile together to one hex file)
if can't is it means the only way i can do is to write a keyboard from zero ?

tmk, 2018-04-12:
oh, you are right. It seems to has backlight supoort somehow.
But I can't remember it  or don't know how it works. Read code and search repo, you will know how to use.

closed 2018-07-06 by tmk