tmk/tmk_keyboard #552: Print Screen is not mapped to PrtSc in XT

osteichthyes, 2018-04-23:
In the XT converter, the PrtSc with modifier to map to * (K7C). 

I propose that without NumLock or Shift, Print Screen is PrtSc, with NumLock, it is K7C

tmk, 2018-04-23:
Ah, ok. I want to know whether it works at the latest commit. Let me know the result when you update it with the latest firmware.

Thanks for heads up and eplanation of XT Numlock.
Through the converter the XT keyboard works like a normal USB keyboard. My converter doesn't intend to emulate the original PrintScreen behaviour. If you want to the original behaviour you have to write keyamp code.

osteichthyes, 2018-04-23:
I built the new one earlier this week and will flash it tonight.

I've never seen a keyboard where PrintScreen and asterisk share a key, I
just assumed the keymap wasn't setup to accommodate the layout of the
original IBM XT.

osteichthyes, 2018-04-23:
I flashed the newest version of the firmware and the issue persists, which makes sense, I don't see anything in the keycodes/keymap which identifies K7C as anything other than asterisk.

The new firmware does appear to be a tad buggy, with some stray keys registering and such. I was a bit confused about why the decision made to switch from functions to macros. Currently, the new firmware is not useable.

tmk, 2018-04-23:
Hmm, bummer. I believe this is righ way to go even though it still needs debug.
It would be appriciated if you can help to debug it. Or you can stick with old code if you don't have time.

I found keymap code have to be updated and I'm working on this now and will push later.

osteichthyes, 2018-04-23:
I'm happy to help debug. My C is middling at best, but I can usually do
okay debugging.

I actually have a few xts, so I can keep one functional on the old code and
another experimental with the new.

osteichthyes, 2018-04-24:
Looks like you pushed a new keymap, I'll  make and flash tonight.

tmk, 2018-04-25:
Updated keymaps and it supports E0-escpaed keys including PrintScreen now. I think PrintScreen is always PrintScreen and '*'(Keypad *) should not be registered now. Test `Alt + PrintScreen` on your keyboard in particular, this is  one of special case in XT keyboard. Let me know your result.

Yes, if you are lucky and sure what you are doing you can ommit pullups. But it is always safe to have them. Though, I do without them for temporal use.

tmk, 2018-04-25:
> I flashed the newest version of the firmware and the issue persists, which makes sense, I don't see anything in the keycodes/keymap which identifies K7C as anything other than asterisk.
>
>The new firmware does appear to be a tad buggy, with some stray keys registering and such. I was a bit confused about why the decision made to switch from functions to macros. Currently, the new firmware is not usable.
>
>I just switched back to the original version of the firmware and the bugs are now gone.

Which commit are you refering to 'original version' and Do you think acbea7f breaks something for your keyboard?

Try prebuilt binary also. My wild guess is GCC version and build tools may affect here.
https://github.com/tmk/tmk_keyboard/tree/master/converter/xt_usb/binary

On my side,
5cbc226 seems to break signal handling and it doesn't work for me until 8a92f25 fixes the issue or revert to original code. After 8a92f25 it works well for me.

In fact I don't have any real XT keyboard and test the code with AT2XT and PS/2 keyboard. This makes debug difficult in nit-picking cases.
https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-XT-Keyboard-Protocol#at2xt-keyboard-converter

Followings are list of important commits related to XT converter for reference.

* 5bd8a67 xt_usb: Update README(the latest)
* 1cb7a24 xt_usb: Clean keymap and change scancode matrix
* acbea7f xt_usb: Replace functions  with macros
* 2b9d4ac xt_usb: Clear flag before enabling interrupt
* 8a92f25 xt_usb: read data on falling edge soft/hard reset
* 5cbc226 Update config.h(this breaks logic?)
* a9bce90 Mostly working. Is unstable, will emit bad codes after a while.
* 3def1c3 Codes appear to be detected correctly, the break codes are broken.

osteichthyes, 2018-04-26:
Sorry, I should have specified commit number. I built after acbea7f, that was unusable, it just kept either not missing keys or registering random keys. I haven't tried any of the newer commits yet. I will this evening.

osteichthyes, 2018-04-26:
Ok, I pulled the latest. The numpad is working intermittently. The printscreen/asterisk is not recognized at all. And, every once and a while, "c" or "z" begin printing nonstop. This is what happened with my build of acbea7f, but which did not happen on 2b9d4ac

The c and z printing nonstop seems to be related to the change from functions to macros.

tmk, 2018-04-26:
Did you build firmware without any modification?

And can you try my prebuilt binary?

tmk, 2018-04-26:
The commit acbea7f makes signal read timing a little fast, a few micro second, this may casue your problem.

Try pull-up resistors if possible. 

Or this patch may cure, change the value like 1 or 5, 10 ... .

```
diff --git a/tmk_core/protocol/xt_interrupt.c b/tmk_core/protocol/xt_interrupt.c
index a0c55e9..25fe04e 100644
--- a/tmk_core/protocol/xt_interrupt.c
+++ b/tmk_core/protocol/xt_interrupt.c
@@ -94,6 +94,7 @@ ISR(XT_INT_VECT)
     } state = START;
     static uint8_t data = 0;
 
+    _delay_us(1);
     uint8_t dbit = XT_DATA_READ();
 
     // This is needed if using PCINT which can be called on both falling and rising edge
```

osteichthyes, 2018-04-26:
Unfortunately, it'd be really, really hard to add resistors to the build.
Is it possible to use  the internal pull up resistors? Where is the pin
defined? Is that in the core library?

tmk, 2018-04-26:
> In your binary, it seems to be working more reliably, but the PrintScrn is still not working properly, it's only being recognized as an asterisk, regardless of whether or not num lock is on.

Ah, ok. It sounds promising.

**Then, GCC version difference is likely to be a problem**, pull-ups and the patch won't help. I'll have to look into recent changes of GCC but I don't think I can do that so soon. Use older GCC for a while or git-revert the commit in your repo until this issue is fixed. And internal pull-ups in AVR is already enabled and that is exactly why this works often without external real resistors.

Now we can return to  PrintScreen again. 
According to Microsoft document, regardless of NumLock state PrintScreen should spit out 'E0 2A E0 37' when pressed and 'E0 B7 E0 AA' when released, while Numpad* sends just '37' and 'B7'.

See this document p17 and p21 for detail if you are interested.
https://download.microsoft.com/download/1/6/1/161ba512-40e2-4cc9-843a-923143f3456c/scancode.doc

Can you check raw scan codes of the key with `hid_listen` tool?
https://github.com/tmk/tmk_keyboard/wiki#pjrc-hid_listen

I like to see scan codes of PrtSc key and Alt+PrtSc.
You will get output like below on terminal.
```
E0 2A E0 37 E0 B7 E0 AA
```

osteichthyes, 2018-04-26:
Regarding GCC, my guess is that the issue is in preprocessing as the only
change is macro is preprocessed.

It makes sense that the IBM PC XT keyboard is not working appropriately as
the layout of the IBM XT is odd. Is there a way to make PrintScreen spit
out 'E0 2A E0 37' and 'E0 B7 E0 AA' when numlock is deactivated and '37'
and 'B7' when numlock is active?
This would replicate the intended functionality of the keyboard. I'm happy
to make the mods and push them as an original XT keymap, I'm just not sure
I understand how the numlock works. I assumed it activated a layer that was
transparent for everything but the numpad. That does not appear to be the
case, it seems like somehow the keycode is set with a pre-set response to
numlock, numlock is not treated as a standard modifier. Is that correct?
Would I need to redefine the scancodes printed by the keycode? Where do I
define the scancodes registered by a keycode under numlock and without
numlock?
Could I just create a new keycode which is numlock dependent and does what
I'm hoping? Then, in my keymap, I could just use that different keycode,
right?

I'll check the raw scancodes this evening.

tmk, 2018-04-27:
> It makes sense that the IBM PC XT keyboard is not working appropriately as
the layout of the IBM XT is odd. Is there a way to make PrintScreen spit
out 'E0 2A E0 37' and 'E0 B7 E0 AA' when numlock is deactivated and '37'
and 'B7' when numlock is active?

No, the scancodes are hardcoded in the XT keyboard firmware and not configurable.

>This would replicate the intended functionality of the keyboard. I'm happy
to make the mods and push them as an original XT keymap, I'm just not sure
I understand how the numlock works. I assumed it activated a layer that was
transparent for everything but the numpad. That does not appear to be the
case, it seems like somehow the keycode is set with a pre-set response to
numlock, numlock is not treated as a standard modifier. Is that correct?
Would I need to redefine the scancodes printed by the keycode? Where do I
define the scancodes registered by a keycode under numlock and without
numlock?

Even in modern computer with USB ports NumLock state is managed in host side, not in keyboard(and converter). And number(0-9) and dot(.) keys on numpad are changed its behaviour by host depending on NumLock state as you know. Note that asterisk(*) is not included there.

Do you really know original XT computer behaviour on the 'PrtSc' key? Maybe you just guess it?
If I'm wrong I'll stand corrected. If you have original XT host hardware you can confirm this.
After looking at your report and scancode table I was almost sure the 'PrtSc/*' key works as PrintScreen when Shift otherwise as Numpad* on original XT computer. Look its double legends on the key, it indicates its shifted behaviour as other number and symbol keys have. Finally I found this descritption at  p5-21 of TechRef.

>Print Screen
>The combination of the Shift and PrtSc (key 55) keys will result
>in an interrupt invoking the print screen routine. This routine
>works in the alphanumeric or graphics mode, with unrecognizable
>characters printing as blanks. 

http://www.minuszerodegrees.net/manuals/IBM_5150_Technical_Reference_6322507_APR84.pdf

You can define the 'PrtSc/*' key in the converter as you described if you still want. You will have to write small C code to check NumLock state but not so big job, I think.

>I'll check the raw scancodes this evening.

Thank you, that would be very appreciated and still useful.

osteichthyes, 2018-04-30:
Sorry, I was traveling. The scancodes are
0x37 0xB7

Sometimes, these are reported on the same line, other times, they are on separate lines. As I'm understanding it, the fact that my system is not recognizing that keycode as print screen is a host, and not a client issue. As I'm understanding it, the PCXT simply recognized the key differently, the key is registering as it should as there is never any real change to the key. The recognition of [Shift] + [*] as print was entirely host based. Am I understanding correctly?

tmk, 2018-05-01:
Thanks. It is exactly what I expected, the key is Numpad*. 37 is make code and B7 is break one for that key, btw. Confirmed that the converter can handle the key 'PrtSc/*' correctly as the result.

Your mordern computer should recognized the key as '*'(Numpad*), not PrintScreen. Yes, not inside .keyboard, it is original PCXT host where the recognition 'Shift+the key=PrintScreen' happens.

When I make converter for old fashioned keyboards I don't care much about its key legend or original behaviour. I usually don't try to emulate origin behaviour in default keymap but users will be able to define it in their own keymap like they want in many cases, perhaps.

If you

tmk, 2018-05-05:
> I'll try newer GCC if I have time.

I setup Archlinux with AVR toolchain packages(avr-gcc 7.3.0-1/avr-binutils 2.30-1/avr-libc 2.0.0-1) in Docker container to built firmware and test it. The firmware still works well with my testing setup. Hmm, still not sure why the latest firmware you built on Arch doesn't work... Let me know if you get any idea on this problem.

The answer to original question is:
The 'PrtSc/*' key should be recognized as 'Numpad *' in modern computer and the current firmware works correctly as such.

Will close this issue now.

closed 2018-05-05 by tmk

osteichthyes, 2018-05-14:
I'm sorry for the late reply, I have been traveling nonstop. I also made
some mods to my XT and it's currently in pieces. I'll finally have some
free time next week to reassemble the XT and will try your new binaries as
well as building my own again to see what happens.

Thanks for being patient.

tmk, 2018-10-07:
The firmware has had wrong mapping for the key on keymap and unimap files.
https://geekhack.org/index.php?topic=94649.msg2666677#msg2666677

With this fix 594d7d85d the '*/PrtSc' key is recognized as 'Keypad *' by default and you can also remanp the key freely now.