tmk/tmk_keyboard #553: Canot flash success?

lisper-inmove, 2018-04-23:
I go to gh60 directory change nothing.

make -f Makefile
dfu-programmer atmega32u4 erase --force
dfu-programmer atmega32u4 flash gh60_lufa.hex
dfu-programmer atmega32u4 reset

but i press 5 it prints 8, t prints i, g prints k, v prints m, and space button without nothing. My pcb is satan, anybody know this error? please help me, thx.

kaihuang201, 2018-04-23:
I ran into the exact same issue about six months ago when I built my first mech keyboard with satan PCB. It turned out that Satan PCB has a different layout from the GH60's so the keymaps are not compatible. I managed to get it working flawlessly by writing my own keymap, which can be found in  my [fork](https://github.com/kaihuang201/tmk_keyboard/tree/master/keyboard/satan) for your reference. I customized my keymap a little bit as you can see in the README, but you can edit it to suit your own needs. Hope this helps. Let me know how it works out!

tmk, 2018-04-23:
What is your pcb; bootleg or clone one?
This supports genuine GH60 Revision A to C at least.

closed 2018-04-23 by tmk

kaihuang201, 2018-04-23:
@tmk Satan GH60 Rev. CHN. is a PCB that closely resembles GH60 but with added support for backlight. Satan, doesn't really conforms to GH60 specs in pin mapping but they call it "GH60" anyway. This is really a non-issue, however, I do suggest adding support for Satan GH60 Rev. CHN to the code base just because it is a fairly common board on the market. I already have a working copy in my [fork](https://github.com/kaihuang201/tmk_keyboard/tree/master/keyboard/satan) so migrating it over shouldn't be too hard.

tmk, 2018-04-24:
Its designer and sellers are friendly or open to the community? It doesn't seem they are, to me at least.
I don't know it is worth to support their products and pitiful users.

lisper-inmove, 2018-04-24:
@kaihuang201 thanks, i am going to try your fork.