tmk/tmk_keyboard #557: AC_BOOTLOADER(BTLD) doesn't work

tmk, 2018-05-17:
Action 'Jump to Bootlloader' doesn't work in Actionmap and Unimap while `KC_BOOTLOADER(BTLD)` still works in Keymap.

It is currently defined for Keymap here but the action should be defined in action.c specifically in `register_code()`.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/keymap.c#L157-L161

RaoulDuke-Esq, 2021-02-25:
Using TMK Keycap Editor:
BLTD works for my M122 (IBMPC-USB Converter(ATmega32U4) - 122-key Terminal)
BLTD does not work for my Pingmaster (IBM4704-USB Converter rev.1 for Alps 102-key).  I have tried it in several positions on the board but it never works.

tmk, 2021-11-12:
`AC_BOOTLOADER` and `AC_BTLD` works now.
- 83b90f4b6fe67

closed 2021-11-12 by tmk