tmk/tmk_keyboard #559: Is it possible to use arduino pro mini with RN-42 for USB<->bluetooth conversion?

LubergAlexander, 2018-05-26:
Hi,

I have a project I am working on and don't want to reinvent the bicycle.

I am using a Leopold FC750R and would like to make it wireless.

Current hardware setup is Arduino pro mini (3.3v) + USB Host Shield mini v2(Circuits@Home) + Sparkfun Bluetooth Mate Silver (RN-42) + of course a battery charger and a battery.

I noticed that USB host shield is supported, but haven't seen any examples which use both bluetooth and usb host.

tmk, 2018-05-26:
I think it is possbile. Note that RN-42 firmware version should be 6.15 or later to support HID.

No documentation/example for your setup, you will have to use time to read source code, though.

LubergAlexander, 2018-05-26:
@tmk Thanks! I will try that. It is indeed 6.15.
If you can provide any pointers to docs or any wiki - would be really helpful!

Mostly I am wondering if any special pin wiring is required? (my Arduino Pro Mini sits right on top of USB host mini (I think just like teensy)) and RN-42 module I use is connected from FTDI side, so hardware serial. My best guess is - no, but maybe some extra wires are required to some reset pins and whatnot.

And another question is RN-42 configuration. Does this firmware handle that automatically on the first start, or I have to preconfigure it manually?

tmk, 2018-05-26:
I think you can use/write software serial code on pro mini to handle RN42.

I used USB host shield on usb_usb converter and RN42 on HHKB Alt controller, you can refer code and document of these.
https://github.com/tmk/tmk_keyboard/tree/master/converter/usb_usb
https://github.com/tmk/tmk_keyboard/tree/master/keyboard/hhkb

KiCad schematic of the projects are available here.
https://github.com/tmk/HHKB_controller
https://github.com/tmk/USB2USB_Converter

HHKB Alt controller can initialize RN42 configuration from its firmware but it is triggered by key combo(both shifts + ScrollLock). But I think you can write code for your firmware to do automatic configuration, it depends on you.
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/hhkb/rn42/rn42_task.c#L169-L183

LubergAlexander, 2018-05-26:
Thanks! I actually already did, and the problem I am facing at the moment, is that there are random, significant lags.  It works end to end, but when I hold a key for some time, it will type a sequence of characters(5-10long) and then gets stuck, then it recovers, and if I type too fast it misses key presses. I have tried to do SQ,16 on the chip, to optimize latency, but it doesn't seem to help at all. Have you encountered anything like that? I am wondering if it is a hardware or software issue, that's why I started exploring alternative firmwares to try to isolate the problem...

tmk, 2018-05-26:
Oh, great! It looks close to finish :smile:
I guess the lags are caused on USB host side, I remember it has some lags with mcu running at 8MHz. My usb-usb converter does overclock 16MHz on 3.3V, instead of 8MHz. I think it should run at 8MHz with fixing/optimising my code but I din't look into. It would be appriciated if you can fix the lags.

tmk, 2018-05-26:
hmm, I just tried the converter at 8MHz and it works without any lag. Ignore my previous message.
It seems that USB host side should run at 8MHz now.

LubergAlexander, 2018-05-26:
I have tried to just print to serial and saw no lag at all(Arduino and usb
host only). It is definitely RN-42, my guess is I have configured it
incorrectly. Any recommendations? All the internets say S~,6 is enough, but
it doesn't seem to be true:)

tmk, 2018-05-26:
I can't remember RN42 commands much anymore. You will have to refer its datasheet.
All of my configuration is what you see in the code.

closed 2018-05-27 by LubergAlexander