tmk/tmk_keyboard #562: Apple Extend Keyboard II on macOS: sporadic transpose of a and h

jonhendry, 2018-07-11:
Occasionally when I type a word that contains the pair "ah" the a and h are being transposed. They appear correctly then swap places. There doesn't appear to be anything in Keyboard preferences to account for this (no automatic spelling correction or text replacements set up.)

I haven't noticed this behavior with any other characters. 

Other than this, works great.

jonhendry, 2018-07-12:
Sadly I don't know. I've tried reproducing it intentionally but haven't been able to. About once a day I notice the letters swapping.

I don't *think* I'm imagining it. I was kind of hoping someone else had seen it.

jonhendry, 2018-07-12:
Ah, I figured it out exactly.

Some of my AEK II’s keys are very sensitive and trigger with only the slightest pressure.

So I end up doing ctrl-t (transpose) sometimes. 

> 
> How can you reproduce the issue? which word exactly?

Thanks for your time! Sorry for the false alarm.

tmk, 2018-07-12:
Nice to hear your catched the cuase in the end!