tmk/tmk_keyboard #566: matrix_print(void) in matrix.c for TMK/Chibios

DmNosachev, 2018-08-24:
Hi,
I'm trying to figure out how to use the debug functions of TMK to troubleshoot my keymap. I've built a replacement controller for Model M 122 using "Blue pill" board (based on STM32F103C8). Keys are organized in 20x8 matrix. I've traced the membrane visually and created the corresponding keymap macro. Everything works fine except one single key, there is some error in mapping. Unfortunately I've deleted the photos of the membrane.
Is there any way to make TMK show row and column of a pressed key in console? I suppose that matrix_print function should do it, but it only outputs row/colums table when I press a key:
```
r/c 0123456789ABCDEFGHIJ
00:
01:
02:
03:
04:
05:
06:
07:
```

tmk, 2018-08-24:
Yes, you can use `matrix_print()` for that purpose but you have to implement it properly.

Assuming this is source code you are refering. https://github.com/DmNosachev/m122_tmk

This line in `matrix_print` is iffy, I think.
https://github.com/DmNosachev/m122_tmk/blob/f8962a92b18d77929edd888a858b4ec0b8222a80/matrix.c#L129

```
pbin_reverse16(matrix_get_row(row));
```
Here `matrix_get_row()` returns `matrix_row_t` and when your matrix column size is 20 `matrix_row_t` is actually defined as `uint32_t`. Check matrix.h.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/matrix.h#L25-L33

While `pbin_reverse16()` expects `uint16_t` as its argument you are giving it `uint32_t` in fact.  This is probably your problem.

`pbin_reverse16()` is alias of `print_bin_reverse16()` and 32bit version of the function is `print_bin_reverse32()`, you should use this instead of it.
https://github.com/tmk/tmk_keyboard/blob/cf6df9db5fa6123b418ca29455d467e4455c0ebd/tmk_core/common/print.h#L143 
https://github.com/tmk/tmk_keyboard/blob/cf6df9db5fa6123b418ca29455d467e4455c0ebd/tmk_core/common/print.h#L86

tmk, 2018-08-24:
Oh, I just found taht tmk_core/common/matrix.c in  has default implementation.  Just remove your own `matrix_print()` from your matrix.c, the default function will be used instead.

https://github.com/tmk/tmk_keyboard/blob/cf6df9db5fa6123b418ca29455d467e4455c0ebd/tmk_core/common/matrix.c#L47-L74

DmNosachev, 2018-08-29:
Thanks for your help, Hasu.
Unfortunately, I have only basic knowledge of C.

I have updated the repo, so it reflects my current setup [https://github.com/DmNosachev/m122_tmk](https://github.com/DmNosachev/m122_tmk).

Removing matrix_print function from matrix.c prevents TMK from compiling:

> /home/quartz/build/tmk_keyboard/keyboard/m122_tmk/../../tmk_core/common/keyboard.c:128: undefined reference to `matrix_print'

Adding the code of matrix_print from tmk_keyboard/tmk_core/common/matrix.c to local matrix.c with print_bin_reverse32 or it's unwrapped equvalent doesn't help. With each key press I can see the colums/rows output and some portion of weird characters.

Maybe it's possible to make a function which outputs the value (or number) of keymap array when user presses a key?

DmNosachev, 2018-09-05:
I'managed to find an error in my physical keymap definition (keymap_common.h), so I don't need matrix debugging anymore.

closed 2018-09-05 by DmNosachev