tmk/tmk_keyboard #567: New at this and have two questions

diskmikey, 2018-08-27:
Please excuse the ignorance but am totally new at programming but am trying to use this code (gh60) to get an older keyboard to work on a newer machine. There are two issues/questions I have...

1) After changing the keyboard matrix and the other files as the project describes, when I try to compile, I get pages of error (like 42 pages) Most of them are cannot find file or directory.....I put the code in a folder on my desktop and am running it from there. I am running it through the Linux (Ubuntu 18.04.1) terminal. Is the directory suppose to be in a specific path?

and that brings me to my second question...

2) I will be hooking this up to a windows machine (as soon as I can get it to compile) do I need to compile it in windows or will the code work on whatever I hook it to after it is built?

LouWii, 2018-09-09:
To answer your second question, you can build the firmware on Linux or Windows, it doesn't matter, the resulting firmware will be the same.

diskmikey, 2018-09-10:
Thank you so much for the info LouWii

closed 2019-05-29 by tmk