tmk/tmk_keyboard #568: How can I make an exclamation point appear when I type a letter "j" or a semicolon?

jackiefoyer, 2018-09-01:
The exclamation point is normally typed using Shift + 1, but I would like to make it appear when I type both "j" or a semicolon. Can this be done?

There doesn't seem to be a way to put this into the keymap, I'm guessing because exclamation point is usually inserted with the user holding the shift key.

tmk, 2018-09-01:
>  when I type both "j" or a semicolon.
what do you mean by this? you want to press both 'j' and semicolon at same time to get '!' ?

You may want to read this.
https://github.com/tmk/tmk_core/blob/master/doc/keymap.md#212-modified-key

jackiefoyer, 2018-09-01:
Maybe this will demonstrate my issue.

If I make a change to use an ACTION_LAYER_TOGGLE in my keymap, I get a compilation error.
https://github.com/tmk/tmk_keyboard/compare/master...jackiefoyer:patch-2

```
:usb_usb jfoyer$ make dfu

mkdir -p obj_usb_usb
Compiling C: keymap.c
avr-gcc -c -mmcu=atmega32u4 -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=4096 -DARDUINO=101 -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(USB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DVERSION=428d9045-dirty -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_usb_usb/keymap.lst -I../../tmk_core/common -I. -I../../tmk_core -I../../tmk_core/protocol/usb_hid -I../../tmk_core/protocol/usb_hid/USB_Host_Shield_2.0 -I../../tmk_core/protocol/usb_hid/arduino-1.0.1/cores/arduino -I../../tmk_core/protocol/usb_hid/arduino-1.0.1/variants/leonardo -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lufa/LUFA-git -I../../tmk_core/common -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_usb_usb_keymap.o.d  keymap.c -o obj_usb_usb/keymap.o 
In file included from keymap.c:18:
keymap_common.h:86:37: error: 'KC_TGL1' undeclared here (not in a function); did you mean 'AC_TGL1'?
     { KC_##K30, KC_##K31, KC_##K32, KC_##K33, KC_##K34, KC_##K35, KC_##K36, KC_##K37,   /* 30-37 */ \
                                     ^~~
keymap.c:42:5: note: in expansion of macro 'KEYMAP_ALL'
     KEYMAP_ALL(
     ^~~~~~~~~~
make: *** [obj_usb_usb/keymap.o] Error 1
```
But this seems to follow the pattern used in keyboard/fc660c/unimap_emu.c, which I assume works.
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/fc660c/unimap_emu.c#L11

How do I get rid of this compilation error?

Then can I change out this line to be this line?

`define AC_TGL1     ACTION_LAYER_TOGGLE(1)`
`define AC_TGL1     ACTION_KEY(KP_EXCLAMATION)`

tmk, 2018-09-02:
It is notation for unimap(actionmap) while you are using 'keymap' file.
With keymap you have to define the action in `fn_acitons[]`.  or just use unimap.c instead of keymap.c.
https://github.com/jackiefoyer/tmk_keyboard/blob/f127d746e3690b792ed13b8b14b531ee2b8330cd/converter/usb_usb/keymap.c#L262-L265

And use `ACTION_MODS_KEY(MOD_LSFT, KC_1)` for exclamation mark as the documentation says.

Assuming you are using usb-usb converter, you can use keymap editor for this. You can define custom action using 'Code Edit' tab. Check J and ; key of this keymap.
https://goo.gl/D6ynDc

DreymaR, 2018-10-08:
Maybe have a look at my TMK files for inspiration? I have an action function to emulate AltGr for instance.

https://github.com/DreymaR/BigBagKbdTrixTMK

closed 2019-05-29 by tmk