tmk/tmk_keyboard #574: Sending raw keycode

lord-re, 2018-09-16:
I've read the faq and doc but i wasn't able to find **a way to send raw keycode to the OS**.

On Linux, i can **xmodmap -pk** to see each keycode and its associated keysyms but unfortunately i can't send raw keycode from tmk.

I'd love using XF86MonBrightnessDown and XF86MonBrightnessUp which are keycodes 232 and 233 (E8 and E9 in hex) but haven't find a way to do it.

tmk, 2018-09-16:
You will have to look into XOrg and kernel codes to find USB 
keycode(usage) you can use. 232 and 233 are keycodes used just on X, not 
on keyboard firmware.

lord-re, 2018-09-17:
Ho !
I thought the keyboard just sent keycodes from 0 to 255 and then the OS did the rest.

DreymaR, 2018-10-08:
Hmmm. The key codes are enumerated in https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/keycode.h and as you see, E7 (RGUI) is the last one. So I'd think that if you add two lines below that in this file you should be able to provide enumerations for E8 and E9 as well? Or would that collide with something else?