tmk/tmk_keyboard #576: Error reading intel hex file with teensy_loader_cli

markspanbroek, 2018-09-25:
After compiling the xt_usb converter, I can no longer load the resulting .hex binary into my Teensy.

This is the error that I see:

```bash
$ teensy_loader_cli -mmcu=atmega32u4 -w xt_usb.hex
Warning, HEX parse error line 1231
error reading intel hex file "xt_usb.hex"
```

The last lines of the .hex file look like this:
```
:104C5C000000000000000000000000000000000048
:104C6C00000000000000007F00547C0000000000E9
:104C7C0000000000000000005574607700610063C4
:104C8C00007562787172000000000000005A5B5CD5
:104C9C005D5E0000005F404D3622391F03D2029C3E
:104CAC00027A024102B401CD01E601FF011F0200AC
:0A4CBC0001020304050607080900C1
:0200000400807A
:0403EC00000000000D
:00000001FF
``` 

(Line 1231 is the next to last line)

The pre-built binary in `binary/xt_atmega32u4.hex` does work correctly.

System: MacOS Mojave, Xcode 10

tmk, 2018-09-25:
what is your toolchain actually? or Xcode 10 support avr arch now?

Post your result of the converter on geekhack!
https://geekhack.org/index.php?topic=94649.0

markspanbroek, 2018-09-26:
Sorry, you're right. Xcode 10 is not relevant :)

I'm using avr-gcc, installed using Homebrew:

```
$ brew info avr-gcc
osx-cross/avr/avr-gcc: stable 8.2.0, HEAD
GNU compiler collection for AVR 8-bit and 32-bit Microcontrollers
```

```
$ brew info avr-binutils
osx-cross/avr/avr-binutils: stable 2.31.1
GNU Binutils for the AVR target
```

tmk, 2018-09-26:
Those tools is way too new and ahead of ones I'm using. I guess they chaged someting on output of intel hex format recently, perhaps.

These lines of the hex file does't seem to be valid for the AVR at least. It is out of flash memory range. But I don't know why the lines were produced.
```
:0200000400807A
:0403EC00000000000D
````

drashna, 2018-10-30:
Yeah, 8.2 is broken. We've had issues with it on QMK as well. 

And there is already a fix, but 8.3 will have to be pushed.  Otherwise, 8.1 or 7.3 is what you want to use.

closed 2019-05-29 by tmk