tmk/tmk_keyboard #579: HyperX FPS Pro US layout to UK

yar05, 2018-10-28:
Hi, I'm looking for a hardware that would enable HyperX FPS Pro keyboard which only comes with US layout to remap some keys so its like an UK layout. 

Is this going to work with it ? 

[403982.250078] usb 3-1.4: new full-speed USB device number 43 using xhci_hcd
[403982.357438] usb 3-1.4: New USB device found, idVendor=0951, idProduct=16d2, bcdDevice=21.11
[403982.357445] usb 3-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[403982.357450] usb 3-1.4: Product: HyperX Alloy FPS Pro Mechanical Gaming Keyboard
[403982.357454] usb 3-1.4: Manufacturer: Kingston

tmk, 2018-10-28:
If the keyobard works well with BIOS and GRUB the converter can handle it probably.

Also USB descriptor including HID report descriptor of the keyboard would be helpful to know whether the converter can work with it.
https://github.com/tmk/tmk_keyboard/wiki/HID-Report-Descriptor

yar05, 2018-11-03:
ok So I've just got the converter USB2USB. Now I want to change the layout of my keybord to the one from http://www.keyboard-layout-editor.com/#/ 

Is there any instructions on how to do this? I can only see the http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?usb_usb but its not as simple as the other one.

Below end result for keyboard-layout-editor raw data that I want to achieve on my keyboard.

`["Esc",{x:1},"F1","F2","F3","F4",{x:0.5},"F5","F6","F7","F8",{x:0.5},"F9","F10","F11","F12",{x:0.25},"PrtSc","Scroll Lock","Pause\nBreak"],
[{y:0.5},"~\n`","!\n1","\"\n2","£\n3","$\n4","%\n5","^\n6","&\n7","*\n8","(\n9",")\n0","_\n-","+\n=",{w:2},"Backspace",{x:0.25},"Insert","Home","PgUp"],
[{w:1.5},"Tab","Q","W","E","R","T","Y","U","I","O","P","{\n[","}\n]",{w:1.5},"|\n\\",{x:0.25},"Delete","End","PgDn"],
[{w:1.75},"Caps Lock","A","S","D","F","G","H","J","K","L",":\n;","@\n'",{w:2.25},"Enter"],
[{w:2.25},"Shift","Z","X","C","V","B","N","M","<\n,",">\n.","?\n/",{w:2.75},"Shift",{x:1.25},"↑"],
[{w:1.25},"Ctrl",{w:1.25},"Win",{w:1.25},"Alt",{a:7,w:6.25},"",{a:4,w:1.25},"Alt",{w:1.25},"Fn",{w:1.25},"Menu",{w:1.25},"Ctrl",{x:0.25},"←","↓","→"]`

tmk, 2018-11-03:
So USB-USB converter works with the "HyperX FPS Pro keyboard", right?

Check first post of  the convefter thread first.
https://geekhack.org/index.php?topic=69169.0

yar05, 2018-11-03:
Well the issue comes form the fact that HyperX FPS Pro is actually a 87 key keyboard.

So there is one physically missing key on it that would fit to the UK layout. 
HyperX FPS Pro :
[image]

HyperX FPS :
[image]

yar05, 2018-11-03:
So I know how to flash the firmware, and I know about the keymap.hex to be replaced with usb_usb.hex to flash it. This works for me.

Question would be how to use editor to let's say change behaviour of key "Action key "\" ? currently it has # and ~ under it for this 87 key keyboard. Can I somehow use that keyboard-editor to use with your software ? In there It's a little bit easier at least it seems like that

yar05, 2018-11-03:
Well I'm using the GB layout in the OS as sometimes I can't change it to the US. By the look of it all of the keys are working fine aside from the below :

[image]
Currently it has the # and ~ under shift.
Is there any way to just remap that to have its original functionality and move the ~ to its original location next to the key 1 as well as Hash symbol to be there with the shift function ?

yar05, 2018-11-03:
Ok perfect I think I know how to use it now :) big thanks it saved me buying another keyboard !

Just one more question. So I'm trying to activate magic command to get help which is 

/* key combination for command */
#define IS_COMMAND() (keyboard_report->mods == (MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT))) 

Which I understand that if I press Lshift + RShift + h it should output some help in text editor ? or am I missing something here ?

tmk, 2018-11-04:
Yes, you are right for the keybind, you need to `hid_listen` tool to get output. But default firmware(used in Keymap Editor) for USB-USB converter is built without magic command function unfortunately while almost all other my products have the function. USB-USB firmware size is rather large and cannot be built with full function unfortunately, you will have to remove MOUSEKEY or EXTRAKEY(media keys) function instead to get magic command enabled here.

https://github.com/tmk/tmk_keyboard/blob/master/converter/usb_usb/Makefile#L90-L97
```
# Build Options
#   comment out to disable the options.
#
MOUSEKEY_ENABLE ?= yes	# Mouse keys
EXTRAKEY_ENABLE ?= yes	# Media control and System control
CONSOLE_ENABLE ?= yes	# Console for debug
#COMMAND_ENABLE ?= yes    # Commands for debug and configuration
#NKRO_ENABLE ?= yes	# USB Nkey Rollover
```

yar05, 2018-11-04:
Ok no I'm a little bit confused. Is there a way to switch between layers without magic key ? 
I thought about using keymap editor to switch between layers and within one of the layers I would have the mousekeys enabled.