tmk/tmk_keyboard #590: what usb hid codes actually work?

GianniGi, 2019-01-21:
Hello, I'm using the USB_USB converter bought from you; I've tried almost all the usb keycodes described in the firmware, by assigning them to virtual layers. The testing method has proven quite unreliable, and I've found that only a small part of those keycodes actyally work. Is there a way to understand what codes actually work? 
Only a few of the KC_KP codes seem to work (windows 10 and linux vitualbox), and KC_PWR and similar seem to be inactive too.

thanks in advance

tmk, 2019-01-21:
You can send those keypad codes using `AC_KP_*` code. `AC_` is action code for newer actionmap and unimap while `KC_` is used for old keymap. I think you want to use `AC_` in your unimap.c for usb-usb converter.

Yes, use `ACTION_USAGE_CONSUMER()` for consumer page usages like volume and media control. Definitions in report.h have some limitations and are used for old keymap.

GianniGi, 2019-01-24:
Sorry for being a newbie, but how do I define "usb hid" action codes? I've tried to insert those codes in keymap.c but I can't figure out where they can be inserted.

tmk, 2019-01-27:
Define `AC_` name for your action like this line in your unimap.c before using it,
```
#define AC_xxxx ACTION_USAGE_CONSUMER(0x6F)
```
And then place `xxxx` in your `UNIMAP()`.

GianniGi, 2019-04-29:
Is it fast or does the output get easily chocked? I'm asking because I need to write with a sustained speed. Thanks

tmk, 2019-04-30:
Probably chocked. Consumer keys and system control keys are send one by one and you can't send two or more keys at once. Those keys are not expected to be used in typing in TMK.

GianniGi, 2019-05-27:
I need a layer that inputs some ascii characters without choking (it should sostain standard writing speed, because I need those symbols to write text and commands for a program). So far I've used a few of the available symbols, but there's a very small number of those.

	GRAVE 
	KP_ASTERISK 
	not working KP_CLEAR ←
	not working KP_CLEAR_ENTRY ¡ 
	works only in vim KP_EQUAL =
	KP_MINUS 
	KP_SLASH  
	NONUS_BSLASH 
	BSLASH 
	COMMA,
	EQUAL 
	KP_COMMA 
	KP_PLUS 
	LBRACKET è 
	MINUS 
	NONUS_HASH 
	QUOTE 
	RBRACKET 
	SCOLON 
	SLASH