tmk/tmk_keyboard #591: Command key doesn’t work well with the iPad

satran, 2019-02-12:
I’m using the firmware on a HHKB. I have mapped one of the keys to LeftGUI. iOS maps Cmd+Space and Cmd+Tab for showing the search and switching application respectively. With the LeftGUI I’m able to perform Cmd+L to focus the address bar of Safari. But the global keys don’t seem to function. When I use the stock firmare of HHKB it seems to work well. Is there something that I have overlooked?

tmk, 2019-02-13:
How did you remap the key, used Keymap Editor or built firmware yourself?
And "stock firmware" means one comes with TMK controller, right?

satran, 2019-02-13:
I used the keymap editor from here: 
http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?hhkb
As for the “stock firmware” I mean the firmware that came with the original board of HHKB. I bought the controller for HHKB as defined in this page: https://geekhack.org/index.php?topic=71517.0 and installed the tmk firmware.

tmk, 2019-02-13:
Ah, OK. I have no idea then. Let me know if you find any clue for this.

closed 2019-05-29 by tmk