tmk/tmk_keyboard #593: terminal to USB PS2 protocol int or usart？

youxiaojie, 2019-03-10:
# PS/2 protocol implementations
# 	USART is recommended if it is available, others are for reference purpose.
# 	INT implementation will drop simultaneous key strokes.
#
#PS2_USE_USART = yes	# uses hardware USART engine for PS/2 signal receive(recomened)
PS2_USE_INT = yes	# uses external interrupt for falling edge of PS/2 clock pin
#PS2_USE_BUSYWAIT = yes	# uses primitive reference code

it seems interupt method with pd0 pd1
But the readme said pd2pd5
I think pd0pd1 is right in default configure.

BTW, I hope remap function keys with Media keys which file I can modify? and where is key definition?

tmk, 2019-03-10:
Where did you find that description? I'll have to fix that. I believe INT implemented properly won't drop any key stroke. You can use INT(erruption) implementation, I recommend it with confidence now.
USART impl. should work well if your pin cofiguration allows, it requres inflexible pin configuration though.

youxiaojie, 2019-03-10:
converter/terminal_usb/README

Keyboard converter for IBM terminal keyboard
============================================
It supports PS/2 Scan Code Set 3 and runs on USB AVR chips such like PJRC Teensy.
I tested the converter on ATMega32U4 with 1392595(102keys) and 6110345(122keys).

Source code: https://github.com/tmk/tmk_keyboard
Article: http://geekhack.org/index.php?topic=27272.0

CONNECTION
----------
Keyboard    ATMega32U4
----------------------
Data:       PD2
Clock:      PD5

And VCC and GND, of course. See RESOURCE for keyboard connector pin assign.

youxiaojie, 2019-03-10:
BTW if I want to change my own key definition for example F24 to volume function where I can modify？

youxiaojie, 2019-03-10:
converter/terminal_usb/Makefile

# T

tmk, 2019-03-10:
Edit the latter one to remap keys.

youxiaojie, 2019-03-10:
thanks greatly. I have told you the two files. Can I pull request?  If I want transfer USB hid by spi nrf24l01 is it feasible?I ask for your advice in geekhack but no answer.

youxiaojie, 2019-03-10:
In the former one double # means KC_##K6C, KC_##K6D?

tmk, 2019-03-10:
I can't say anything about nrf24, I don't know at all.

I can't understand your question about "KC_##6C, ...". can you rephrase it?

youxiaojie, 2019-03-10:
if you don't know nrf24  you know spi protocol? I see you are expert so I ask your option.
This is the first define keymap two #
#define KEYMAP(
K08,K10,K18,K20,K28,K30,K38,K40,K48,K50,K57,K5F,
K07,K0F,K17,K1F,K27,K2F,K37,K3F,K47,K4F,K56,K5E,

K05,K06, K0E,K16,K1E,K26,K25,K2E,K36,K3D,K3E,K46,K45,K4E,K55,K5D,K66, K67,K6E,K6F, K76,K77,K7E,K84,
K04,K0C, K0D,K15,K1D,K24,K2D,K2C,K35,K3C,K43,K44,K4D,K54,K5B, K5C, K64,K65,K6D, K6C,K75,K7D,K7C,
K03,K0B, K14,K1C,K1B,K23,K2B,K34,K33,K3B,K42,K4B,K4C,K52, K53,K5A, K63, K6B,K73,K74,K7B,
K83,K0A, K12,K13,K1A,K22,K21,K2A,K32,K31,K3A,K41,K49,K4A, K51,K59, K61,K62,K6A, K69,K72,K7A,K79,
K01,K09, K11, K19, K29, K39, K58, K60, K68,K70,K71,K78
) {
{ KC_NO, KC_##K01, KC_NO, KC_##K03, KC_##K04, KC_##K05, KC_##K06, KC_##K07 },
{ KC_##K08, KC_##K09, KC_##K0A, KC_##K0B, KC_##K0C, KC_##K0D, KC_##K0E, KC_##K0F },
{ KC_##K10, KC_##K11, KC_##K12, KC_##K13, KC_##K14, KC_##K15, KC_##K16, KC_##K17 },
{ KC_##K18, KC_##K19, KC_##K1A, KC_##K1B, KC_##K1C, KC_##K1D, KC_##K1E, KC_##K1F },
{ KC_##K20, KC_##K21, KC_##K22, KC_##K23, KC_##K24, KC_##K25, KC_##K26, KC_##K27 },
{ KC_##K28, KC_##K29, KC_##K2A, KC_##K2B, KC_##K2C, KC_##K2D, KC_##K2E, KC_##K2F },
{ KC_##K30, KC_##K31, KC_##K32, KC_##K33, KC_##K34, KC_##K35, KC_##K36, KC_##K37 },
{ KC_##K38, KC_##K39, KC_##K3A, KC_##K3B, KC_##K3C, KC_##K3D, KC_##K3E, KC_##K3F },
{ KC_##K40, KC_##K41, KC_##K42, KC_##K43, KC_##K44, KC_##K45, KC_##K46, KC_##K47 },
{ KC_##K48, KC_##K49, KC_##K4A, KC_##K4B, KC_##K4C, KC_##K4D, KC_##K4E, KC_##K4F },
{ KC_##K50, KC_##K51, KC_##K52, KC_##K53, KC_##K54, KC_##K55, KC_##K56, KC_##K57 },
{ KC_##K58, KC_##K59, KC_##K5A, KC_##K5B, KC_##K5C, KC_##K5D, KC_##K5E, KC_##K5F },
{ KC_##K60, KC_##K61, KC_##K62, KC_##K63, KC_##K64, KC_##K65, KC_##K66, KC_##K67 },
{ KC_##K68, KC_##K69, KC_##K6A, KC_##K6B, KC_##K6C, KC_##K6D, KC_##K6E, KC_##K6F },
{ KC_##K70, KC_##K71, KC_##K72, KC_##K73, KC_##K74, KC_##K75, KC_##K76, KC_##K77 },
{ KC_##K78, KC_##K79, KC_##K7A, KC_##K7B, KC_##K7C, KC_##K7D, KC_##K7E, KC_NO },
{ KC_NO, KC_NO, KC_NO, KC_##K83, KC_##K84, KC_NO, KC_NO, KC_NO, },
}

tmk, 2019-03-10:
I think you have to check keymap doc first and then wiki pages.
https://github.com/tmk/tmk_core/blob/master/doc/keymap.md
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap

youxiaojie, 2019-03-10:
tell you the truth I have read it... but still puzzled. Thanks for your patience.I take 3 monthes to find main () and find it in v-usb and fupa.

youxiaojie, 2019-03-11:
Previous Section 	 < Day Day Up >  	Next Section

Using Arguments with #define

By using arguments, you can create function-like macros that look and act much like functions. A macro with arguments looks very similar to a function because the arguments are enclosed within parentheses. Function-like macro definitions have one or more arguments in parentheses, and these arguments then appear in the replacement portion, as shown in Figure 16.2.

Figure 16.2. Parts of a function-like macro definition.

graphics/16fig02.gif

Here's a sample definition:

#define SQUARE(X) X *X

It can be used in program like this:

z = SQUARE(2);

This looks like a function call, but it doesn't necessarily behave identically. Listing 16.2 illustrates using this and a second macro. Some of the examples also point out possible pitfalls, so read them carefully.
Listing 16.2. The mac_arg.c Program

/* mac_arg.c -- macros with arguments */

#include <stdio.h>

#define SQUARE(X) X*X

#define PR(X)   printf("The result is %d.\n", X)

int main(void)

{

    int x = 4;

    int z;

    printf("x = %d\n", x);

    z = SQUARE(x);

    printf("Evaluating SQUARE(x): ");

    PR(z);

    z = SQUARE(2);

    printf("Evaluating SQUARE(2): ");

    PR(z);

    printf("Evaluating SQUARE(x+2): ");

    PR(SQUARE(x+2));

    printf("Evaluating 100/SQUARE(2): ");

    PR(100/SQUARE(2));

    printf("x is %d.\n", x);

    printf("Evaluating SQUARE(++x): ");

    PR(SQUARE(++x));

    printf("After incrementing, x is %x.\n", x);

    return 0;

}
[... 230 lines cut ...]
    PR(2, "x = %.2f, y = %.4f\n", x, y);

    return 0;

}

In the first macro call, X has the value 1, so #X becomes "1". That makes the expansion look like this:

print("Message " "1" ": " "x = %g\n", x);

Then the four strings are concatenated, reducing the call to this:

print("Message 1: x = %g\n", x);

Here's the output:

Message 1: x = 48

Message 2: x = 48.00, y = 6.9282

Don't forget, the ellipses have to be the last macro argument:

#define WRONG(X, ..., Y)  #X #_ _VA_ARGS_ _ #y

Previous Section 	 < Day Day Up >  	Next Section

closed 2019-03-16 by youxiaojie