tmk/tmk_keyboard #597: Can't compile TMK - ./halconf_community.h: No such file or directory

lebastaq, 2019-03-16:
Hey guys,

I'm having trouble compiling TMK for my Teensy 3.2

I followed the instructions:

- Cloned TMK
- Cloned Chibios into tmk_core/tools/chibios/ChibiOs
- Cloned Chibios contrib into tmk_core/tools/chibios/ChibiOs-Contrib

Went into folder keyboards/teensy3.2_tkl and make

Here's the error log:
[https://pastebin.com/BHEizVwC](pastebin)

Does anyone have some hinsight ?
So far it's been a difficult process

Thanks
Cheers

lebastaq, 2019-03-17:
Followed these instructions https://github.com/qmk/qmk_firmware/blob/master/keyboards/chibios_test/teensy_lc_onekey/instructions.md

Started a new project using the console and copied over the teensy 3.2 configuration:

`MCU = cortex-m4
F_CPU = 16000000
MCU_FAMILY = KINETIS
MCU_SERIES = K20x
MCU_STARTUP = k20x7
BOARD = PJRC_TEENSY_3_1
ARMV = 7`

Still not working, but getting closer...

closed 2019-03-17 by lebastaq

SkyCoops, 2019-08-01:
Hello @lebastaq, have you find a solution yet?

I am encoutering the same problem. Any help would be appreciated.

lebastaq, 2019-08-01:
Hey,

I didn't find a solution.
Instead, I bought a pro micro and used QMK

Best regards

SkyCoops, 2019-08-01:
That's very unfortunate.

I am feeling that I have to get a pro micro since it is the standard here.

Let's desolder everything...

Thank you, regards