tmk/tmk_keyboard #602: why magic-x magic-v magic-c don't work?

youxiaojie, 2019-03-17:
why magic-x magic-v magic-c don't work?
magic-d magic-k magic-s work!

tmk, 2019-03-17:
what's keyobard are you using?

youxiaojie, 2019-03-18:
# Build Options
#   comment out to disable the options.
#
#BOOTMAGIC_ENABLE = yes	# Virtual DIP switch configuration(+1000)
#MOUSEKEY_ENABLE = yes	# Mouse keys(+4700)
#EXTRAKEY_ENABLE = yes	# Audio control and System control(+450)
CONSOLE_ENABLE = yes	# Console for debug(+400)
COMMAND_ENABLE = yes    # Commands for debug and configuration
#SLEEP_LED_ENABLE = yes  # Breathing sleep LED during USB suspend
#NKRO_ENABLE = yes	# USB Nkey Rollover

/* key combination for command */
#define IS_COMMAND() ( \
    (keyboard_report->mods == (MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT))) || \
    (keyboard_report->mods == (MOD_BIT(KC_RALT) | MOD_BIT(KC_RCTL)))  \
)

without change.
with hid_listen, I can see magic-d control the debug message magic-s to see some info,  magic-k to see hid-report? 8byte. great work.

tmk, 2019-03-18:
Ah, M122 is 2kro, so some combination of keys are blocked.

closed 2019-03-18 by tmk