tmk/tmk_keyboard #605: Add #[wiki/keymap-examples](https://github.com/tmk/tmk_keyboard/wiki/Keymap-examples) to homepage (readme).

chenzhihuai1990, 2019-03-23:
Add [wiki/keymap-examples](https://github.com/tmk/tmk_keyboard/wiki/Keymap-examples) to homepage [readme](https://github.com/tmk/tmk_keyboard/). In order to let more people share their idea.

I have just found that there is a page for us to share our example, I have just added my own. I think this page is hard to found out, why not add a link to it in homepage?

tmk, 2019-03-24:
it seems to be good idea. I will do.

GianniGi, 2019-05-28:
In the page https://github.com/tmk/tmk_keyboard/wiki/Keymap-examples
there are many non working links. Every link should instead point to GitHub. You can link the author page for attribution, but you mix mirror everything in GitHub.

For example this link doesn't work anymore
"Reverse-shifted for numbers"

tmk, 2019-05-28:
Note that you can edit wiki page. 
Feel free to edit the page to correct them for the community!

Thank you

closed 2021-11-11 by tmk