tmk/tmk_keyboard #616: ADB: Add Power button support

hadess, 2019-06-18:
The 4th pin on the Mini-DIN ADB plug is reserved for the power button. Ideally, if that pin was wired to the micro controller, it would just work as the power button on a multimedia keyboard, and ignored if not connected.

hadess, 2019-06-18:
Is setting `ADB_PSW_BIT` to `1` and wiring the pin to PD1 enough?

If so, I'll send a PR to update the docs to show that.

tmk, 2019-06-18:
When pressing Power button the pin is pulled low level and it sends scan code '7F7F' at same time.
So you don't have to watch the pin because we receive the scan code. Do you find the fact?

Yes, if you really need to use the pin you can define ADB_PSW_BIT and use `psw_in()` to check the pin.
https://github.com/tmk/tmk_keyboard/blob/cf6df9db5fa6123b418ca29455d467e4455c0ebd/tmk_core/protocol/adb.c#L54

hadess, 2019-06-19:
> When pressing Power button the pin is pulled low level and it sends scan code '7F7F' at same time.
> So you don't have to watch the pin because we receive the scan code. Do you find the fact?

I'm guessing not since it didn't generate a "Power" keycode when used in Linux. I'll have to see about flashing a debug firmware so I can see whether it generates anything internally.

The keyboard in question is an Apple Keyboard II (M0487).

tmk, 2019-06-19:
Check it with `xev`, when you tap the button you will see something in output from the command.
Use `hid_listen` to see debug print, and hold the button and press 'v' you will see TMK version output.
Also try to hold the button and press 'x' to see scan codes from the keyboard.

It seems that with default keymap the button sends nothing, you can assign what you want to it with editing keymap.

hadess, 2019-06-20:
I used `evtest` to test the keyboard's scancode, it didn't output anything.
hid_listen and your explanation was what I needed though. I expected the power button to work as a power button, not as the "magic command".

Is there an existing way to make the power button work as a power button rather than as the "magic command"?

tmk, 2019-06-20:
You are build firmware from source code and using `keymap_plain.c`, perhaps? It doesn't assing nothing on power button for some reason. You can change the keymap file, of course.
If you can use it as 'magic key' it means you power button spit scan code '7F7F'.

I don't use the the keymap anymore and use `unimap` instead.

hadess, 2019-06-20:
> You are build firmware from source code and using `keymap_plain.c`, perhaps?

I used the default (from `make -f Makefile.teensy in converter/adb_usb`). `unimap` isn't used by default, and it indeed looks like `plain` is used by default. Forcing `unimap` (with `make -f Makefile.teensy UNIMAP_ENABLE=yes`) worked splendidly. Thanks!

Should the default be changed? Or should we simply make some changes to the README to mention that using `unimap` is recommended?

tmk, 2019-06-20:
Great. I'll update some later.

closed 2019-06-20 by tmk

hadess, 2019-06-21:
> Great. I'll update some later.

Can you explain what change you'll be making? I don't know which option you choose, but I'm guessing you weren't interested in a merge request for it.

tmk, 2019-06-21:
I intend that power button works as 'magic key' when hold it and power key(consumer page) when you tap by default configuration. This is what firmware of Unimap/Keymap Editor offers currently.

And some repository cleaning will be planned and unimap will be default keymap file.