tmk/tmk_keyboard #617: No Device Present

fredfortier, 2019-06-20:
I'm trying to configure the Hasu USB controller installed in my HHKB Pro 2. I'm unfamiliar with the HHKB internals so I'm just naively following the instructions. I'm using Linux on a recent Thinkpad X1 (also tried with an X400). My HHKB Pro 2 is connected to the computer via the USB port of the Hash controller. Here what happens when I type following command from the HHKB:
```
$ dfu-programmer --version
dfu-programmer 0.7.2

$sudo dfu-programmer atmega32u4 erase --suppress-validation --force --debug 300
     target: atmega32u4
    chip_id: 0x2ff4
  vendor_id: 0x03eb
    command: erase
      quiet: false
      debug: 300
device_type: AVR
------ command specific below ------
   validate: false

dfu.c:320: dfu_device_init( 1003, 12276, 0x7ffde4e3e130, true, false )
dfu.c:324: dfu_device_init(000003eb, 00002ff4)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.008402] [00004700] libusb: debug [libusb_get_device_list]
[ 0.008410] [00004700] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.008413] [00004700] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.008416] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  0: 0x0bda, 0x0411
[ 0.008420] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  1: 0x1d6b, 0x0003
[ 0.008424] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  2: 0x1d6b, 0x0002
[ 0.008428] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  3: 0x0bda, 0x0328
[ 0.008432] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  4: 0x1d6b, 0x0003
[ 0.008437] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  5: 0x06cb, 0x009a
[ 0.008444] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  6: 0x13d3, 0x56b2
[ 0.008450] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  7: 0x8087, 0x0a2b
[ 0.008457] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  8: 0x046d, 0x0825
[ 0.008465] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338:  9: 0x17e9, 0x4307
[ 0.008472] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 10: 0xfeed, 0xcafe
[ 0.008478] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 11: 0x047d, 0x8018
[ 0.008485] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 12: 0x0b95, 0x6802
[ 0.008492] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 13: 0x048d, 0x1165
[ 0.008499] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 14: 0x1038, 0x1260
[ 0.008505] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 15: 0x1a40, 0x0201
[ 0.008512] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 16: 0x0bda, 0x5411
[ 0.008518] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 17: 0x056a, 0x5147
[ 0.008524] [00004700] libusb: debug [libusb_get_device_descriptor]
dfu.c:338: 18: 0x1d6b, 0x0002
dfu-programmer: no device present.
[ 0.008534] [00004700] libusb: debug [libusb_exit]
[ 0.008536] [00004700] libusb: debug [libusb_exit] destroying default context
```
I have no idea where to go from here. Any pointers would be appreciated.

tmk, 2019-06-20:
did you push program button on the pcb?

You should get output like below in /var/log/kern.log or other log file when you press the button.
```
Jun 20 10:18:36 desk kernel: [2251082.558167] usb 3-3.2: USB disconnect, device number 70
Jun 20 10:18:36 desk kernel: [2251083.034435] usb 3-3.2: new full-speed USB device number 71 using xhci_hcd
Jun 20 10:18:36 desk kernel: [2251083.184019] usb 3-3.2: New USB device found, idVendor=03eb, idProduct=2ff4
Jun 20 10:18:36 desk kernel: [2251083.184023] usb 3-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 20 10:18:36 desk kernel: [2251083.184026] usb 3-3.2: Product: ATm32U4DFU
Jun 20 10:18:36 desk kernel: [2251083.184028] usb 3-3.2: Manufacturer: ATMEL
Jun 20 10:18:36 desk kernel: [2251083.184030] usb 3-3.2: SerialNumber: 1.0.0
```

fredfortier, 2019-06-20:
> did you push program button on the pcb

I'm ashamed to say that I don't know what this means. Are you referring to the red button on the back of the Hasu controller?

tmk, 2019-06-20:
yes, the red button. Refer to this also.
https://github.com/tmk/tmk_keyboard/wiki#flash-firmware

fredfortier, 2019-06-24:
Same problem:
```
$ sudo dfu-programmer atmega32u4 erase --force
dfu-programmer: no device present.
```
I unplugged cable and plugged in back when holding the red button. I can tell that it booted in an alternate mode because I could no longer use the keyboard. However, I got the same error when typing the command above my built-in keyboard.

tmk, 2019-06-24:
Yes, TMK HHKB Alt Controller uses  'atmega32u4'.

Just plug usb cable normally and then push the button to start bootloader(program/flash mode).

Do you see a line like below in output of 'lsusb' command?
```
$ lsusb
...
Bus 003 Device 049: ID 03eb:2ff4 Atmel Corp. atmega32u4 DFU bootloader
...
```

which os are you using?
'mtp-probe' or 'systemd-logind' may cause, you can try removing these.
I have seen 'mtp' or something like that before and don't have it on my ubuntu now at least. It is called as 'mtp-tools' package in ubuntu. I never seen 'systemd-logind' in my os, stopping/removing this may be worth to try if possible.

And can you post some closeup pics of your controller to check if there are any hardware problem, perhaps ?

fredfortier, 2019-06-25:
> Just plug usb cable normally and then push the button to start bootloader(program/flash mode).
That's what I'm doing but the removing issues occur after pushing the button. 

> Do you see a line like below in output of 'lsusb' command?
No

> which os are you using?
Fedora 30

I'll try plugging the keyboard after a system restart to see if it resolves the remove issue. I'll also try on my other Thinkpad which is running Debian. If none of those work, I'll re-open the keyboard and take the pics.

tmk, 2019-06-25:
Also try it on Windows if possible.