tmk/tmk_keyboard #631: avrdude: ser_open(): can't open device "COM11": No such file or directory

Lonelyzhc, 2019-09-26:
When I enter “avrdude -patmega32u4 -cavr109 -b57600 -Uflash:w:usb_usb.hex -PCOM11”, it shows“avrdude: ser_open(): can't open device "COM11": No such file or directory”。In fact, I used the u2u PCB file you designed.And the Atmega32u4 chip on this u2u comes from an Arduino Pro micro.The strange thing is that you can upload files on the Arduino IDE with this u2u.Can you help me analyze the reasons for this?

tmk, 2019-09-27:
You will have to looking into how the IDE works and emulate it?

tmk, 2019-09-27:
And you may want to compare schematics between promicro and the converter. The difference causes something?

Lonelyzhc, 2019-10-05:
After trying a lot of methods failed, I gave up this unworkable U2U.Then I made two new U2Us, but the two U2Us still have problems.After soldering the U2U and installing the driver, I can download the firmware to U2U.However, although the backlight is on after the keyboard is plugged in, the computer does not respond no matter which button is pressed.

Lonelyzhc, 2019-10-05:
However, when I press the "CapsLock" button on the keyboard that comes with the laptop, the "CapsLock" indicator on the keyboard that is connected to the U2U will light up.My keyboard model is cherry MX Board 1.0.

Lonelyzhc, 2019-10-05:
I was wondering if this is a problem between the max3421E and the keyboard communication. But I don't know how to find U2U hardware problems.

tmk, 2019-10-05:
Try other keyobards than the "cherry MX Board 1.0" to confirm whether the problem is specific to the keyboard or not.

Lonelyzhc, 2019-10-06:
I found another keyboard connected to U2U and found it to work. This is just too exciting, thank you very much for your help.

Lonelyzhc, 2019-10-06:
Unfortunately, U2U is not compatible with "cherry MX Board 1.0".

tmk, 2019-10-07:
Can you share USB descriptor of the keyboard? It would be helpful.

https://github.com/tmk/tmk_keyboard/wiki/USB-Descriptor

Lonelyzhc, 2019-10-07:
Sorry, because my major is not in software code. I really want to do this but I have tried it several times tonight without success.I will ask my friend to help me tomorrow. I will try to share it because I know this information may help.

tmk, 2020-12-28:
@Lonelyzhc
Try the latest firmware code and report back if you still have the keyboard and the converter.
Some bugs were fixed since then and those changes may solved this problem.

Thanks

closed 2020-12-28 by tmk