tmk/tmk_keyboard #632: one key stroke cause entire column of keys to be registered, sort of

xinanhuang, 2019-10-07:
So I just designed my first keyboard and I'm using an atmega32u4 for the controller. I made sure the pcb design is completely working and added diodes to prevent ghosting when multiple keys are pressed. However, after modifying the onekey.c (and referencing his code: https://github.com/w4ilun/pocket-keyboard)  from the keyboard folder and flashed the firmware I wasn't able to get the keyboard completely functional. The symptom so far is that when I press one key the entire column gets registered. (excepted the keys from the 7th row, which seems to work fine).

here is my matrix.c for configuring pins.

/////////////////////////////////////////////////////////
/*
 * scan matrix
 */
#include <stdint.h>
#include <stdbool.h>
#include <avr/io.h>
#include <util/delay.h>
#include "print.h"
#include "debug.h"
#include "util.h"
#include "matrix.h"

#ifndef DEBOUNCE
#   define DEBOUNCE 5
#endif
static uint8_t debouncing = DEBOUNCE;

/* matrix state(1:on, 0:off) */
static matrix_row_t matrix[MATRIX_ROWS];
static matrix_row_t matrix_debouncing[MATRIX_ROWS];

static matrix_row_t read_cols(void);
static void init_cols(void);
static void unselect_rows(void);
static void select_row(uint8_t row);

void matrix_init(void)
{   
    debug_enable = true;
    // initialize row and col
    unselect_rows();
    init_cols();

    // initialize matrix state: all keys off
    for (uint8_t i=0; i < MATRIX_ROWS; i++) {
        matrix[i] = 0;
        matrix_debouncing[i] = 0;
    }
}

uint8_t matrix_scan(void)
{
    for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
        select_row(i);
        _delay_us(300);  // without this wait read unstable value.
        matrix_row_t cols = read_cols();
        if (matrix_debouncing[i] != cols) {
            matrix_debouncing[i] = cols;
            if (debouncing) {
                debug("bounce!: "); debug_hex(debouncing); debug("\n");
            }
            debouncing = DEBOUNCE;
        }
        unselect_rows();
    }

    if (debouncing) {
        if (--debouncing) {
            _delay_ms(1);
        } else {
            for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
                matrix[i] = matrix_debouncing[i];
            }
        }
    }

    return 1;
}

inline
matrix_row_t matrix_get_row(uint8_t row)
{
    return matrix[row];
}

/* Column pin configuration
 * col: 0  1  2  3  4  5  6  7  8  9  
 * pin: B0 B1 B2 B3 B4 B5 B6 F0 F1 F4
 */
static void  init_cols(void)
{
    // Input with pull-up(DDR:0, PORT:1)
    DDRB  &= ~(1<<0 | 1<<1 | 1<<2 | 1<<3 | 1<<4 | 1<<5 | 1<<6);
    PORTB |=  (1<<0 | 1<<1 | 1<<2 | 1<<3 | 1<<4 | 1<<5 | 1<<6);
    DDRF  &= ~(1<<0 | 1<<1 | 1<<4);
    PORTF |=  (1<<0 | 1<<1 | 1<<4);    
}

static matrix_row_t read_cols(void)
{
    return (PINB&(1<<0) ? 0 : (1<<0)) |
           (PINB&(1<<1) ? 0 : (1<<1)) |
           (PINB&(1<<2) ? 0 : (1<<2)) |
           (PINB&(1<<3) ? 0 : (1<<3)) |
           (PINB&(1<<4) ? 0 : (1<<4)) |
           (PINB&(1<<5) ? 0 : (1<<5)) |
           (PINB&(1<<6) ? 0 : (1<<6)) |
           (PINF&(1<<0) ? 0 : (1<<7)) |
           (PINF&(1<<1) ? 0 : (1<<8)) |
           (PINF&(1<<4) ? 0 : (1<<9)) ;
}

/* Row pin configuration
 * row: 0   1   2   3   4   5   6 
 * pin: D0  D1  D2  D3  D4  D5  D6 
 */

static void unselect_rows(void)
{
    // Hi-Z(DDR:0, PORT:0) to unselect
    DDRD  &= ~0b11010000;
    PORTD &= ~0b11010000;    
}

static void select_row(uint8_t row)
{
    // Output low(DDR:1, PORT:0) to select
    switch (row) {

        case 0:
            DDRD  |= (1<<0);
            PORTD &= ~(1<<0);
            break;
        case 1:
            DDRD  |= (1<<1);
            PORTD &= ~(1<<1);
            break;
        case 2:
            DDRD  |= (1<<2);
            PORTD &= ~(1<<2);
            break;
        case 3:
            DDRD  |= (1<<3);
            PORTD &= ~(1<<3);
            break;
        case 4:
            DDRD  |= (1<<4);
            PORTD &= ~(1<<4);
            break;
        case 5:
            DDRD  |= (1<<5);
            PORTD &= ~(1<<5);
            break;
        case 6:
            DDRD  |= (1<<6);
            PORTD &= ~(1<<6);
            break;

    }    
}
/////////////////////////////////////////////////////////

Schematic:
[image]

tmk, 2019-10-07:
`unselect_rows()` is not consitent to pin usages in the schematics. Check there, first.

xinanhuang, 2019-10-07:
Holly shit I'm so dumb thanks that fixed it!

closed 2019-10-07 by xinanhuang