tmk/tmk_keyboard #635: IBM XT Converter generally unreliably

acid2000, 2019-11-10:
I've been doing a lot of testing between Soarer's Converter, TMK and QMK to find what is best for my IBM 5150. This issue is a way of stopping people wasting their time.

By far the most reliable for daily use is Soarer's converter. It seems to have no issues with any of the three keyboards I tested with. TMK & QMK seem to have persistent issue interpreting the protocol correctly. This eventually results in either phantom key presses or stuck keys.

I tried a number of different power supplies to ensure this wasn't a result of undervoltage. The issues persisted.

My recommendation is not to use the XT converter for these keyboards.

acid2000, 2019-11-11:
I've done a lot of work to bottom out the issue. You can find further details here:
https://github.com/qmk/qmk_firmware/issues/7321

tmk, 2019-11-11:
Try the TMK binary with 1K pull-up resistors. 9K ohm you are refering here may be too weak and the external pull-up resistors are must-have. You can't depend on chip's internal pull-up's.
https://github.com/qmk/qmk_firmware/issues/7321#issuecomment-552246676

How we can reproduce the issue? And supply outputs on hid_listen when it happens.

acid2000, 2019-11-12:
I’m going away for a few weeks. When I get back I’ll provide the output. My QMK ticket has the output of their QMK toolbox tool which has the scan codes.

The biggest problem I have with changing the resistors is that it works perfectly fine with Soarers converter ( internal pull-ups alone ).

If it was a pull-up problem I should be seeing this all the time with soarers, right.

It’s also worth changing the documentation on the converter if 1-10k is no longer recommended.

acid2000, 2019-11-12:
It would also be great if there was some confirmation that this issue can/can’t be reproduced with the latest codebase. I’ve got two setups, different keyboards and wiring and can see the same on both.

Just hold down Q was wait for faults at around 600 characters.

tmk, 2019-11-12:
No hurry, take time.
I usually don't have time to test QMK codes and debug it, test with prebuild firmware in TMK repo and report here if you want me to help you.
https://github.com/tmk/tmk_keyboard/tree/master/converter/xt_usb/binary

TMK emulates how original XT PC handles signals from keyboard as possible and it reads data line at diffrent timing from Soarer's. Weak pull-up resistors may cause problem.

At least users have not reported similar issue in the thread so far. You may want to post there to ask people. I don't have 5150 keyobard though, I'll try the 'holding Q' test when I get time anyway.
https://geekhack.org/index.php?topic=94649.0

tmk, 2019-11-12:
Tested holding Q for 5 mins and no issue happened in my site.

```
Waiting for new device:.................                                                                                                                                                                                                      
Listening:                                                                                                                                                                                                                                    
Keyboard init.                                                                                                                                                                                                                                
[C]Keyboard start.                                                                                                                                                                                                                            
AA 2A 36 2F                                                                                                                                                                                                                                                                                                                                                                                                                                 
        - Version -                                                                                                                                                                                                                           
DESC: convert XT keyboard to USB                                                                                                                                                                                                              
VID: 0xFEED(t.m.k.) PID: 0x6512(XT keyboard converter) VER: 0x0001                                                                                                                                                                            
BUILD: fb24846-dirty (22:28:42 Oct  6 2018)                                                                                                                                                                                                   
OPTIONS: LUFA MOUSEKEY EXTRAKEY CONSOLE COMMAND NKRO KEYMAP_SECTION 4096                                                                                                                                                                      
GCC: 4.9.2 AVR-LIBC: 1.8.0svn AVR_ARCH: avr5                                                                                                                                                                                                  
AF B6 AA 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 1
0 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10
 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 
10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 1
0 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 ...
```

acid2000, 2019-11-24:
So i've now changed to 1k pull ups. It seems better but I still have issues. Notice the stray 21's

Waiting for device:
Listening:
10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 21 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 21 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10

tmk, 2019-11-25:
Are you sure using genuine TMK  firmware? You should see lines following lines with the firmware. Did you omit the lines on your previous post.

If you are using the correct firmware I don't have idea now.

```
Waiting for new device:................
Listening:
Keyboard init.
[C]Keyboard start.
```

acid2000, 2019-11-25:
Yes I’m using the firmware you linked in this post. I cropped stuff out of hid_listen.

I didn’t see those other lines in the tool.

acid2000, 2020-05-03:
Cross posted from the Geekhack forums for completeness. 
Hid listen results

[textTyped.txt](https://github.com/tmk/tmk_keyboard/files/4571810/textTyped.txt)
[textTyped-hidlistneout.txt](https://github.com/tmk/tmk_keyboard/files/4571811/textTyped-hidlistneout.txt)
[holdingDownQ.txt](https://github.com/tmk/tmk_keyboard/files/4571812/holdingDownQ.txt)
[holdingDownQ-2.txt](https://github.com/tmk/tmk_keyboard/files/4571813/holdingDownQ-2.txt)
[nothingBeingPressed.txt](https://github.com/tmk/tmk_keyboard/files/4571814/nothingBeingPressed.txt)
---

I’ve been testing the latest TMK firmware binaries in the repo.

Overall it is better than when I was testing in September but still has lots of issues.

I’m using an XT 83 key 5150 UK layout with a 5V Pro Micro without reset connected. It’s either connected into a powered USB hub or with a Y cable into a 2.1amp PSU to ensure the voltage doesn’t drop. It has strong pull-ups and the cable to the board itself is short.

I have built a few pro micros with Soarers before. With XTs I usually power it up with the RAW pin and bypass the regulator all together to make sure the power doesn’t dip. I can’t remember if I did this in this one.

It works well with Soarers and is my daily driver.

I’ve got some hid listen output for a few different states.
* No input, power on and don’t touch anything. I left it for around 5m. I was expecting to see nothing but got some things that look like errors.
* Holding down Q. I have two different outputs with slightly different results. I’ve tried a few keys and Q always seems to show issues.
* General typing. I’ve included the ASCII output from the keyboard and the corresponding hid_listen results.

Testing is a PITA for me. For some reason I can’t get the Soarers fw into the bootloader from Windows. I need to bridge the reset pin by hand. I have a few 5150 keyboards and in all of them my adapter is inside the keyboard glued in place and wrapped in electrical tape.

Back in September there were lots of problems with stuck keys. This looks like that is solved (only did a bit of testing) but instead I do get an incorrect keypresses. Is this a timing error?

I’m really hoping that one day soon these issues can be resolved. I’m quite familiar with the TMK codebase now and already have some mods I’ve written I’d like to push up when they’ve been tested.

i-am-shodan, 2020-05-07:
I’ve now complete replaced the wiring and have moved to a teensy 2.0.

Very similar issues.

DalvikVM, 2020-05-07:
Hello!

I am having the same issues too. Tried pull up resistors with a separate power source and without. It works fine with soarers on a teensy 2.0. Usually it'll work for a few seconds, and then jam up. It seems that Ctrl gets stuck on and the f key gets jammed on also. If I hit space after the f key goes nuts, it'll stop, but Ctrl still seems to be jammed.  Usually Ill get the phantom key presses even when it is just sitting, but it takes a bit longer.

Now, some side notes. I am using a Adafruit Feather ATMEGA32u4 to run this thing. And the only way I could get the converter to not output a device descriptor error on windows immediately was to set the processor frequency to:
F_CPU = 8000000

Which, if I had to take a wild guess, may be messing with the timing of the key codes

Then it'll connect properly.

That and the bootloader are the only things I've changed from stock code.

Here's the output I get from QMK Toolbox HID Console with the caterina bootloader with no interaction:

*** QMK XT Keyboard converter (FEED:6512:0001)

> 2B 95 13 15 93 13 2F 2F 3E 17 11 3D 9E 20 1B BC 3C

*** HID device disconnected (FEED:6512:0001)
It'll go completely batty if I start typing, as in unplug the keyboard and reboot my computer sort of batty.

Heres HalfKay bootloader (same device) with random typing, but not holding any keys:

*** HID device connected: QMK XT keyboard converter (FEED:6512:0001)
  > 39 B9 39 B9 39 B9 39 B9 39 B9 39 B9 39 B9 32 B2 24 A4 2C 2C 2C 2C 2C 59 AC 38 39 B9 B8 38 73 39 39 B9 B8 71 39 B9 39 73 E2 38 24 31 26 25 B1 24 25 24 A4 25 4B 92 A4 39 B9 24 A4 24 A4 24 A4 2C AC 3D 9E 23 A3 20 24 32 32 18 98 19 33 66 99 18 31 5E 97 18 
*** HID device disconnected (FEED:6512:0001)

For whatever reason the halfkay bootloader seems to work better, with it not doing anything while being left alone. Still got phantom key presses during typing, no Ctrl issues thus far, still have the F_CPU set to 8000000. 

My gut tells my it's a clocking issue somehow....

I'm new to the TMK/QMK world so I'm still figuring this stuff out. Probably an easy and dumb solution.

Thanks for any suggestions or help!

i-am-shodan, 2020-05-07:
Interesting you mention the F key. With my Q test when it faults I get Fs.

i-am-shodan, 2020-05-07:
Worth also mentioning is that my 5150 has absolutely no reset pin, the header inside is missing the connection.

i-am-shodan, 2020-05-07:
Why F instead of Q?

According to the docs XT outputs scan set 1 which TMK takes and outputs as USB. I’ve looked up the Q and F scan codes. I think these are 10 and 21 respectively. If I’m right in binary this is:
Q=00001010
F=00010101

They have very similar representations in binary, and if F was shifted right then they would be the same.

My guess is this is what TMK is doing probably because for some reason it is not reading data at the right time.

tmk, 2020-05-08:
@DalvikVM , @i-am-shodan 

XT converter has issue that can miss start bit signal possibly. I think the issue was fixed somehow in IBMPC converter. But I still don't totally know why phantom key press happens without touching keys.

Can you test with this a bit old prebuild firmware of IBMPC converter?
https://github.com/tmk/tmk_keyboard/blob/8f86d125c353999372996bce6548a32e9264248a/converter/ibmpc_usb/binary/ibmpc_usb_atmega32u4.hex

DalvikVM, 2020-05-08:
Starting out, I switched to a teensy 2.0 to test it. (which has the same issues with QMK BTW). I couldn't use the firmware on the feather, as it kept throwing a usb error back at me. 

So far it's been bratty for me. It works good and doesn't freeze up my computer, but it'll randomly spew gibberish as I'm typing. Not exactly ideal. But there aren't any phantom key presses or repeated key so far. I'm using it to type this out now. Interesting, it just 6yuhbn 8iojk,m0p25tc.7ep10p;[12zed[](url) there 8ij,i

OK. There it went. It moved the focus to another window, worked again for a moment and then spewed out the above. Heres what hid picked up.

*** HID device connected: t.m.k. IBM PC keyboard converter (FEED:1BEE:0001)
  > 
    
    TMK:fefe1028/LUFA
    
    
    USB configured.
    
    Keyboard start.
    I256 X757 rAA W787 R1289 wF2 
    ERR:11
    ID:FFFF(1)
    S1324 L1325 r32 rB2 r12 r92 r34 rB4 r39 rB9 r0E r8E r2A r1F rAA r9F r14 r94 r1E r9E r13 r93 r14 r94 r17 r31 r97 r22 rB1 rA2 r39 rB9 r18 r16 r98 r14 r96 r94 r33 rB3 r2A r1F rAA r18 r9F r98 r39 rB9 r21 rA1 r1E r13 r9E r93 r39 rB9 r17 r97 r14 r94 r28 rA8 r1F r9F r39 rB9 r30 rB0 r12 r92 r12 r92 r31 r39 rB1 rB9 r30 rB0 r13 r93 r1E r9E r14 r94 r14 r94 r15 r95 r39 rB9 r21 rA1 r18 r13 r98 r93 r39 rB9 r32 rB2 r12 r92 r34 rB4 r39 rB9 r2A r17 rAA r97 r14 r94 r39 rB9 r11 r91 r18 r98 r13 r93 r25 rA5 r1F r9F r39 r07 r15 r23 r30 r31 r87 r95 rA3 rB0 rB1 rB9 r13 r12 r93 r92 r1E r9E r26 rA6 r26 rA6 r15 r95 r39 rB9 r22 rA2 r18 r98 r18 r98 r20 rA0 r39 rB9 r21 rA1 r18 r13 r98 r93 r39 rB9 r1E r9E r39 rB9 r11 r91 r23 rA3 r17 r97 r26 rA6 r12 r92 r0E r0E r0E r0E r0E r0E r0E r0E r0E r8E r0E r8E r0E r8E r13 r93 r12 r92 r1E r9E r26 rA6 r26 rA6 r15 r95 r39 rB9 r22 rA2 r18 r98 r18 r98 r20 r39 rA0 rB9 r1E r9E r31 rB1 r20 rA0 r39 rB9 r20 r18 rA0 r98 r12 r92 r1F r9F r31 rB1 r28 rA8 r14 r94 r39 rB9 r21 rA1 r13 r93 r12 r92 r12 r92 r2C rAC r12 r92 r39 rB9 r16 r96 r19 r99 r39 rB9 r32 rB2 r15 r95 r39 rB9 r2E rAE r18 r32 r98 rB2 r19 r99 r16 r96 r14 r94 r12 r13 r92 r93 r33 rB3 r39 rB9 r30 rB0 r16 r96 r14 r94 r39 rB9 r17 r97 r14 r94 r28 rA8 r26 rA6 r26 rA6 r39 rB9 r13 r93 r1E r9E r31 rB1 r20 rA0 r18 r32 r98 rB2 r26 rA6 r15 r95 r39 rB9 r1F r9F r19 r99 r12 r92 r11 r91 r39 rB9 r22 rA2 r17 r97 r30 rB0 r30 rB0 r12 r13 r92 r93 r17 r97 r1F r9F r23 rA3 r39 rB9 r1E r1F r9E r9F r39 rB9 r2A r17 rAA r97 r28 rA8 r32 r39 rB2 rB9 r14 r94 r15 r95 r19 r99 r17 r31 r97 rB1 r22 rA2 r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r34 rB4 r39 rB9 r2A r17 rAA r97 r26 rA6 r26 rA6 r39 rB9 r19 r18 r99 r98 r1F r9F r14 r94 r39 rB9 r14 r23 r94 rA3 r12 r39 r92 rB9 r0E r8E r0E r8E r0E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r1C r9C r1C r9C r2A r31 rAA rB1 r18 r98 r14 r94 r39 rB9 r12 r92 r2D rAD r1E r9E r2E rAE r14 r94 r26 rA6 r15 r95 r39 rB9 r17 r97 r20 r12 rA0 r92 r1E r9E r26 rA6 r39 rB9 r0E r8E r34 rB4 r34 rB4 r0E r8E r39 rB9 r2A r30 rAA rB0 r16 r96 r14 r94 r39 r14 rB9 r94 r23 rA3 r12 r13 r92 r12 r93 r92 r39 rB9 r1E r9E r13 r12 r93 r92 r31 rB1 r28 rA8 r14 r94 r39 rB9 r1E r9E r31 rB1 r15 r95 r39 rB9 r19 r99 r23 rA3 r1E r9E r31 rB1 r14 r94 r18 r32 r98 rB2 r39 rB9 r25 rA5 r12 r92 r15 r95 r39 rB9 r19 r99 r13 r93 r12 r92 r12 r92 r1F r9F r12 r92 r0E r8E r0E r8E r0E r8E r1F r9F r1F r9F r12 r92 r1F r9F r39 rB9 r18 r98 r13 r93 r39 rB9 r13 r12 r93 r92 r19 r99 r12 r92 r1E r9E r14 r94 r12 r92 r20 rA0 r39 rB9 r25 rA5 r12 r92 r15 r95 r39 rB9 r1F r9F r18 r98 r39 rB9 r1E r9E r0E r8E r21 r1E rA1 r9E r13 r93 r34 rB4 r39 rB9 r2A r17 rAA r97 r32 rB2 r39 rB9 r0E r8E r0E r8E r28 rA8 r32 rB2 r39 rB9 r16 r96 r1F r17 r9F r31 r97 rB1 r22 rA2 r39 rB9 r17 r97 r14 r94 r39 r14 rB9 r94 r18 r98 r39 rB9 r14 r94 r15 r95 r19 r99 r12 r92 r39 rB9 r14 r94 r23 rA3 r17 r97 r1F r9F r39 rB9 r18 r16 r98 r96 r14 r94 r39 r3E r40 r42 r44 r03 r11 r1F r20 r2C rBE rC0 rC2 rC4 r83 r91 r9F rA0 rAC rB9 r31 rB1 r18 r98 r11 r91 r31 r18 rB1 r11 r98 r91 r34 rB4 r39 rB9 r2A r2A r17 r97 rAA r0E r8E r2A r17 rAA r31 r97 rB1 r14 r94 r12 r13 r92 r93 r12 r92 r1F r9F r14 r94 r12 r92 r20 rA0 r0E r8E r17 r31 r97 r22 rB1 rA2 r0E r8E r0E r8E r0E r8E r0E r8E r17 r31 r97 r22 rB1 rA2 r33 rB3 r39 rB9 r17 r97 r14 r94 r39 rB9 r24 rA4 r16 r96 r1F r9F r14 r94 r39 rB9 r07 r15 r16 r23 r30 r31 r39 r7 24 33 0B ERR:FF
  > 
  > Ar
  > 
  > rr14 r2E r16 r39 r18 r34 r47 r12 r86 r94 rAE 
  > ERR:FF
  > r8B rA7 r83 r92 r32 rCF rAC r99 rB2 r19 r4F r99 rCF rr19 r27 r1A r4F r03 r2C rA7 r8B r9A r83 rAC r99 rA6 rCF r81 r98 rB1 rAA rB3 r2A rA0 rAF rAA r12 r92 r20 rA5 r9D r1D r25 rrrrrr93 rA1 rA2 r89 r97 rA4 
  > ERR:FF
  > r87 
  > ERR:FF
  > rB9 r0F r13 r15 r8F r93 r21 r95 r15 rA1 r95 r18 r98 r2E rAE r16 r96 r1F r9F r39 rB9 r39 rB9 r14 r94 r23 rA3 r12 r13 r92 r12 r93 r92 r39 rB9 r0D r8D r0D r8D r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r23 rA3 r12 r13 r92 r12 r93 r92 r39 rB9 r09 r17 r24 r25 r33 r89 r97 rA4 rA5 rB3 r17 r97 r14 r94 r39 rB9 r0B r19 r26 r27 rr1E r1D r2A r03 r11 r1F r20 r2C 
  > ERR:FF
  > r90 
  > ERR:FF
  > r81 
  > ERR:FF
  > rB3 r33 rB3 rAA rA0 rAF r12 r92 
*** HID device disconnected (FEED:1BEE:0001)

DalvikVM, 2020-05-08:
BTW, thanks for the help! Everything from "OK there it went" down I used another keyboard for.

tmk, 2020-05-09:
No problem! Thanks for your time.

i-am-shodan, 2020-05-09:
I tested the firmware you linked above.

Generally it is much better but I still get an F instead of a Q. I think the error rate has gone from 1/200 keys to about 1/1000.
I can see an error in the output of hid_listen when it happens which I've attached.

[qTest-against8f86d125c353999372996bce6548a32e9264248a.txt](https://github.com/tmk/tmk_keyboard/files/4604890/qTest-against8f86d125c353999372996bce6548a32e9264248a.txt)

tmk, 2020-05-10:
@i-am-shodan Thank you for the test.
As you says the converter mistakes Q(10h) for F(21h) due to misread of its start bit(0).

```
... r10 r10 r10 r10 r21
ERR:20
init
```

On IBM XT keyboards Q(10h) should be read on signal line as below: https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-XT-Keyboard-Protocol#1-ibm-xttwo-start-bits

    0001 0000 10
    `--' `--' ||
     1    0   |`- start bit(0)
              `-- start bit(1)

But firmware read the start bit(0) as 1 wrongly, and recognized as F(21h) as if it is on XT Clone keyboards. https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-XT-Keyboard-Protocol#2-clonesone-start-bit

    0001 0000 11
    |`---'`---'|
    |  2    1   `-- start bit(1)
    `-------------- ignored(causes Timeout Error(ERR:20) in next read)

The firmware seems to be still slow to read start bit(0) correctly, this is root of this problem.
https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-XT-Keyboard-Protocol#note-for-start0

This makes totally sense to me now,  though, I optimistically expected that this happens more rarely. The firmware should be fixed to read the start bit(0) correctly and I wan to do that, but it is difficult and time consuming job if it is possible.

I added adhoc error check for this problem and I think this works for actual use. I can't test the error check code at all because of luck of IBM XT keyboard in hand, though.
https://github.com/tmk/tmk_keyboard/commit/c092948222d941e7ea5be507666da5dedbcc812d

But I still don't have clear idea on a lot of ramdom key and phantom key press that reported before, recent code changes may cause somehow.

Can you try these firmwares? 
I'm also interested in whether there is any diffrence between the two revisions.

"A bit old firmware"(which I linked above) with the error check.
https://github.com/tmk/tmk_keyboard/blob/ibmpc_startbit_error/converter/ibmpc_usb/binary/ibmpc_usb_atmega32u4.hex

The latest firmware with the error check.
https://github.com/tmk/tmk_keyboard/blob/ibmpc_error_handling/converter/ibmpc_usb/binary/ibmpc_usb_atmega32u4.hex

You will see 'XT_ERR' in hid_listen log when the start bit(0) occurs. And the converter should read scan code correctly even with the error.

i-am-shodan, 2020-05-10:
Thanks for looking into this. I will test the FW tonight or tomorrow.

I may end up recompiling them as I need a custom key map. The XT doesn’t have a pause key which makes getting into the bootloader a pain.

Worth saying I don’t get any phantom key presses. I tested before (last year I think) and I did then so I think that has been fixed. The only problem I have is misrepresenting a keypress for another.

Keep up the good work, feels like it’s one bugfix away from working well.

i-am-shodan, 2020-05-10:
I would love to know how Soarers converter works around this. It seems to accurately catch the start(0) bit.

I tried to reverse engineer the firmware with GHIDRA but without an integrated simulator/debugger it is incredibly difficult to get a handle on how it operates.

DalvikVM, 2020-05-10:
OK. So, this is inexperience talking here, but isn't the ATMEGA32u4's internal clock rather inaccurate? From what I was reading it has a sufficiently accurate clock for basic timers and the like, but for anything accurate, it isn't terribly great. (Again, my understanding is rather rudimentary, so this is kinda spitballing) https://hackaday.com/2013/01/03/accurate-timers-with-an-avr/

At the speed that some type and/or holding a button, might the clock get thrown off and miss or mess up the starting bit? I would assume it gets "reset" somehow when there is a break?

I'm currently on the old bit firmware with error handling you posted and it just went bonkers. Here's the output:

`*** HID device connected: t.m.k. IBM PC keyboard converter (FEED:1BEE:0001)
  > r14 r94 r23 rA3 r12 r39 r92 rB9 r2A r1E r9E r14 r94 r32 rB2 r12 r92 r22 rA2 r1E r9E rAA r04 r03 r84 r83 r16 r96 r05 r85 r28 rA8 r1F r9F r39 rB9 r2E rAE r26 rA6 r18 r98 r2E rAE r25 rA5 r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r17 r31 r97 rB1 r14 r94 r12 r13 r92 r93 r31 rB1 r1E r9E r26 rA6 r39 rB9 r2E rAE r18 r98 r0E r8E r26 rA6 r18 r98 r2E rAE r25 rA5 r39 rB9 r13 r93 r1E r9E r14 r94 r23 rA3 r12 r13 r92 r93 r39 rB9 r17 r31 r97 rB1 r1E r9E r2E rAE r2E rAE r16 r96 r13 r93 r1E r9E r14 r94 r12 r92 r36 r35 rB6 rB5 r39 rB9 r2A rAA r21 rA1 r13 r93 r0E r8E r0E r8E r2A r21 rAA rA1 r13 r93 r18 r32 r98 rB2 r39 rB9 r11 r91 r23 rA3 r1E r9E r14 r94 r39 rB9 r2A r17 rAA r97 r39 rB9 r11 r1E r91 r1F r9E r9F r39 rB9 r13 r12 r93 r92 r1E r9E r20 rA0 r17 r31 r97 r22 rB1 rA2 r39 rB9 r14 r94 r17 r97 r39 rB9 r0E r8E r0E r8E r17 r97 r14 r94 r39 rB9 r26 rA6 r18 r98 r1F r9F r12 r92 r1F r9F r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r17 r97 r14 r94 r39 rB9 r26 rA6 r18 r98 r1F r9F r12 r92 r1F r9F r39 rB9 r17 r97 r14 r94 r28 rA8 r1F r9F r39 rB9 r14 r94 r17 r32 r97 rB2 r12 r92 r17 r31 r97 r22 rB1 rA2 r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r0E r8E r0E r8E r0E r8E r20 rA0 r18 r98 r12 r92 r0E r8E r0E r8E r0E r8E r23 rA3 r1E r1F r9E r9F r39 rB9 r1E r9E r39 rB9 r1F r9F r16 r96 r21 rA1 r21 rA1 r17 r97 r2E rAE r12 r92 r31 rB1 r14 r94 r26 rA6 r15 r95 r39 rB9 r1E r9E r2E rAE r2E rAE r16 r96 r13 r93 r1E r9E r14 r94 r12 r92 r39 rB9 r2E rAE r26 rA6 r18 r98 r2E r25 rAE rA5 r39 rB9 r21 rA1 r18 r13 r98 r93 r39 rB9 r32 rB2 r0D r8D r0E r8E r0E r8E r30 rB0 r1E r1F r9E r9F r17 r97 r2E rAE r39 rB9 r14 r94 r17 r32 r97 rB2 r12 r13 r92 r93 r1F r9F r39 rB9 r1E r9E r31 rB1 r20 rA0 r39 rB9 r14 r94 r23 rA3 r12 r39 r92 rB9 r26 r17 rA6 r97 r25 rA5 r12 r92 r33 rB3 r39 rB9 r30 rB0 r16 r96 r14 r94 r39 rB9 r17 r97 r14 r94 r39 rB9 r20 rA0 r18 r98 r12 r92 r1F r9F r39 rB9 r26 rA6 r18 r98 r1F r9F r12 r92 r39 rB9 r14 r94 r17 r32 r97 rB2 r12 r92 r39 rB9 r0D r8D r0D r8D r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r21 rA1 r18 r13 r98 r93 r39 rB9 r1E r9E r31 rB1 r15 r95 r14 r94 r23 r17 rA3 r31 r97 rB1 r22 rA2 r39 rB9 r1E r9E r2E rAE r21 r2E rA1 rAE r16 r96 r13 r93 r1E r9E r14 r94 r12 r92 r33 rB3 r39 rB9 r17 r97 r14 r94 r39 rB9 r17 r97 r1F r9F r31 rB1 r28 rA8 r14 r94 r39 rB9 r14 r94 r12 r13 r92 r93 r17 r97 r17 r97 r30 rB0 r26 rA6 r15 r95 r39 rB9 r22 rA2 r13 r93 r12 r92 r1E r9E r14 r94 r34 rB4 r39 rB9 r2A r0A r8A rAA r2A r1E rAA r9E r22 rA2 r1E r9E r17 r31 r97 rB1 r33 rB3 r39 rB9 r32 rB2 r15 r95 r39 rB9 r16 r31 r96 rB1 r20 r12 rA0 r13 r92 r93 r1F r9F r14 r94 r1E r9E r31 rB1 r20 rA0 r17 r31 r97 rB1 r22 rA2 r39 rB9 r17 r97 r1F r9F r39 rB9 r13 r93 r1E r9E r14 r94 r23 r12 rA3 r13 r92 r93 r39 rB9 r13 r93 r16 r96 r20 rA0 r12 r92 r32 rB2 r12 r92 r31 rB1 r14 r94 r1E r9E r13 r93 r15 r95 r33 rB3 r39 rB9 r1F r9F r18 r98 r39 rB9 r14 r94 r23 r17 rA3 r97 r1F r9F r39 rB9 r17 r97 r1F r9F r39 rB9 r25 rA5 r17 r97 r31 rB1 r20 rA0 r1E r9E r39 rB9 r1F r9F r19 r17 r99 r97 r14 r94 r30 rB0 r1E r9E r26 rA6 r26 rA6 r17 r31 r97 rB1 r22 rA2 r2A r0B r8B rAA r39 rB9 r1D r2F rAF r9D r1C r9C r1C r9C r2A r17 rAA r97 r21 rA1 r39 r14 rB9 r94 r23 r12 rA3 r39 r92 rB9 r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r2A r1E rAA r14 r9E r94 r39 rB9 r14 r94 r23 r12 rA3 r13 r92 r39 r93 rB9 r0E r8E r0E r8E r39 rB9 r1F r9F r19 r99 r12 r92 r12 r92 r20 rA0 r39 rB9 r14 r94 r23 rA3 r1E r9E r14 r94 r39 rB9 r1F r9F r18 r32 r98 rB2 r12 r92 r39 rB9 r14 r94 r15 r95 r19 r99 r12 r92 r11 r39 r91 rB9 r1E r9E r31 r20 rB1 rA0 r39 rB9 r0E r8E r35 rB5 r18 r98 r13 r93 r39 rB9 r4B r4B r4B rCB r4B rCB r0E r8E r4D r4D r4D r4D rCD r23 rA3 r18 r98 r26 rA6 r20 rA0 r17 r31 r97 rB1 r22 rA2 r39 r14 rB9 r94 r23 rA3 r12 r92 r39 rB9 r30 rB0 r16 r96 r14 r94 r14 r18 r94 r31 r98 rB1 r0D r8D r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r1E r9E r39 rB9 r30 rB0 r16 r96 r14 r94 r14 r18 r94 r31 r98 rB1 r33 rB3 r39 rB9 r32 r17 rB2 r97 r22 rA2 r23 rA3 r14 r94 r39 rB9 r14 r94 r23 rA3 r12 r92 r39 rB9 r14 r94 r17 r32 r97 rB2 r17 r31 r97 rB1 r22 rA2 r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r2E rAE r26 rA6 r18 r98 r2E rAE r25 rA5 r39 rB9 r22 rA2 r12 r92 r14 r94 r39 rB9 r14 r94 r23 rA3 r13 r93 r18 r98 r11 r91 r31 rB1 r39 rB9 r18 r98 r21 rA1 r21 rA1 r39 rB9 r1E r9E r31 rB1 r1F r9F r39 rB9 r32 rB2 r17 r97 r1F r9F r1F r9F r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r20 rA0 r39 rB9 r32 rB2 r17 r97 r1F r9F r1F r9F r39 rB9 r18 r98 r13 r93 r39 rB9 r32 rB2 r12 r92 r1F r9F r1F r9F r39 rB9 r16 r96 r19 r99 r39 rB9 r14 r94 r23 rA3 r12 r39 r92 rB9 r1F r9F r14 r13 r94 r93 r0E r8E r1E r9E r0E [... cut ...]
    I33029 X33531 rAA W33553 r3B W33580 r3D W33591 R33592 wF2 
  > ERR:12
    ID:FFFF(1)
    S33606 L33607 r00 !CS1_ERR!
    I33607 X34109 rAA W34131 r3B W34157 r3D W34169 R34169 wF2 
  > ERR:12
    ID:FFFF(1)
    S34183 L34184 r00 !CS1_ERR!
    I34184 X34686 rAA W34708 r3B W34734 r3D W34746 R34747 wF2 
  > ERR:12
    ID:FFFF(1)
    S34761 L34761 r00 !CS1_ERR!
    I34762 X35263 rAA W35286 r3B W35311 r3D W35324 R35324 wF2 
  > ERR:12
    ID:FFFF(1)
    S35338 L35339 r00 !CS1_ERR!
    I35339 
*** HID device disconnected (FEED:1BEE:0001)
`

tmk, 2020-05-10:
@i-am-shodan ,

My speculation is that Soarer's reads data line at rising edge of clock, it is not difficult to read the start bit and read data as below in that case. 

IBM XT: start(1), bit0, bit1, bit2, bit3, bit4, bit5, bit6, bit7, stop(0)
Clones: start(1), bit0, bit1, bit2, bit3, bit4, bit5, bit6, bit7

https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-XT-Keyboard-Protocol#speculation-on-xt-protocol-of-soarers-converter

Reading at rising edge works with IBM XT keyboards and some clones but it is not how original XT host does and won't be compatible with a few clone keyboards.

I have tried to emulate XT keyboard interface comprised of TTL descrite ICs as possible but my codes misses out the first start bit. I'll have to polish my ISR code, where its function prologue that pushs regsiters into stack take time.

https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-XT-Keyboard-Protocol#host-side-schematics
https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-XT-Keyboard-Protocol#note-for-start0

DalvikVM, 2020-05-10:
So I switched to the other firmware you posted and it lasted less than a minute. It kept switching focus off the window as I was trying to type, and then went completely berserk. Once it went off, I didn't touch the keyboard till it stopped. Here's the output of that one:

 `*** HID device connected: t.m.k. IBM PC keyboard converter (FEED:1BEE:0001)
  > 
    
    TMK:59f750d0/LUFA
    
    
    USB configured.
    
    Keyboard start.
    I254 X755 rAA W779 R1280 wF2 
    ERR:11
    ID:FFFF(1)
    S1315 L1316 r2A r14 rAA r94 r23 rA3 r12 r13 r92 r12 r93 r39 r0F r8F r92 r15 rB9 r95 r11 r91 r1E r1F r9E r9F r31 rB1 r28 rA8 r14 r94 r39 rB9 r1E r9E r31 r39 rB1 r0F r11 r07 r15 r30 r17 r24 r13 r21 rA1 r8F r91 r93 r87 r95 rB0 rB9 rA4 r97 r1F r9F r1F r9F r16 r96 r12 r92 r17 r97 r1F r9F r1F r9F r16 r96 r12 r92 r39 rB9 r14 r94 r17 r97 r26 rA6 r26 rA6 r39 rB9 r14 r94 r23 r12 rA3 r92 r39 rB9 r21 rA1 r17 r97 r13 r93 r1F r9F r14 r94 r39 rB9 r12 r92 r13 r93 r13 r93 r18 r13 r98 r93 r34 rB4 r39 rB9 r2A r17 rAA r97 r39 rB9 r24 rA4 r16 r96 r1F r9F r14 r94 r39 rB9 r1F r9F r11 r91 r17 r97 r14 r94 r2E rAE r23 rA3 r12 r92 r20 rA0 r39 rB9 r14 r94 r18 r98 r39 r14 rB9 r94 r23 r12 rA3 r92 r39 rB9 r1F r9F r12 r92 r2E rAE r18 r31 r98 rB1 r20 rA0 r39 rB9 r1F r9F r12 r92 r2E rAE r18 r31 r98 rB1 r20 rA0 r39 rB9 r21 rA1 r17 r97 r13 r93 r32 rB2 r11 r91 r1E r9E r13 r93 r12 r92 r39 rB9 r1E r9E r31 r20 rB1 rA0 r39 rB9 r17 r97 r14 r94 r39 rB9 r25 rA5 r12 r92 r12 r92 r19 r99 r39 rB9 r1F r9F r11 r91 r17 r97 r14 r94 r2E rAE r23 rA3 r17 r31 r97 rB1 r21 rA1 r0E r8E r22 rA2 r39 rB9 r21 rA1 r18 r98 r2E rAE r16 r96 r1F r9F r39 rB9 r18 r98 r21 rA1 r21 rA1 r39 rB9 r18 r98 r21 rA1 r39 rB9 r14 r94 r23 rA3 r12 r92 r39 rB9 r11 r91 r17 r31 r97 rB1 r20 rA0 r18 r98 r11 r91 r39 rB9 r1E r1F r9E r39 r9F rB9 r2A r17 rAA r97 r39 rB9 r14 r94 r15 r95 r19 r99 r12 r92 r34 rB4 r39 rB9 r2A r30 rAA rB0 r16 r96 r14 r94 r39 rB9 r31 rB1 r18 r98 r39 rB9 r12 r92 r13 r93 r13 r93 r18 r13 r98 r93 r1F r9F r39 rB9 r17 r97 r39 rB9 r14 r94 r23 rA3 r12 r39 r92 rB9 r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r31 rB1 r39 rB9 r14 r94 r23 rA3 r12 r92 r39 rB9 r18 r16 r98 r96 r14 r94 r19 r99 r16 r96 r14 r94 r39 rB9 r15 r95 r12 r92 r14 r94 r34 rB4 r39 rB9 r2A r32 rB2 rAA r15 r95 r39 r30 rB0 rB9 r0E r8E r18 r98 r14 r94 r23 rA3 r12 r13 r92 r93 r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r0E r8E r1C r9C r1C r9C r2A r14 rAA r23 r94 rA3 r12 r39 r01 r0F r10 r1E r1D r2A r2B r03 r11 r1F r20 r2C r2D r05 r06 r13 
  > ERR:FF
  > r3E 
  > ERR:FF
  > r08 r00 !CS1_ERR!
    I52226 X52728 rAA W52763 r42 W52783 r01 W52ERR:12
    ID:F10 rrr12 r1F r20 r2C r05 r13 r14 
  > ERR:FF
  > rC2 r92 r42 r12 rC2 r92 r12 r92 r12 r92 r12 r92 r42 rC2 r42 rC2 r12 r92 r42 rC2 r12 r92 r12 r92 r42 rC2 r96 r16 r38 r96 r16 r96 r16 r39 r98 r45 r50 4D ERR:19
  > rr2B 12 
  > ERR:FF
  > 
  > ERR:FF
  > 
  > ERR:FF
  > rB4 rB9 99 
  > r81 ERR:FF
  > r9E ERR:FrrrrA4 A3 rrrrrrrr4 rB3 A0 9D r8 9E r9F 85 
  > ERR:FF
  > rA4 r32 r91 rA1 r97 r8F r87 rB0 r17 r97 r98 r12 r17 rB2 r97 r92 r93 r13 r93 r13 r93 r13 r93 r95 r15 r13 r93 r13 r93 r95 r15 r13 r93 r95 r15 r15 r95 r15 r95 r15 r95 r15 r13 r93 r95 r15 r95 r15 r95 r15 r95 r15 r13 r93 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r95 r15 r13 r93 r13 r13 r13 r13 r13 r13 r13 r93 r13 r93 r13 r93 r95 `

DalvikVM, 2020-05-10:
OK. I think I get basically what you're saying. I have some reading ahead it looks like! Thanks again for you're help sir!

DalvikVM, 2020-05-10:
I think there may be an issue with my keyboard, Soarers is now freaking out. There may be a bad connection in my wiring job. I have my keyboard open for the resistors and i think I may have wiggled one of the wires loose. In light of this, the data may be inaccurate. Probably wait for i-am-shodan before doing anything. Meanwhile, I have some soldering to do.

tmk, 2020-05-10:
@DalvikVM 
Are you using Teensy2.0?

`ERR:FF` means 'keyboard buffer full' or `key detecting error` according to Technical reference. Keyboard seems to register many keys at one time and go mad. I'd like to suspect keyboard hardware side. I don't think converter can read what keyboard sends.

Soarer's firmware works with same setup?

And you can quote log text easily using ```(three backquotes).
https://help.github.com/en/github/writing-on-github/basic-writing-and-formatting-syntax#quoting-code

tmk, 2020-05-10:
@DalvikVM 
OK. I see. Yes, you need to check wiring and solder joint probably.

DalvikVM, 2020-05-10:
I am using a teensy 2.0 now. The feather won't work with any of the firmwares properly. I think there is something not right somewhere. Soarers worked fine till about 5 minutes ago. I think there may be a loose connection.

DalvikVM, 2020-05-10:
Sorry about the code formatting thing. This is my first go with posting on GitHub (as you probably can tell) 😬

tmk, 2020-05-10:
My prebuild firmwares are compiled with 16MHz configuration and they don't work with Adafruit Feather with 8MHz crystal. Current firmware is written for controller with 16MHz crystal. Use Teensy2.0.

DalvikVM, 2020-05-10:
I got an afternoon of running Ethernet, so I'll have to get back to you tomorrow once I get the wiring fixed.

i-am-shodan, 2020-05-10:
> "A bit old firmware"(which I linked above) with the error check.
https://github.com/tmk/tmk_keyboard/blob/ibmpc_startbit_error/converter/ibmpc_usb/binary/ibmpc_usb_atmega32u4.hex

> The latest firmware with the error check. https://github.com/tmk/tmk_keyboard/blob/ibmpc_error_handling/converter/ibmpc_usb/binary/ibmpc_usb_atmega32u4.hex

I’m not sure if this is good news or bad news. I ran both of the firmwares and even checked the version info with the magic+V command to be sure I flashed the right ones.

Neither firmwares showed an error! I held down Q for around 3 minutes must of been a couple of hundred key presses.

I’m assuming your interrupt handler in these builds is quicker? Do you have any good tests I could run that might trigger it?

tmk, 2020-05-11:
Thanks for the test.

I believe the interrupt handler works at same timing as before and it still misses catching the start bit(0). It seems like the error checking(and recovery) code works somehow, but it fails to display the error message probably.

Good to hear that there is no difference between the two revisions for firmware. I'm working on merging this error checking code into master branch in addition to other updates.

tmk, 2020-05-11:
I just updated repository.

This is prebuilt firmware file of the latest repo.

https://github.com/tmk/tmk_keyboard/blob/42b7bf894253db095bca0e29cd3d9de912ba4131/converter/ibmpc_usb/binary/ibmpc_usb_atmega32u4.hex

@i-am-shodan 
Can you test that when you have time?

I think the start bit(0) error will happen with holding down Q for a few mins as before. And you will see its error message like `ISR:xxxx` in hid_listen log now unless I did something wrong.

i-am-shodan, 2020-05-11:
@tmk here you go

[log.txt](https://github.com/tmk/tmk_keyboard/files/4610562/log.txt)

I had to hold down Q for a few minutes.

DalvikVM, 2020-05-11:
So, after a rewire ( it looks like a loose solder joint was doing some wacky stuff to the converter) I got the keyboard working just fine with Soarers again. I reflashed TMK and held the q button for two minutes. No issues. But as I'm typing this there was something that just popped up in the hid output Ill post here. 

```
Waiting for new device:...
Listening:

TMK:b59f7655/LUFA

USB configured.

Keyboard start.
I222 X723
PROTO:02
rAA W746 R1247 wF2
ERR:11
ID:FFFF(1)
S1282 L1283 r0E r8E r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r90 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 [... cut ...] 12 r13 r92 r12 r93 r92 r34 rB4 r39 rB9 r2A r0A r8A r23 rA3 rAA r18 r19 r98 r99 r12 r92 r21 rA1 r16 r96 r26 rA6 r26 rA6 r15 r95 r39 rB9 r11 r91 r17 r97 r14 r23 r94 rA3 r14 r39 rB9 r94 r23 rA3 r12 r92 r39 rB9 r19 r99 r13 r93 r18 r98 r19 r99 r12 r13 r92 r39 r93 rB9 r21 rA1 r18 r98 r13 r93 r32 rB2 r1E r9E r14 r94 r14 r94 r17 r31 r97 r22 rB1 rA2 r34 rB4 r0E r8E r0E r8E r39 rB9 r14 r94 r0E r8E r2A r0B r8B rAA r1C r9C r1C r9C
```

Ill keep using it and see what else pops up.

Thanks again for the help!

DalvikVM, 2020-05-11:
So, another bug I found is the Num Lock button throws an error when pushed and the 0 button doesn't output anything outside of hid_listen

```
ERR:11
rC5 r45 wED
ERR:11
rC5 r45 wED
ERR:11
```

i-am-shodan, 2020-05-11:
I’m now running master with a custom payout but it looks like Num Lock is working AND outputting an error.

0 works for me when NumLock is pressed(but remember I have a custom layout)

tmk, 2020-05-12:
@i-am-shodan @DalvikVM 
Thanks for the report.

The start bit(0) error recovery logic seems to be work well. The `ISR:xxE0` in log indicates start bit(0) error occurs more occasionally than I expected.

The `wED` is 'set indicator' command and it will fail with `ERR:1x`  because XT keyboard doesn't support the command. This will happen with NumLock, ScrollLock and CapsLock. This error may be harmful for XT keyobards and it stops working perhaps. I'll get rid of the command for XT keyboars soon later.

Also can you test for 'keyboard buffer full'(Overrun) error?
Press and release many keys with your palm as possible, you will see `!OVERRUN!` in log like below. Confirm that there is no stuck key or weird behavior after that.

```
Waiting for new device:........
Listening:

TMK:b59f7655/LUFA

USB configured.

Keyboard start.
I629 X1130
PROTO:01
r3D W4256 r35 W4260 r2C W4264 R4264 wF2 rFA rBF rBF ID:BFBF(3)
S4269 wF8 rFA L4272 r44 r42 r3E r3C r00
!OVERRUN![r00 b:4B43 r4B !FULL! rFF ]

ERR:40
r4D rF0 r4D rF0 r4B r00
!OVERRUN![b:0000 r00 b:00F0 r00 !FULL! rFF ]

ERR:40
r35 r2C r3E r3D r3C r34 r00
!OVERRUN![]
r44 r32 r4B r43 r42 r4D r41 r33 r29 r3A rF0 r4D rF0 r42 rF0 r41 r00
!OVERRUN![r00 b:F04B rF0 !FULL! rFF !FULL! rFF ]
```

DalvikVM, 2020-05-12:
Well, I got this as soon as plugged in my keyboard this round.

```
Listening:

TMK:b59f7655/LUFA

USB configured.

Keyboard start.

ERR:F0

PROTO:02
I258 X760 rAA W788 r01 W807 r0F W809 R809 wF2
ERR:12
ID:FFFF(1)
S821 L821
PROTO:03
r1E
PROTO:02
r1D r2A r03 r11 r1F r20 r2C r05 r13 r21 r22 r2F r07
ERR:F0
!FF! rFF
!OVERRUN![rC7 ]
r47 rC7 r47 rC7 r47 rC7 r9A r1A r9A r1A
Device disconnected.
```

I wonder if theres a short somewhere...

tmk, 2020-05-12:
@DalvikVM 
I think so. It may be your wirings or keyboard hardware itself perhaps.

What is actually your keyobard model number?
If you took the keyboard apart before check grounding between switch assembly, controller, bottom plate and cables. Bad ground connection can cause random key press from my experience with other model F.

i-am-shodan, 2020-05-12:
Log below.

Waiting for device:
Listening:
r0E r8E r14 r16 r09 r04 r11 r05 r06 r13 r21 r15 r0A r24 r0B r03 r12 r23 r22 r17 r20 r2E r2B r1E r2D r32 r1F r2F r33 r2C r25 r25 r25 r25 r25 r25 r25 r25 r25 r25 r25 r25 r10 r0F r02 r01 r18 r1D r26 r39 r35 r19 r34 r34 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r27 r82 r81 r99 r27 rA7 r8F r90 r9D r0F r10 r01 r02 r02 r02 r02 r82 r81 r01 r02 rA5 rA6 r08 r44 r98 r92 rAF rAE r88 r9F rA0 rAD r86 r97 rA2 r85 r95 rA1 r05 r12 r06 r17 rC4 r15 r82 r81 r8F r90 r21 r22 r2E r2F r2D r83 r91 r84 r92 r85 r86 rAF r0F rA2 rAE r44 r95 rA1 rAD r93 r01 rB2 r3A r43 rA3 r10 r11 r02 r05 r06 r42 r2A r03 r04 r13 r41 r12 r21 r22 r52 r3E r3D r15 r23 r23 rBD r23 rBE rC1 r23 rC2 rA3 r97 rB3 rB4 r95 r89 r8A rA4 rD2 r82 rAA r92 r93 rA1 rA2 rB9 rB5 rBA rC3 r90 r83 r84
ERR:F0
!FF! rFF
!OVERRUN![r81 b:C48F rC4 !FF! rFF rAB r94 ]

ERR:F0
r1F r20 r9F rA0 r1E r1F r20 r9E r9F rA0 r1D r1E r9D r9E r0E r8E r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r10 r90 r1D r1E r9D r9E r0E r8E

i-am-shodan, 2020-05-12:
I’ve now been running the ibmpc firmware with a custom key map for a day as my main keyboard. Some thoughts:
- Best firmware yet
- I checked the hid_listen logs at the end of the day. I did notice some start(0) errors but I didn’t notice this during use.
- No stuck keys or phantom key presses yet!
- I used to have a rare issue with Soarers converter where a 1 might occasionally come out as a double keypress. I no longer get this issue.

I would probably get rid of the XT converter source tree just to prevent people trying to use it.

DalvikVM, 2020-05-12:
@i-am-shodan 
I'm glad to know it's working! That means my issues are hardware related! (Something I can actually fix!)

@tmk I do believe you are correct. My keyboard came with the cord hacked off and I've been using the wire and end that was left inside the keyboard to attach to the teensy. When I move the wires by the keyboard, it freaks out, so the connector is bad.

I've always used the connectors when rebuilding these keyboards, do either of you have a suggestion to replace them?

Thanks!

i-am-shodan, 2020-05-12:
My teensy is hanging off the header inside. The spiral cord was just too heavy for my liking.

DalvikVM, 2020-05-12:
Yeah, I think its gonna work now. I forgot to reinstall the ground screw in the main board. It must've had good enough of a connection to function for a while, but finally got loose again when I pulled it apart this time. 

Thanks for the help, both of you!

Would it be hard to add these code changes into qmk? Would it be something I could do?  I'm trying to build a bluetooth version with Adafruit's ATMEGA32u4 with bluetooth. QMK has the built in option to use it.  OK, I'm lazy, I get it. ;P

 If its a big deal, don't bother.

Here's the key mash:

```
ERR:19
r08 r22 r15 r16 r23 r09 r17 r19 r1D r2A r14 r21 r24 r11 r12 r20 r2C r2D r13 r2F r0A r1E r2B
ERR:F0
!FF! rFF
!OVERRUN![r05 !FULL! rFF ]

ERR:40
r31 r9D rAA r92 r9F rA0 rAD r85 r93 rAF rB1 rA4 rB2 r8B r99 rB8 r9E rAB r91
ERR:F0
!FF! rFF
!OVERRUN![r88 !FF! rFF ]

ERR:F0
```

DalvikVM, 2020-05-12:
Here's a pic of what I've got internally btw, if you're interested...
[image]

DalvikVM, 2020-05-13:
Ok. I'm partway through porting the ibmpc_usb over to qmk. Provided everything goes well I should have a working version soon...

tmk, 2020-05-13:
@DalvikVM 
Yes, the grounding screw was a must in my Model F too.
Good luck with QMK. Backport to TMK if you have a bug in my code.

tmk, 2020-05-22:
Can anyone test fixed firmwares with IBM XT keboard?
Hold down Q key for  a few minutes and see if `PRT:23` shows up in hid_listen log output.

That log output means that the firmware still misses start bit.

1. This prevents other interrupt routines from intevening in converter interrupt routine.
https://github.com/tmk/tmk_keyboard/tree/ibmpc_timer0isr_fix/converter/ibmpc_usb/binary

2. This defines converter interrupt routine with assembly to reduce interrupt latency, in additon to 1.
https://github.com/tmk/tmk_keyboard/tree/ibmpc_isr_fix/converter/ibmpc_usb/binary

DalvikVM, 2020-05-23:
I'll give it a go as soon as I can. Might be a few days though. Thanks!

i-am-shodan, 2020-06-01:
So the first firmware looks like it’s fine. I held down Q for about 5 minutes, no issue.

The second one has problems. I flashed it twice and couldn’t get a single key out.

tmk, 2020-06-01:
Thanks for the test.
Didn't you see any `PRT:23` in hid_listen log during the first test? That would be good indication if the message is not there.

Did the second one display anything in hid_listen log? It works with my AT keyobards2

i-am-shodan, 2020-06-01:
I didn’t see any PRT:23 messages in the log and in the second firmware the teensy didn’t even come up as a USB device.

tmk, 2020-06-02:
OK. I'll update firmware in main branch with change of the 1. 

I can't know why the second firmware doesn't work but
its change is not needed and not preferable if the former one works.

tmk, 2020-06-03:
Thank you, guys.
I updated repository finally and am closing this issue.

Feel free to open new issue if you find any problem.

closed 2020-06-03 by tmk