tmk/tmk_keyboard #637: windows 10 terminal converter

dehghanGit, 2019-11-26:
terminal usb converter does not work in windows 10.

tmk, 2019-11-26:
The converter works on other OSes?

dehghanGit, 2019-11-27:
I have no other OS.
But the Soarer converter works well in my Windows 10.
the same issue: https://github.com/tmk/tmk_keyboard/issues/485

tmk, 2019-11-28:
Then, toolchain version causes perhaps. Some Mac users have reported, iirc.
What version compiler did you use? I still use gcc 5.4.0, btw.

dehghanGit, 2019-11-28:
[avr-gcc (GCC) 5.4.0] and [avr-gcc (GCC) 4.9.2] and [avr-gcc.exe (WinAVR 20100110) 4.3.3]
Same result :/

tmk, 2019-11-29:
Hmm,
Can you supply detail of your keyboard like model number?
The terminal conveter supports only 'Code Set 3' while Soarer's supports CodeSet 1, 2 and 3. If your keyobard is not terminal keyboard(CodeSet3) it doesn't work with terminal converter.

And, does your converter have pull-up resistors? Without them the converter may miss singnal, TMK firmware can read data lines at different timing than Soarer's.

Also debug prints on hid_listen would be helpful.

dehghanGit, 2019-11-29:
IBM module m, terminal RJ45
Model number: 1395764

No I didn't use any resistor.  I connected the board directly to the keyboard.  Please tell me resistors value and how to connect or give me any article link to test again.

tmk, 2019-11-29:
OK. That keyobard should speak terminal scan code set 3.

I'd recommend 1Kohm for pull-up resistors.
But try hid_listen to get debug prints before soldering the resistors. I think TMK also can read signal correctly without the resistors when Soarer's can. I'll tune my code if needed.

Comparing debug prints on hid_listen from TMK firmware and Soerer's would be very helpful.
https://www.pjrc.com/teensy/hid_listen.html

dehghanGit, 2019-11-29:
Soarer's
keys: Ctrl, Shift, Alt, d, o, o, Up, Down, Right, 5, 0, 0, F1, F5
Listening:
r11 +E0 dE0
rF0 r11 -E0 uE0
r12 +E1 dE1
rF0 r12 -E1 uE1
r19 +E2 dE2
rF0 r19 -E2 uE2
r44 +12 d12
rF0 r44 -12 u12
r44 +12 d12
rF0 r44 -12 u12
r44 +12 d12
rF0 r44 -12 u12
r63 +52 d52
rF0 r63 -52 u52
r60 +51 d51
rF0 r60 -51 u51
r6A +4F d4F
rF0 r6A -4F u4F
r73 +5D d5D
rF0 r73 -5D u5D
r70 +62 d62
rF0 r70 -62 u62
r45 +27 d27
rF0 r45 -27 u27
r07 +3A d3A
rF0 r07 -3A u3A
r27 +3E d3E
rF0 r27 -3E u3E
---------------------------------------------------------------------
TMK terminal converter:
Output with or without keyboard connection:
(Even when the microchip is connected without a keyboard):

Keyboard start.
wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wwFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF wFF...

It will continue until the microchip plug out.

dehghanGit, 2019-12-14:
noting?

tmk, 2019-12-15:
Hmm, TMK seems to fail to reset the keyboard at startup for some reason. I have no idea. Can you share your firmware file? I'd like to try it on my site.

Also try this new another TMK converter implementation.
https://geekhack.org/index.php?topic=103648.0

closed 2022-06-07 by tmk