tmk/tmk_keyboard #640: [adb2usb] weird Keyboard recognition

waszqba, 2020-01-29:
Hello,
I have built the adb2tmk converter using Pro Micro Arduino compatible and TMK with the `unimap_plain` slightly tailored to suit my needs and `config.h ` suited with VID and DEV_ID to match my ISO AEK II directly soldered to the pro micro with 1KOhm pull-up resistor.

There are 3 scenarios which occur:
- Directly after flash:
  -  L/R Modifiers are distinguished
  - `§` and grave keys are swapped
  jumping to bootloader and back has no effect
- Unplugged and plugged USB
  - L/R modifiers are not distinguished
  - `§` and grave keys are still swapped
  at this stage, jumping to bootloader and waiting 8 seconds for the controller to reboot brings us to scenario #3
- Replugged and rebooted
  - L/R modifiers are distinguished
  - `§` and grave keys seem to be in their places
  rebooting gives no further effect safely looping us back to scenario #3

Another weird thing is that the `§` is being sent when the GRV is set in keymap and the grave key is emitted when defined as KC_NUBS, but that's a small thing I can cope with and don't mind as long as I can just set the keymap and rely on it

tmk, 2020-01-29:
Can you test with default firmware downloaded from TMK keymap editor without edit?
http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?adb_usb_rev1

And share output from hid_listen when the problems occurs.
https://github.com/tmk/tmk_keyboard/wiki#debug-console

And show me pic of your ISO AEK, I don't know much where the § is.

waszqba, 2020-01-31:
Sure, will try on weekend and let you know :)

waszqba, 2020-02-03:
I wasn't able to run the PJRC debug tool, I got an error stating I have a bad CPU (???)

However after flashing the rev1 you linked I got similar results:
- Flashed -> correct position
- Replugged - wrong position
- rebooted - correct position

Bad news is that I cannot flash my device anymore, since trying to jump into bootloader simply reboots the keyboard without any time to actually spend in bootloader. Any help?

This is my layout:
[image]

waszqba, 2020-02-03:
I tried the second revision from the editor, but I got nothing, the keyboard wasn't even recognized as a HID, so I'm back to my version for now, but open to experiments to help identify the issue

tmk, 2020-02-03:
hmm, I guess your hardware of converter has something wrong. Show me photo of your pro micro and wiring this time. Rev2 firmware is not for your converter, it features different controller ATmega32u2.

And how did you change files when you built firmware? Patch output of diff command would be helpful if the difference is slight.

tmk, 2020-02-03:
And I am just curious, the AEK layout is for German or other language?

waszqba, 2020-02-03:
Here are changes
```patch
Index: converter/adb_usb/Makefile
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- converter/adb_usb/Makefile	(revision 3b21f85e9bb731adcbd40be1b93b7d39ef7748c5)
+++ converter/adb_usb/Makefile	(date 1579631679439)
@@ -1,5 +1,5 @@
 # Target file name (without extension).
-TARGET ?= adb_usb
+TARGET ?= adb_usb_pro_micro
 
 # Directory common source filess exist
 TMK_DIR ?= ../../tmk_core
@@ -18,7 +18,7 @@
 # atmega32u4 	Teensy2.0
 # atemga32u4	TMK Converter rev.1
 # atemga32u2	TMK Converter rev.2
-MCU ?= atmega32u2
+MCU ?= atmega32u4
 
 # Processor frequency.
 #     This will define a symbol, F_CPU, in all source code files equal to the
 
Index: converter/adb_usb/config.h
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- converter/adb_usb/config.h	(revision 3b21f85e9bb731adcbd40be1b93b7d39ef7748c5)
+++ converter/adb_usb/config.h	(date 1580189587373)
@@ -19,12 +19,12 @@
 #define CONFIG_H
 
 
-#define VENDOR_ID       0xFEED
-#define PRODUCT_ID      0x0ADB
+#define VENDOR_ID       0x05AC
+#define PRODUCT_ID      0x020C
 #define DEVICE_VER      0x0101
-#define MANUFACTURER    t.m.k.
-#define PRODUCT         ADB keyboard converter
-#define DESCRIPTION     convert ADB keyboard to USB
+#define MANUFACTURER    Apple
+#define PRODUCT         Extended Keyboard II
+#define DESCRIPTION     ISO Apple Extended Keyboard II
 
 /* matrix size */
 #define MATRIX_ROWS 16  // keycode bit: 3-0
Index: converter/adb_usb/unimap_plain.c
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- converter/adb_usb/unimap_plain.c	(revision 3b21f85e9bb731adcbd40be1b93b7d39ef7748c5)
+++ converter/adb_usb/unimap_plain.c	(date 1580298227427)
@@ -16,10 +16,33 @@
 */
 #include "unimap_common.h"
 
+/* id for user defined functions */
+enum function_id {
+    LSHIFT_LPAREN,
+};
+
+enum macro_id {
+    RAMBOX,
+    TEAMS,
[... 136 lines cut ...]
+            // Shift parentheses example: LShft + tap '('
+            // http://stevelosh.com/blog/2012/10/a-modern-space-cadet/#shift-parentheses
+            // http://geekhack.org/index.php?topic=41989.msg1304899#msg1304899
+            if (record->event.pressed) {
+                if (record->tap.count == 0 || record->tap.interrupted) {
+                    register_mods(MOD_BIT(KC_LSHIFT));
+                }
+                return;
+            }
+            if (record->tap.count == 0 || record->tap.interrupted) {
+                return unregister_mods(MOD_BIT(KC_LSHIFT));
+            }
+            add_weak_mods(MOD_BIT(KC_LSHIFT));
+            send_keyboard_report();
+            register_code(KC_9);
+            unregister_code(KC_9);
+            del_weak_mods(MOD_BIT(KC_LSHIFT));
+            send_keyboard_report();
+            record->tap.count = 0;  // ad hoc: cancel tap
+
+            break;
+    }
+}

```

waszqba, 2020-02-03:
I'll take the photo tomorrow as I left my board at work. Regarding your question: according to [this article](https://support.apple.com/en-us/HT201794) it seems to be international English ISO layout

tmk, 2020-02-03:
Ah, It is 3.3V version, you have to close J1 solder jumper to supply 5V probably.

closed 2020-02-03 by tmk

waszqba, 2020-02-03:
It's 5v version, I double-checked that before the whole operation

tmk, 2020-02-03:
How did you checked?
You have to check voltage of VCC pin of the microcontroller.

waszqba, 2020-02-03:
yup, and it showed me 4.9V which I consider close enough since my multimeter tend's to lower the results, also, it has 16MHz quartz and AFAIK they are only made in 8MHz/3.3v and 16MHz/5v variants

tmk, 2020-02-03:
Are you sure? Pro Micro is totally mess for users unless you got it from Sparkfun or reliable sources.
Try close J1.

With Pro Micro you can save little money but often waste time :(

waszqba, 2020-02-03:
Well, anywhere I check there's info that 16MHz (present in my model) requires 5v to be reliable, when powering of USB I get the readings mentioned above, but I'll short the J1 just in case, after all it can't hurt it with the desired 5v, can it? :)

I'll let you know tomorrow how it went

tmk, 2020-02-03:
And you had better not to use the Apple's vendor/product ID with this converter, perticular on MacOS.  It may make the OS confused.

waszqba, 2020-02-03:
well my goal was to make it actually work as an apple keyboard so the OS won't ask me to go through that silly layout check routine, however I only changed it recently already after I noticed the issue with the keymap scramble (sounds like a title of some adventure movie :D)

waszqba, 2020-02-04:
INB4 I know my soldering is no engineering marvel
[image] [image]

I tried shorting out the J1 connector but no luck, the issue persists. About the wiring: I wired the pro micro directly to the lines the connector was wired to, everything was checked with multimeter and noted for reference in order not to mix the wires, so I'm pretty certain about that. All the keys, including Magic key, work, but this could be inferred from previous messages.

I'm afraid we'll have to reopen this issue :(

Still wondering though, why would it mess up the keyboard detection on power up but work correctly after soft reboot?

tmk, 2020-02-04:
Yes, if the problem still be there, your firmware setup causes probably. Then, I don't know why the default firmware didn't work for your converter.

The converter detects if keyboard is ANSI or ISO only when startup and MacOS does dirty trick under the hood to hide their trivial keyboard wierdness from users. See source code from Apple in this comments, it may explain.
https://github.com/tmk/tmk_keyboard/issues/35#issuecomment-41874635

The converter does the trick,  so that it will be problem if MacOS does it again.

The default firmware or plain built firmware(without the VID/PID setting) should work on MacOS.

waszqba, 2020-02-04:
But the thing is not only the characters are messed up on macOS, it seems that the controller itself messes up the keymap, as the momentary layer key changes places, so I don't think that it's the OS's thing. Also, modifiers are nod being distinguished properly in some cases, but after reboot they are, so there must be something in the detection mechanism. If there only was some way to actually debug it in code... I'll try the stock firmware on the bridged J1 at home and let you know if it changes anything. 

- [x]  stock firmware on J1 open
- [x] My firmware on J1 open
- [x] My firmware on J1 closed
- [ ] Stock firmware on J1 closed

waszqba, 2020-02-04:
Huh, funny thing, I tried stock firmware on another board and a fresh controller with J1 open, now it acts like this, but I'm unsure if this occurred on my main board, didn't test it, but probably acts the same:
- reconnect -> keys in place
- reboot -> keys swapped

waszqba, 2020-02-05:
tested on my main board with J1 closed and actually... it seems to work properly when quickly reconnected, but when slowly, the keymap is scrambled... the behavior of this seems so random...

waszqba, 2020-02-05:
I tried to flash my firmware with stock config on J1 closed and now GRV is being read as actual GRV, but the overall behavior hasn't changed, keys are swapped and modifiers are mixed on long reconnect and correct on quick

tmk, 2020-02-05:
What do you mean by 'reconnect'? Plug into USB port or hot-plug keyboard into the converter? You better connect keyboard to converter first and then plug into USB port every time.

Stock firmware should work for the grave and ISO key when you plug as mentioned above. For ISO keyboard, hot-plugging(without power cycle of the converter) is not supported at this time.

Bootloader of ProMicro by magic command is not officially supported yet, it is another issue. Use reset pin or button for a while.

tmk, 2020-02-05:
And output from hid_listen would be helpful. The tool shoul work if your setup is ok.

Test with Windows and Linux and know difference from MacOS, that would be helpful to understnad the problem also. I don't have any MacOS to test, btw.

I'm assuming you are testing this on MacOS, is that right?

waszqba, 2020-02-06:
yup, testing on macOS
although not supported, the magic bootloader jump works fine in my case
by replugging I mean the USB, the keyboard is hard-wired to the controller, so hot pluging is not an option
I'll try to get the hid_listen to work, but I need to tinker around it, because it throws me that CPU error of sort...

waszqba, 2020-02-11:
Sooo I tried many ways on many macs, but the only version of hid_listen I got working is the one wrapped in node.js

Couldn't find any info on what should normally be expected on the output, but I hope this will be somewhat informational. I can parse the data returned from hid_listen ofc and make id read anything you want

[debug_diff.txt](https://github.com/tmk/tmk_keyboard/files/4184345/debug_diff.txt)
[debug_improper.txt](https://github.com/tmk/tmk_keyboard/files/4184346/debug_improper.txt)
[debug_proper.txt](https://github.com/tmk/tmk_keyboard/files/4184347/debug_proper.txt)

tmk, 2020-02-11:
hmm, it looks something wrong with hid_listen on MacOS. What's is your MacOS version? I guess the tools is not compatible with the latest MacOS, pherhaps. But I never heard it from the comunity yet. Compiling hid_listen yourself on your MacOS may solve this. You can get soruce of the tool here: https://github.com/PaulStoffregen/hid_listen

Also test on Windows may tell you something useful.

For example, this is output from the tool when the converter start up with M0116 keyboard.
```
Device disconnected.
Waiting for new device:......
Listening:

TMK:fb84cac5/LUFA

Scan:
 addr:2, reg3:6601

Keyboard:
hadler: 01, ISO: no

Mouse:

Scan:
 addr:2, reg3:6E01

USB configured.

Keyboard start.
```

waszqba, 2020-02-11:
I'm on Catalina 10.15.3
Will try to compile @ home & share the output. As far as I understand you wand just the output from the keyboard boot, right?

waszqba, 2020-02-11:
Having troubles compiling this program, it was written in 2008 and made for 32-bit arch which is unsupported by current OS version :( Will try to make it but I'm afraid we'll have to try to read the output from the nodyfied version

waszqba, 2020-02-11:
well, I got this darn thing compiled and working
At first I was unsure of the output given by the node version, but it seems to be pretty much the same. However, here it is, the output of the original hid_listen, when I plug in the board:
```
Waiting for device:.......
Listening:

TMK:d8228a37+/LUFA

Scan:
RRRRRRRRRRRRRRRR
Keyboard:
Rhadler: 00, ISO: no
R
Mouse:
R
Scan:
RRRRRRRRRRRRRRRR
USB configured.

Keyboard start.
RRR
```
and this is that I see after quickly reconnecting the keyboard:
```
Waiting for new device:.
Listening:

TMK:d8228a37+/LUFA

Scan:
 addr:2, reg3:6A05

Keyboard:
hadler: 05, ISO: yes

Mouse:

Scan:
 addr:2, reg3:6803

USB configured.

Keyboard start.

```
just after this I made a soft reboot and this version came up:
```
Waiting for new device:.........
Listening:

TMK:d8228a37+/LUFA

Scan:
 addr:2, reg3:6103

Keyboard:
hadler: 03, ISO: no

Mouse:

Scan:
 addr:2, reg3:6703

USB configured.

Keyboard start.

```
after normal, slow replug, result of which can be seen in the first paste, soft reboot seems to result in exactly the same situation as directly after hot-replug:
```
Waiting for new device:.........
Listening:

TMK:d8228a37+/LUFA

Scan:
 addr:2, reg3:6805

Keyboard:
hadler: 05, ISO: yes

Mouse:

Scan:
 addr:2, reg3:6203

USB configured.

Keyboard start.

```

tmk, 2020-02-11:
It doesn't recognize your keyobard as ISO correctly for some reason.

R in the first log means connection is not stable basically, I also see a few R when plugin-in keyboard but the R's in the log seem to me be too many. If you don't use proper connector and solder them, instead of using jumper wires.

Also pro micro's startup sequence may causes. Pro micro has two variant of its bootloader, old bootloader requires 8sec before start user application or converter firmware in our case while new Sparkfun one takes only 750ms. I think pro Micro is shipped with new one even if it is a clone but I'm sure.
https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/troubleshooting-and-faq#ts-reset

Meanwhile, Teensy and my TMK converters start up user application immediately without bootloader intervention. I'm not sure this causes acctually, though.

Keyboard is recognized by this part of code, and you can tweak its timing by changing this wait line.
https://github.com/tmk/tmk_keyboard/blob/master/converter/adb_usb/matrix.c#L83-L97
https://github.com/tmk/tmk_keyboard/blob/master/converter/adb_usb/matrix.c#L76

I don't have any Pro Micro, MacOS and ISO ADB keyboard to test this ufnortunately. You may want to ask help from other ISO keyboard users with MacOS in the thread in Geekhack.

Let us know if you have some clue.

> TMK:d8228a37+/LUFA
Are you including your change for this test?
Please test with plain firmware without any of your change till we get default firmware worked in you environment as expected. I don't want to be confused.
You can use Makefile.rev1 to build firmware for your pro micro like `make -f Makefile.rev1`

tmk, 2020-02-12:
Can you share hid_listen binary you built?
It seems that MacOS 10.15 drops support for 32-bit binary and hid_listen for Mac hosted in PJRC site is 32-bit. Users on Catalina will have same problem with the tool. If you can share it would be helpful for them.

waszqba, 2020-02-12:
I'll test with stock firmware. One tap on reset has no visible effect, only doubletap lands me in bootloader on my board, so I guess I have the newer version.
By stock/plain you mean the one downloaded from editor or just to compile a clean one from source at master?

waszqba, 2020-02-12:
> Can you share hid_listen binary you built?
> It seems that MacOS 10.15 drops support for 32-bit binary and hid_listen for Mac hosted in PJRC site is 32-bit. Users on Catalina will have same problem with the tool. If you can share it would be helpful for them.

Sure, how do you want it shared? I can either host it on Mega and share a link or put it somewhere else

waszqba, 2020-02-12:
stock firmware from editor:
after plug in:
```
TMK:fb84cac5+/LUFA

Scan:
RRRRRRRRRRRRRRRR
Keyboard:
Rhadler: 00, ISO: no
R
Mouse:
R
Scan:
RRRRRRRRRRRRRRRR
USB configured.

Keyboard start.
```
after hot-replug:
```
TMK:fb84cac5+/LUFA

Scan:
RRRR
Keyboard:
hadler: 05, ISO: yes

Mouse:

Scan:
 addr:2, reg3:6A03

USB configured.

Keyboard start.
```
Soft reboot directly after that:
```
TMK:fb84cac5+/LUFA

Scan:
 addr:2, reg3:6503

Keyboard:
hadler: 03, ISO: no

Mouse:

Scan:
 addr:2, reg3:6403

USB configured.

Keyboard start.
```
slow plugin, soft reboot (like hot-replug):
```
TMK:fb84cac5+/LUFA

Scan:
 addr:2, reg3:6005

Keyboard:
hadler: 05, ISO: yes

Mouse:

Scan:
 addr:2, reg3:6003

USB configured.

Keyboard start.
```

waszqba, 2020-02-12:
freshly pulled master, built from scratch, completely clean, built using makefile.rev1:
after plug-in
```
TMK:6209ceeb/LUFA

Scan:
RRRRRRRRRRRRRRRR
Keyboard:
Rhadler: 00, ISO: no
R
USB configured.

Keyboard start.
RR
```
after hot-replug:
```
TMK:6209ceeb/LUFA

Scan:
 addr:2, reg3:6405

Keyboard:
hadler: 05, ISO: yes

USB configured.

Keyboard start.
```
Soft reboot directly after that ( this time the listener catches the boot loader command)
```
bootloade
Device disconnected.
Waiting for new device:.........
Listening:

TMK:6209ceeb/LUFA

Scan:
 addr:2, reg3:6103

Keyboard:
hadler: 03, ISO: no

USB configured.

Keyboard start.
```
slow plugin, soft reboot (like hot-replug):
```
TMK:6209ceeb/LUFA

Scan:
 addr:2, reg3:6905

Keyboard:
hadler: 05, ISO: yes

USB configured.

Keyboard start.
```

waszqba, 2020-02-12:
ok, The initialization wait time seems to be a viable workaround for me for now. I've tried 150 ms but 1 oof 3 times the keyboard wouldn't boot properly, so I set it to 200ms and now it works like magic, no hiccups, every boot, hard or soft, is correct with handler 5 and iso detected :D
Thank you @tmk for all the time you've spent on this topic and the help you provided <3
I am really grateful and I hope I can pay the favor back somehow. If there is any way I can help figuring out bugs or contributing anyhow to this project, I am willing to. Giant Kudos to you!

tmk, 2020-02-12:
Great to hear that.
I see, if you don't see the 'R' error with the longer delay works  the errors meant slow ISO AEK startup, not your wiring problem nor pro micro bootloader. It seems the converter starts talking too early before the keyboard wakes up.
I'll update code to make the delay longer, probably 1000ms would be safe for other boards considering Pro micro booloader 750ms wait.

And again, could you share your hid_listen binary?

waszqba, 2020-02-13:
In my experience 200ms would be just enough, I don't see a single `R` in the log with this setting :)
And with the 150ms Even when I got a couple of `R`s in the first row the keyboard worked properly (those 2 of 3 times), so I think those 200ms will be sufficient for most, if not all boards.
and [here's](https://mega.nz/#!HlcgESwT!q4b1lBmHhAF7ih-SS0qmS1Qwha2wjjoKvn2BB_9q7rk) the binary you asked for. It's in my 'Public Domain' directory so I won't delete it when cleaning my cloud someday

tmk, 2020-02-13:
I confirmed that there is the problem even with US ANSI layout AEK/AEKII, and 200ms is enough for AEKs. AEK startup is a bit  slower than other ADB keyboards for some reason. I updated the delay codes in repo with commit  ce5e565 for AEKS using 1000ms for safety of unknown devices.

And thanks for the hid_listen binary. That would be helpful for users on the latest MacOS.